AndyQ/NFCPassportReader #104: 1.1.4 does not work on iOS 12 (library not loaded)

Jeroenbb94, 2021-04-29:
When running version 1.1.4 on an iOS 12 device you get the following crash message on startup:

```
dyld: Library not loaded: /System/Library/Frameworks/CryptoTokenKit.framework/CryptoTokenKit
  Referenced from: /private/var/containers/Bundle/Application/80449E41-D4CD-4FA5-936A-A37C043B0487/CMApp.app/Frameworks/NFCPassportReader.framework/NFCPassportReader
  Reason: image not found
```

Adding the CryptoTokenKit framework to the app won't help either, because the framework is iOS 13 only (https://developer.apple.com/documentation/cryptotokenkit). Could 1.1.4 be resolved to support iOS 12 again or can de pod spec be updated to minimal iOS 13?

Jeroenbb94, 2021-04-29:
The crash is happing because of the import CryptoTokenKit in Utils.swift. I'll try to change it to the following:
```
#if canImport(CryptoTokenKit)
    import CryptoTokenKit
#endif
```

Jeroenbb94, 2021-04-29:
Above didn't resolve it, though adding `-weak_framework "CryptoTokenKit"` to the spec.xconfig in the pod spec did since it is a framework that is being used in the codebase but not specified for cocoapods.

zaidahmed2468, 2021-05-31:
Do you have an eta when you will fix this issue (give support to iOS 12) ?

AndyQ, 2021-07-29:
This should be fixed in 1.1.6 (due soon)

closed 2021-07-29 by AndyQ