AndyQ/NFCPassportReader #110: Slower passport read

danydev, 2021-06-21:
I just upgraded from 1.0.10 to 1.1.4 and I noticed that the auth flow is much slower.

With 1.0.10 I can auth in under ~500ms, while with 1.1.4 it's in the order of 4-6 seconds.

Note how PACE is successful, so it's not losing time with an unsuccessful authentication, it's just slower.

I have upgraded just to provide some fixes the library inserted in the last months, but this is kind of a big deal, there is a UX degradation for our users, so this is blocking our release.

random idea: maybe we can a settings to attempt first BAC and fallback to PACE if bac fails?

danydev, 2021-06-26:
@AndyQ do you have any insight or thoughts for this?

AndyQ, 2021-06-26:
Interesting - I've not actually noticed PACE being that much slower than BAC. 

It is doing more work and there is more back and forth between the passport and the device so it's possible that different passports may be slower than others. 

I had thought there was an option for disabling PACE but just checking looks like not however it would be fairly trivial to add. 

I'll try to get it in in the next release - which also includes a fix for some invalid ECDAa signatures - but not quite sure when that will be yet!

danydev, 2021-06-26:
ok. For what it worth I tried on an italian identity document.

Whenever there is the release I'll try again to confirm the performance problem. 

Eventually, I'll do a first attempt with BAC, and then if it fails, I'll try with the flag disabled. That will work for me. Thanks!

AndyQ, 2021-07-29:
I've just released v1.1.5 - this now supports the ability to disable both Chip Authentication and PACE if desired.

There are two now parameters on the readPassport method - shipCA and skipPACE.  These both default to false.

danydev, 2021-08-04:
While I implemented the feature with the new flag, I realised that's not the best solution. This because every time I call `readPassport` a new action sheet is displayed to the user, so it's not something I can do transparently.

To be clear, this is the UX with the current solution.
- call readPassport with pace disabled
- user sees iOS action sheet about reading NFC
- user puts the document near NFC reader
- BAC fails for any reason
- call readPassport with pace disabled
- user sees iOS action sheet about reading NFC
- user puts the document near NFC reader
- done

While ideally I would like to do something like
- call readPassport with priority to BAC auth
- user sees iOS action sheet about reading NFC
- user puts the document near NFC reader
- BAC fails for any reason
- NFCPassportReader automatically fallbacks on PACE
- done

Do you think we could do something about it? Maybe just replace the new `skipPACE` with something like `givePriorityToBAC`?

danydev, 2021-08-05:
Adding logs but with verbose this time:

Logs with PACE: https://pastebin.com/KGKTycw1
Logs with BAC: https://pastebin.com/mzNTX56k

AndyQ, 2021-08-05:
Interesting!  I've just tested on my passport and doing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-256 takes about 1.2 seconds.

2021-08-05 11:32:58.0750 - Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-256
...
2021-08-05 11:32:59.2930 - Restarting secure messaging using AES encryption

I can't test 3DES encryption as I don't have a passport that supports that sadly.

danydev, 2021-08-05:
@AndyQ I see...please save the pastebins above, since I'm not sure they contain sensitive information, I'm going to delete them in the next 30 min

AndyQ, 2021-08-05:
One other thing, from what I understand from the ICAO standards, PACE should be picked if available, then Chip Authentication done if available and the secure messaging channel setup by each should then be used. BAC should only be used if PACE/CA is not available OR there is an error.

AndyQ, 2021-08-05:
> @AndyQ I see...please save the pastebins above, since I'm not sure they contain sensitive information, I'm going to delete them in the next 30 min
You can remove them

danydev, 2021-08-05:
> One other thing, from what I understand from the ICAO standards, PACE should be picked if available, then Chip Authentication done if available and the secure messaging channel setup by each should then be used. BAC should only be used if PACE/CA is not available OR there is an error.

makes sense on paper, I didn't find the exact specification where this is specified. Anyway not all use cases must be ICAO compliant, I think that many applications just need to read the data, they may even not care about verifying signatures and stuff. As long as an SDK allows me to follow ICAO standard or applying variations, it's fine, I guess.

AndyQ, 2021-08-05:
Unfortunately switching round the order of authentication is non-trivial and would add much more complexity which I'm currently not willing to do. 

However, for info, the Async branch which I've pushed up whilst being iOS 15 specific - would make it much simpler to implement your scenario if you so wished.

closed 2021-08-05 by AndyQ

wellbranding, 2022-07-07:
@danydev Hello, we as well facing issues with the Italian passport. Can you check if yours is this one?
[](https://www.consilium.europa.eu/prado/en/ITA-AO-02003/index.html)
Have you resolved the issue?

danydev, 2022-07-07:
mine should be the latest https://www.consilium.europa.eu/prado/en/ITA-AO-02004/index.html not really sure since they are really similar.

The work around that I applied is basically this one

```
call readPassport with pace disabled
user sees iOS action sheet about reading NFC
user puts the document near NFC reader
BAC fails for any reason
call readPassport with pace enabled
user sees iOS action sheet about reading NFC
user puts the document near NFC reader
done
```

wellbranding, 2022-07-07:
@danydev Amazing news! Just to make it clear, have you tried this logic with different documents/countries? From my understanding this is a reverse approach from what we have in the library by default. In the library we first read with pace enabled, if it fails we do bac, while you on the other hand read with pace disabled, right?

danydev, 2022-07-07:
correct. This is a reverse approach to what the library does by default.

Yes, I tested it with several electronic documents (passports, identity cards) of several countries

wellbranding, 2022-07-07:
@danydev i talked to the team and we would gladly pay! This is especially important for us since we have a potential client from Italy, but the solution does not work and we want to debug it beforehand....

emirbeytekin, 2022-07-28:
Hello,
I tried with a Turkish passport and was able to withdraw data without any problems ON `skipFace: TRUE`

RobinFarmer, 2022-08-09:
@AndyQ This is interesting and even with the newest changes still an issue for Dutch passports.

PACE is very slow which in return causes timeouts. while BAC is almost instant in comparison.

```
2022-08-09 8:34:17.7160 - Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-256
2022-08-09 8:34:24.3080 - Restarting secure messaging using AES encryption
```

```
2022-08-09 8:36:43.2810 - Starting Basic Access Control (BAC)
2022-08-09 8:36:43.5250 - Basic Access Control (BAC) - SUCCESS!
```

AndyQ, 2022-08-09:
@RobinFarmer Are you able to see which bits of the code are taking the time?

RobinFarmer, 2022-08-09:
@AndyQ 

This is the full log:

```
2022-08-09 8:34:17.7160 - Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-256
2022-08-09 8:34:17.7540 - Doing PACE Step1...
2022-08-09 8:34:17.9260 - Doing PACE Step2...
2022-08-09 8:34:17.9260 -    Using General Mapping (GM)...
2022-08-09 8:34:17.9270 - Generating ECDH mapping keys from parameterSpec - 929
2022-08-09 8:34:17.9420 - Sending public mapping key to passport..
2022-08-09 8:34:20.0490 - Received passports public mapping key
2022-08-09 8:34:20.0500 - Doing ECDH Mapping agreement
2022-08-09 8:34:20.0680 - Doing PACE Step3 - Key Exchange
2022-08-09 8:34:20.0740 - Generated Ephemeral key pair
2022-08-09 8:34:20.0740 - Sending ephemeral public key to passport
2022-08-09 8:34:22.3470 - Doing PACE Step4 Key Agreement...
2022-08-09 8:34:22.3470 - Computing shared secret...
2022-08-09 8:34:22.3580 - Deriving ksEnc and ksMac keys from shared secret
2022-08-09 8:34:22.3580 - Generating authentication token
2022-08-09 8:34:22.3590 - Sending auth token to passport
2022-08-09 8:34:24.3080 - Auth token from passport matches expected token!
2022-08-09 8:34:24.3080 - Restarting secure messaging using AES encryption
2022-08-09 8:34:24.3080 - PACE SUCCESSFUL
2022-08-09 8:34:24.3080 - PACE Succeeded
```

It seems it's the sending commands that both take around 2 seconds to complete. Is this sufficient or do you want to see the actual bits?