AndyQ/NFCPassportReader #113: Active Authentication failing with error "Wrong length Le: SW2 indicates the exact length..."

pmitche, 2021-08-25:
There seems to be a problem with the APDU sent when issuing an INTERNAL AUTHENTICATE command through `TagReader.doInternalAuthentication` to perform Active Authentication. I get the following error message on my Norwegian passport which supports both CA and AA:

> .....
> TagReader [SM - unprotected] [0x99, 0x02, 0x6c, 0x80, 0x8e, 0x08, 0x55, 0x94, 0x80, 0x4e, 0x4a, 0x81, 0x7a, 0xca, ], sw1:0x6c sw2:0x80
> Error reading tag: sw1 - 0x6C, sw2 - 0x80
> reason: Wrong length Le: SW2 indicates the exact length - (exact length :128)
> doInternalAuthentication failed - Wrong length Le: SW2 indicates the exact
> length - (exact length :128)
> .....
> 

It seems as something related to this was fixed in #28, however a subsequent commit 223e50fb02692ad9de414fa837d63242de58795a caused the `SecureMessaging.buildD097` function to never be called when issuing the INTERNAL AUTHENTICATE command with `expectedResponseLength` of 256.

https://github.com/AndyQ/NFCPassportReader/blob/420ba74525f2f322982bd7ff82337816f44111f8/Sources/NFCPassportReader/TagReader.swift#L52-L58

https://github.com/AndyQ/NFCPassportReader/blob/420ba74525f2f322982bd7ff82337816f44111f8/Sources/NFCPassportReader/SecureMessaging.swift#L60-L63

If I force the `SecureMessaging.buildDO97` function to run (by for example setting `expectedResponseLength` to 255 in `TagReader.doInternalAuthentication` or reverting the added `&& apdu.expectedResponseLength < 256` from commit 223e50fb02692ad9de414fa837d63242de58795a) Active Authentication completes successfully.

Any ideas on the correct way to address this?

Passport details:
 * Norwegian
* Supports BAC, not PACE
* Supports PA, AA and CA
* DG1, DG2, DG3, DG14, DG15

invisible66, 2021-09-01:
Hey, did you find any solution?

AndyQ, 2021-09-03:
I do recall there were some issues with Chip Authentication when the LE was 256 - but I can't remember exactly now. 

However, looking I think it may be a bug that only occurs on some passports!  I think that removing the check for length < 256 is the correct fix however I'll do some testing and to check.

invisible66, 2021-09-03:
When I removed `apdu.expectedResponseLength < 256 ` condition from `SecureMessaging.protect` Active Authentication completed successfully.. but I'm not sure I'm doing it right..

pmitche, 2021-09-06:
Great, thanks! I'll remove the check for length < 256 locally until there's an update available. I'd be happy to help if there's anything I can do

AndyQ, 2021-09-16:
I've tested this on all the passports I have access to and seems to work fine - just released 1.1.7 with this fix so closing.

closed 2021-09-16 by AndyQ