AndyQ/NFCPassportReader #116: Error reading tag: sw1 - 0x6A, sw2 - 0x86. when scan idcard

enaifuose, 2021-09-16:
Hi,

first, i want to thank u for the great work you do.

i m trying to test the NFC scan in id card. but i get the following errors : 

[CoreNFC] 00000002 827685a0 -[NFCTagReaderSession setAlertMessage:]:90  (null)
Error reading tag: sw1 - 0x6A, sw2 - 0x86
reason: Incorrect parameters P1-P2
Re-selecting eMRTD Application
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
Error reading tag: sw1 - 0x6D, sw2 - 0x00
reason: Instruction code not supported or invalid
ERROR - Instruction code not supported or invalid
BAC Failed

Any help please.

danydev, 2021-10-06:
When reporting issues it's kind of useful to specify type (id card, passport, etc..) and country of the document you are using

shakhzodiosdev, 2022-03-01:
The same issue while reading driving license. Have you solved?

shakhzodiosdev, 2022-03-02:
@enaifuose Any solutions?

RobinFarmer, 2022-08-04:
Ran into the same problem using a Latvian ID card.

@shakhzodiosdev did you happen to find a solution for this?

shakhzodiosdev, 2022-08-04:
As I remember you should try to add ISO7816 application identifiers for NFC Tag Reader Session in your plist :
[image]

your plist should look like that ) I am not sure that it will help you but anyway you can try )

RobinFarmer, 2022-08-04:
Thanks just tried it but unfortunately the problem remains. 

I did already have the the following:

```
A0000002471001
A0000002472001
```

I added the ones you said but yea same problem.

shakhzodiosdev, 2022-08-04:
@RobinFarmer These identifiers I found from different sources) I am sure that some of them I do not need also) I just tried to add each identifier that I could find while searching.

shakhzodiosdev, 2022-08-04:
@RobinFarmer I am sorry earlier I had said the same problem with driving license. Actually I could not solve the issue with driving license and as I remember I had not problem with reading from ID card.

RobinFarmer, 2022-08-04:
@shakhzodiosdev Thanks,

I did find https://www.eftlab.com/knowledge-base/211-emv-aid-rid-pix/ which has lots of identifiers including:
[image]

I can't it to work yet but will continue to look into it.

AndyQ, 2023-02-03:
I'm closing this as whilst ID cards may work, its not something I can test as I don' have access to any.

If you are able to to get more information, please submit a PR or a new issue with more details

closed 2023-02-03 by AndyQ