AndyQ/NFCPassportReader #117: Working directly with masterlist in .ml file without transforming to .pem

arturocalvoacn, 2021-09-21:
The authority we use publishes the masterlist in .ml format.
I have seen that some [Android libraries](https://github.com/jllarraz/AndroidPassportReader/blob/beb8703b605347339100d83879db0b661b15b4f1/app/src/main/java/example/jllarraz/com/passportreader/network/MasterListService.kt#L27) accept .ml files for passive authentication.
Is it any way to do that as well from iOS without running the ML-to-PEM Python script?

AndyQ, 2022-06-16:
This is tricky - I'm actually using the OpenSSL CMS functions to do the certificate chain matching.  I'm not sure its worth me trying to implement chain building myself so I'm closing this but will give it a little thought as I agree it would be simpler and take some steps out.

Also one additional thing I'd need to consider is that there are multiple formats that countries publish in.

closed 2022-06-16 by AndyQ

shuvo1997, 2025-01-01:
@AndyQ Can you please share OpenSSL Script to do the certificate chain matching?