AndyQ/NFCPassportReader #119: PACE-IM implementation

pauphi, 2021-09-28:
Hi!

Does anyone have a working implementation of PACE-IM, or is there currently an implementation in the works?

I'm working on a project where we have access to several passport / identity card specimen for EU countries, including specimen that support PACE-IM-ECDH / PACE-IM-DH. Our Android application currently supports PACE-IM using JMRTD, however our iOS application lacks this support.

We have tried implementing this ourselves, but to no avail as of now. We tried basing it on [OpenPACE](https://github.com/frankmorgner/openpace/blob/master/src/pace_mappings.c) (and partly on [JMRTD](https://sourceforge.net/p/jmrtd/code/HEAD/tree/trunk/jmrtd/src/main/java/org/jmrtd/protocol/PACEProtocol.java)), as it seemed as if the current implementation of PACE-GM is at least somewhat based on OpenPACE. However OpenPACE does not seem to implement the `pseudoRandomFunction` specified in Doc9303-11 (it is implemented by JMRTD)? I'm sure we also have several other faults other than a broken `pseudoRandomFunction` implementation..

Eventually PACE fails with the current logs:

```
Error reading tag: sw1 - 0x63, sw2 - 0x00
PACEHandler: Step3 KeyAgreement - Error - InvalidMRZKey
   OpenSSLError: 
PACE Failed - Invalid data passed - PACE Failed

```

I'd be happy to share what we currently have to whoever is interested in helping out, and to help verify anyones PACE-IM implementation using the specimen we have access to.

danydev, 2021-10-06:
I'm curious, are you able to read data from those passports using BAC authentication or they dropped support for it?

pauphi, 2021-10-06:
I’m able to read data using BAC on those passports. Active authentication is not supported however (replaced by chip authentication)

yen-dang-ts, 2024-05-29:
I got same issue

AndyQ, 2026-11-09:
I've added an initial implementation of PACE-IM - would be interested to see how you get on with it

closed 2026-11-09 by AndyQ