AndyQ/NFCPassportReader #124: Chip Authentication not working

jennytellne-cygni, 2021-11-04:
Hi,

I was debugging some issues and noticed in the logs that Chip Authentication success was always false, but set to .success in passport.chipAuthenticationStatus.

So first of all:
ChipAuthenticationHandler.swift line 68 should return false since chip authentication failed:
`guard chipAuthPublicKeyInfos.count > 0 else {
            completedHandler?( true )
            return
        }` should be `guard chipAuthPublicKeyInfos.count > 0 else {
            completedHandler?( false )
            return
        }`

PassportReader.swift line 40 should set self.passport.chipAuthenticationStatus with regards to the success bool returned from the closure:
`self.caHandler?.doChipAuthentication() { [unowned self] (success) in
                                    self.passport.chipAuthenticationStatus = .success
                                    self.readNextDataGroup(completedReadingGroups: completed)
                                }`
should be 
`self.caHandler?.doChipAuthentication() { [unowned self] (success) in
                                    self.passport.chipAuthenticationStatus = success ? .success : .failed
                                    self.readNextDataGroup(completedReadingGroups: completed)
                                }`

So this will fix things visually in the app, but chip authentication still always fails for me. 
I have tested on multiple norwegian specimen: passport, special passport, and two types of ids. Also tested on real swedish passports. Chip authentication fails for all of these documents but should succeed.

It seems to fail in the function `sendPulbicKey(oid : String, keyId : Int?, pcdPublicKey : OpaquePointer, completed: @escaping (ResponseAPDU?, NFCPassportReaderError?)->())` with the following message: 

Error reading tag: sw1 - 0x6F, sw2 - 0x00
reason: No precise diagnosis
ERROR! - No precise diagnosis
Finished Chip Authentication - success - false

Any help is appreciated!

pauphi, 2021-11-04:
It seems as if reverting the change added in [8470bfa311a7f515ebfbe73261637f43bc183988](https://github.com/AndyQ/NFCPassportReader/commit/8470bfa311a7f515ebfbe73261637f43bc183988) will cause chip authentication to successfully complete, but active authentication will then fail again. Tested on Norwegian passports.

**Log with current version**
```
tagReaderSession:connected to tag - starting authentication
Error reading tag: sw1 - 0x6A, sw2 - 0x82
reason: File not found
Re-selecting eMRTD Application
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
BACHandler - Doing mutual authentication
BACHandler - complete
BAC Successful
Reading tag - COM
TagReader - Number of data bytes to read - 23
DG Found - ["DG1", "DG2", "DG3", "DG14", "DG15"]
Reading tag - DG14
TagReader - Number of data bytes to read - 362
Performing Chip Authentication - number of public keys found - 1
Starting Chip Authentication!
Error reading tag: sw1 - 0x6F, sw2 - 0x00
reason: No precise diagnosis
ERROR! - No precise diagnosis
Finished Chip Authentication - success - false
Reading tag - SOD
TagReader - Number of data bytes to read - 2026
Reading tag - DG1
TagReader - Number of data bytes to read - 91
Reading tag - DG2
TagReader - Number of data bytes to read - 17659
Reading tag - DG15
TagReader - Number of data bytes to read - 162
Performing Active Authentication
Active Authentication (RSA) successful
Passport passed SOD Verification
Parse SOD - Using Algo - SHA256
      - Hashes     - [NFCPassportReader.DataGroupId.DG1: "8960DE3A667C690922CE712724C898D6393A579F9867980ACEFEBF9890D135D6", NFCPassportReader.DataGroupId.DG2: "5D45B4F96CE576EA8450C1D537DB51E9F3EF58A033CEC83B0C3E7DED41FDC205", NFCPassportReader.DataGroupId.DG15: "A9524CFEB9AF4DB9EEB2EA01A8C8F4AB9D0D53C5B5DAEFEF913B5100B714C84C", NFCPassportReader.DataGroupId.DG3: "BC4162A963EC6E8DFA522288391CA05FEB37F859F93554F19450B589B6D2CD21", NFCPassportReader.DataGroupId.DG14: "57B84B921638A743333FEF6F49FB162D36598CFB568DEF0C2825E3101961629D"]
Passport passed Datagroup Tampering check
```

**Log with [commit 8470bfa311a7f515ebfbe73261637f43bc183988](https://github.com/AndyQ/NFCPassportReader/commit/8470bfa311a7f515ebfbe73261637f43bc183988) reverted:**
```
tagReaderSession:connected to tag - starting authentication
Error reading tag: sw1 - 0x6A, sw2 - 0x82
reason: File not found
Re-selecting eMRTD Application
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
BACHandler - Doing mutual authentication
BACHandler - complete
BAC Successful
Reading tag - COM
TagReader - Number of data bytes to read - 23
Error reading tag: sw1 - 0x6C, sw2 - 0x00
reason: Wrong length Le: SW2 indicates the exact length - (exact length :0)
ERROR - Wrong length Le: SW2 indicates the exact length - (exact length :0)
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
BACHandler - Doing mutual authentication
BACHandler - complete
BAC Successful
Reading tag - COM
TagReader - Number of data bytes to read - 23
DG Found - ["DG1", "DG2", "DG3", "DG14", "DG15"]
Reading tag - DG14
TagReader - Number of data bytes to read - 362
Performing Chip Authentication - number of public keys found - 1
Starting Chip Authentication!
Public Key successfully sent to passport!
Restarting secure messaging using DESede encryption
Finished Chip Authentication - success - true
Reading tag - SOD
TagReader - Number of data bytes to read - 2026
Reading tag - DG1
TagReader - Number of data bytes to read - 91
Reading tag - DG2
TagReader - Number of data bytes to read - 17659
Reading tag - DG15
TagReader - Number of data bytes to read - 162
Performing Active Authentication
Error reading tag: sw1 - 0x6C, sw2 - 0x80
reason: Wrong length Le: SW2 indicates the exact length - (exact length :128)
doInternalAuthentication failed - Wrong length Le: SW2 indicates the exact length - (exact length :128)
Passport passed SOD Verification
Parse SOD - Using Algo - SHA256
      - Hashes     - [NFCPassportReader.DataGroupId.DG15: "A9524CFEB9AF4DB9EEB2EA01A8C8F4AB9D0D53C5B5DAEFEF913B5100B714C84C", NFCPassportReader.DataGroupId.DG2: "5D45B4F96CE576EA8450C1D537DB51E9F3EF58A033CEC83B0C3E7DED41FDC205", NFCPassportReader.DataGroupId.DG3: "BC4162A963EC6E8DFA522288391CA05FEB37F859F93554F19450B589B6D2CD21", NFCPassportReader.DataGroupId.DG1: "8960DE3A667C690922CE712724C898D6393A579F9867980ACEFEBF9890D135D6", NFCPassportReader.DataGroupId.DG14: "57B84B921638A743333FEF6F49FB162D36598CFB568DEF0C2825E3101961629D"]
Passport passed Datagroup Tampering check
```

pauphi, 2021-11-10:
Changing to this in `SecureMessaging.swift` seems like it may work: 

```swift
let isMSE = apdu.instructionCode == 0x22
if apdu.expectedResponseLength > 0 && (isMSE ? apdu.expectedResponseLength < 256 : true) {
    tmp += " and DO97"
    do97 = try self.buildD097(apdu: apdu)
}
```

danydev, 2021-11-10:
@AndyQ this is quite interesting.

AndyQ, 2022-06-12:
Thanks! No idea how I missed this one!

AndyQ, 2022-06-12:
Fixed in 1.1.9

closed 2022-06-12 by AndyQ