AndyQ/NFCPassportReader #125: Active Authentication with ECDSA signature not working

pauphi, 2021-12-02:
Hi! 

I've noticed that Active Authentication with ECDSA signature does not work. Have tested on documents with both `ecdsa-plain-SHA224` and `ecdsa-plain-SHA384`. The logs are not very helpful, just displaying `Error verifying Active Authentication ECDSA signature`. Active Authentication with RSA signature works as expected.

If I'm not mistaken it's not a huge problem as most documents either support Chip Authentication as well, or use Active Authentication with RSA signature - however, I've seen specimen from Japan that **only** have Active Authentication w/ ECDSA.

pauphi, 2021-12-06:
Opened a pull request: #129

AndyQ, 2022-02-11:
Thanks I'll get this added for the next release!

AndyQ, 2022-06-12:
Merged and in 1.1.9 - thanks!

closed 2022-06-12 by AndyQ