AndyQ/NFCPassportReader #131: PACE Integrated mapping problem

enaifuose, 2022-02-02:
Hi,

first i wanna thank AndyQ for this greate work.

I try to read id card using PACE authentication method.  firstly i have trouble to get cardAccess data but finnaly i resolved by setting the APDU request in readCardAccess methode by :

`NFCISO7816APDU(instructionClass: 0x00, instructionCode: 0xA4, p1Parameter: 0x02, p2Parameter: 0x0C, data: Data([0x3f,0x00]), expectedResponseLength: 256)`

neverthless, the PACE handler return the error **PACEHandler: Step2IM - IM not yet implemented**. see the image for the app log :

[image]

Please can anyone help me to do the Integrated mapping implimentation.

thanks.

enaifuose, 2022-02-07:
Hi,

still in trouble with this problem, any one can help.

AndyQ, 2022-02-11:
My issue is that I don't have access to any passports in order to add the IM yet!  Its a little more complex than the GM.

enaifuose, 2022-02-14:
Thank you for your interest,

i do lot of research and finally i do some attempts to recreat the implimentation in JMRTD here : 

https://github.com/rgex/JMRTD/blob/e551cd0105e4aed10371f80d38a32baefbeaded5/jmrtd/src/main/java/org/jmrtd/protocol/PACEProtocol.java#L506

unfortunately, i do not have any experience with cryptokit or openssl.

So, if you can do the implimentation in your SDK by swift, i can run the tests for you.

i m available,

cordialy.

rbrouwer, 2022-02-23:
JMRTD's source code is located at the following location:
https://sourceforge.net/p/jmrtd/code/

So I would recommend using:
https://sourceforge.net/p/jmrtd/code/HEAD/tree/trunk/jmrtd/src/main/java/org/jmrtd/protocol/PACEProtocol.java
as code to look into.

AndyQ, 2022-02-23:
The problem is that you can't realistically implement it without being able to test as you go as each step depends on output from the previous, you can't capture anything repeatable as all the inputs change each time!

Now I could in theory take a guess at how it could work, push that out ask someone to test which would almost immediately fail and then go back-and-forth many times to figure out why (and then repeat), but sadly I don't have time/willingness to do that!

However, if anyone with a PACE-IM supported passport fancies implementing it - I'd be very very happy to receive a pull request!

longcon9999, 2024-04-23:
I always got error 6D 00 (Instruction code not supported or invalid) when select master file, so I can not using PACE. Please help me, @AndyQ . I also tried change cmd like @enaifuose but still did not work. Thank you

AndyQ, 2026-11-09:
Have added an initial implementation of PACE-IM - please test and raise any issues you find.