AndyQ/NFCPassportReader #132: Driving License issue

shakhzodiosdev, 2022-02-23:
Hi,

I am facing with the problem that I right now can read from NFC passport and e-passport but not driving license. Actually there is no zone of MRZ in driving license and I am trying to send manually the document number with checksum, birthdate with checksum and issue date with checksum. But it returns [NFCTagReaderSession _connectTag:error:]:635  Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x282ba0a80 {Error Domain=nfcd Code=15 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error}}} such an error. Any ideas about it?) I guess on driving license MRZ generation works a little bit different but it is just thoughts. Thanks.

rbrouwer, 2022-04-12:
NFC Passsports follow ICAO standards while European drivers license follow the ISO18013 standard.

The eDL MRZ is different. Opening the NFC chip would requires BAP instead of BAC identification. Another applications ID should be used. Also PACE likely requires work and several datagroups are different in that standard.

It would be really nice to have that support, but those type of documents are completely different and could very much be a NFCDriversLicense repo/project.

shakhzodiosdev, 2022-04-12:
@rbrouwer Thanks for response. It would be great if had that one also :)

AndyQ, 2022-06-16:
Sadly I don't have a compatible driving licence to even look into this.  If at such time I can get one I'll look at what's needed/

closed 2022-06-16 by AndyQ