AndyQ/NFCPassportReader #150: Chip Authentication 6F error code

emreyasar, 2022-07-07:
When I try to Chip Authentication, in the sendMSESetATIntAuth method, APDU command completes with error. The error log is below.

Performing Chip Authentication - number of public keys found - 1
Starting Chip Authentication!
Error reading tag: sw1 - 0x6F, sw2 - 0x00
reason: No precise diagnosis
ERROR! - No precise diagnosis
Finished Chip Authentication - success - false

What does "No precise diagnosis" mean ? Can you help me ?

Groops78, 2022-07-07:
Coule you send me a complete log of apdu?

emreyasar, 2022-07-07:
In order not to be too long, I deleted the data group read logs except for DG14. I can add them if you want.

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2836acd80>)
tagReaderSession:connecting to tag - iso7816(<NFCISO7816Tag: 0x2836acd80>)
tagReaderSession:connected to tag - starting authentication
TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00, 0x00]
TagReader - Received response
TagReader [unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]
TagReader - Received response
TagReader [unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x04]
TagReader - Received response
TagReader [unprotected] [0x31, 0x14, 0x30, 0x12, ], sw1:0x90 sw2:0x00
TagReader - Number of data bytes to read - 20
TagReader - sending [0x00, 0xB0, 0x00, 0x02, 0x14]
TagReader - Received response
TagReader [unprotected] [0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x04, 0x04, 0x04, 0x02, 0x01, 0x02, 0x02, 0x01, 0x12, ], sw1:0x90 sw2:0x00
Read CardAccess - data 31143012060A04007F00070202040404020102020112
Starting Password Authenticated Connection Establishment (PACE)
Performing PACE with id-PACE-ECDH-IM-AES-CBC-CMAC-256
TagReader - sending [0x00, 0x22, 0xC1, 0xA4, 0x0F, 0x80, 0x0A, 0x04, 0x00, 0x7F, 0x00, 0x07, 0x02, 0x02, 0x04, 0x04, 0x04, 0x83, 0x01, 0x01, 0x00]
TagReader - Received response
TagReader [unprotected] [], sw1:0x90 sw2:0x00
Doing PACE Step1...
TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x02, 0x7C, 0x00, 0x00]
TagReader - Received response
TagReader [unprotected] [0x7c, 0x22, 0x80, 0x20, 0x77, 0x91, 0x4f, 0xcc, 0x53, 0x23, 0x52, 0x14, 0xdd, 0xdd, 0x40, 0xa7, 0xb2, 0x64, 0xa9, 0x28, 0x59, 0x1c, 0xb1, 0xf6, 0xde, 0x51, 0xd6, 0x7b, 0xab, 0x34, 0xf4, 0x63, 0x22, 0x71, 0x8e, 0x56, ], sw1:0x90 sw2:0x00
Doing PACE Step2...
   Using Integrated Mapping (IM)...
PACEHandler: Step2IM - IM not yet implemented
   OpenSSLError: 
PACE Failed - Invalid data passed - PACE Failed
Re-selecting eMRTD Application
TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01, 0x00]
TagReader - Received response
TagReader [unprotected] [], sw1:0x90 sw2:0x00
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
TagReader - Received response
TagReader [unprotected] [0xf2, 0x52, 0xe7, 0x36, 0x8b, 0xc2, 0xce, 0x92, ], sw1:0x90 sw2:0x00
BACHandler - Doing mutual authentication
TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x10, 0x4D, 0x72, 0x42, 0xA0, 0x46, 0xAE, 0x4D, 0x9E, 0x16, 0xCF, 0x2E, 0x57, 0x03, 0xB5, 0x55, 0xA6, 0x4A, 0x1C, 0xBE, 0x82, 0x98, 0xE1, 0x04, 0xD8, 0x10, 0x76, 0xAD, 0x1E, 0x7D, 0xCE, 0x0F, 0x61, 0x6D, 0x69, 0x6E, 0xB1, 0x74, 0x25, 0x3D, 0x00]
TagReader - Received response
TagReader [unprotected] [0x5e, 0x4f, 0x7f, 0xe2, 0x30, 0xa3, 0x28, 0x9e, 0x04, 0xdb, 0xd7, 0x13, 0x67, 0x68, 0xa1, 0x3f, 0x59, 0xef, 0xe0, 0xce, 0x2a, 0xfb, 0xe9, 0xef, 0xef, 0x6f, 0x86, 0xe2, 0x12, 0xec, 0x59, 0x71, 0xf7, 0x0a, 0x51, 0x7e, 0x17, 0x92, 0x50, 0x24, ], sw1:0x90 sw2:0x00
BACHandler - complete
BAC Successful
Reading tag - COM
TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1E]
TagReader - Received response
TagReader [SM - unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x04]
TagReader - Received response
TagReader [SM - unprotected] [0x60, 0x18, 0x5f, 0x01, ], sw1:0x90 sw2:0x00
TagReader - Number of data bytes to read - 24
TagReader - sending [0x00, 0xB0, 0x00, 0x02, 0x18]
TagReader - Received response
TagReader [SM - unprotected] [0x5f, 0x01, 0x04, 0x30, 0x31, 0x30, 0x37, 0x5f, 0x36, 0x06, 0x30, 0x34, 0x30, 0x30, 0x30, 0x30, 0x5c, 0x06, 0x61, 0x75, 0x63, 0x6b, 0x6c, 0x6e, ], sw1:0x90 sw2:0x00
DG Found - ["DG1", "DG2", "DG3", "DG11", "DG12", "DG14"]
Reading tag - DG14
TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x0E]
TagReader - Received response
TagReader [SM - unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x04]
TagReader - Received response
TagReader [SM - unprotected] [0x6e, 0x82, 0x02, 0x91, ], sw1:0x90 sw2:0x00
TagReader - Number of data bytes to read - 657
[... 12 lines cut ...]
TagReader - sending [0x00, 0xB0, 0x02, 0x84, 0x11]
TagReader - Received response
TagReader [SM - unprotected] [0xf1, 0x98, 0x49, 0x5b, 0x63, 0xd2, 0x43, 0x2d, 0x50, 0x95, 0x41, 0xb3, 0x90, 0xe7, 0xf1, 0xf2, 0xab, ], sw1:0x90 sw2:0x00
Performing Chip Authentication - number of public keys found - 1
Starting Chip Authentication!
TagReader - sending [0x00, 0x22, 0x41, 0xA4, 0x0C, 0x80, 0x0A, 0x04, 0x00, 0x7F, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x04, 0x00]
TagReader - Received response
TagReader [SM - unprotected] [0x99, 0x02, 0x6f, 0x00, 0x8e, 0x08, 0x8d, 0x88, 0x1f, 0x9d, 0xd9, 0x70, 0x58, 0x9e, ], sw1:0x6f sw2:0x00
Error reading tag: sw1 - 0x6F, sw2 - 0x00
reason: No precise diagnosis
ERROR! - No precise diagnosis
Finished Chip Authentication - success - false
Reading tag - SOD
TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1D]
...
....
....
....
....
....
error 20 at 0 depth lookup:unable to get local issuer certificate
Parse SOD - Using Algo - SHA512
      - Hashes     -  ......................
Passport passed Datagroup Tampering check
tagReaderSession:didInvalidateWithError - Session invalidated by user

emreyasar, 2022-07-21:
Is there any comment about this error ?

Groops78, 2022-07-21:
I think the implicit selection is not well configured

emreyasar, 2022-07-25:
@AndyQ do you have any comment ?

Groops78, 2022-07-25:
Do you have the atr/ats answer?

AndyQ, 2022-07-25:
I'll take a look in a day or so when I'm back. It's possible it's not correctly selecting the correct certificate for some reason or I need to explicitly select the certificate rather than relying on the fact that as there is only one then passport will automatically just use that.

emreyasar, 2022-08-01:
It's happening because I haven't imported the latest changes in SecureMessaging to my local. Now it completed CA successfully. Thanks for your help.

closed 2022-08-01 by emreyasar