AndyQ/NFCPassportReader #151: Italian passports support

viktor-idenfy, 2022-07-08:
Hello, we at **iDenfy** use the code from this library for our NFC solution (I also messaged you @AndyQ in person, we at iDenfy would like to become an official backer of this library and help with documentation, supported document lists as well as drive more adoption). Right now we face issues with Italian passports. The client could not read NFC data (we do data extracting on the device and then send raw data to the **backend for further processing).** 

Right now we added more logs but would like to know, if perhaps you have **Italian documents or tested with them**, before asking the client to test and possibly fail again. Specifically, we are interested in [https://www.consilium.europa.eu/prado/en/ITA-AO-02003/image-135189.html](https://www.consilium.europa.eu/prado/en/ITA-AO-02003/image-135189.html), but other types would suit as well. Right now we're thinking about using [this solution](https://github.com/AndyQ/NFCPassportReader/issues/110#issuecomment-1177257368), have you tested it perhaps?

AndyQ, 2022-07-08:
I've tested a number of Italian passports and they worked fine - but this was about a year or so ago now.  Its possible that they have implemented PACE-IM which isn't supported (couldn't say), BUT the library should revert back to BAC in this situation.

You mentioned that you couldn't read the data - was the passport detected, and did it attempt to start reading?

Unfortunately without any logs I couldn't really say where to start looking.

wellbranding, 2022-07-12:
@danydev please send your email to v.vostrikov123@gmail.com. We at iDenfy gonna invite you to test our KYC and can actually pay you, because we can't find people like you... So frustrating.

wellbranding, 2022-07-14:
@danydev send you few emails. Have you received them? :) Perhaps it went to spam

danydev, 2023-01-25:
I replied and tested with them. The italian passport works flawlessy. This can be closed @AndyQ

closed 2023-01-25 by AndyQ