AndyQ/NFCPassportReader #156: Unable to read Emirates ID Card

zaidahmedpf, 2022-08-15:
After scanning MRZ I got this error.
NFCPassportReader.NFCPassportReaderError.ResponseError("Incorrect parameters P1-P2", 106, 134)

unitedadi, 2022-08-18:
2022-08-18 13:4:05.4340 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2022-08-18 13:4:05.4340 - reason: Conditions of use not satisfied
2022-08-18 13:4:05.4340 - PACE Failed - falling back to BAC
2022-08-18 13:4:05.4340 - Re-selecting eMRTD Application
2022-08-18 13:4:05.4440 - Starting Basic Access Control (BAC)
2022-08-18 13:4:05.4440 - BACHandler - deriving Document Basic Access Keys
2022-08-18 13:4:05.4440 - BACHandler - Getting initial challenge
2022-08-18 13:4:05.4670 - BACHandler - Doing mutual authentication
2022-08-18 13:4:05.4780 - Error reading tag: sw1 - 0x6A, sw2 - 0x86
2022-08-18 13:4:05.4790 - reason: Incorrect parameters P1-P2
2022-08-18 13:4:08.3480 - tagReaderSession:didInvalidateWithError - Session invalidated by user

This is the complete Log

danydev, 2023-01-25:
@zaidahmedpf do you mind provide the logs with full debug details?

NirajAkratech, 2023-09-13:
Hi @danydev 

I am Unable to read Emirates ID Card, Here is logs

```
2023-09-13 14:14:16.7650 - tagReaderSessionDidBecomeActive
2023-09-13 14:14:19.0490 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2824fc390>)
2023-09-13 14:14:19.0520 - tagReaderSession:connected to tag - starting authentication
2023-09-13 14:14:19.053707+0400 e-Passport[378:43752] [CoreNFC] 00000002 81831170 -[NFCTagReaderSession setAlertMessage:]:90  (null)
2023-09-13 14:14:19.0540 - TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
2023-09-13 14:14:19.1320 - TagReader - Received response
2023-09-13 14:14:19.1320 - TagReader [unprotected] [], sw1:0x90 sw2:0x00
2023-09-13 14:14:19.1330 - TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]
2023-09-13 14:14:19.1910 - TagReader - Received response
2023-09-13 14:14:19.1920 - TagReader [unprotected] [], sw1:0x6a sw2:0x82
2023-09-13 14:14:19.1920 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-09-13 14:14:19.1930 - reason: File not found
2023-09-13 14:14:19.1930 - PACE Failed - falling back to BAC
2023-09-13 14:14:19.1930 - Re-selecting eMRTD Application
2023-09-13 14:14:19.1940 - TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01]
2023-09-13 14:14:19.4580 - TagReader - Received response
2023-09-13 14:14:19.4590 - TagReader [unprotected] [], sw1:0x90 sw2:0x00
2023-09-13 14:14:19.4590 - Starting Basic Access Control (BAC)
2023-09-13 14:14:19.4590 - BACHandler - deriving Document Basic Access Keys
2023-09-13 14:14:19.4590 - Calculate the SHA-1 hash of MRZ_information
2023-09-13 14:14:19.4600 - 	MRZ KEY - 118127031889052482610117
2023-09-13 14:14:19.4620 - 	sha1(MRZ_information): 32715D02C2C83CC6872DD08F48488085A46CC96E
2023-09-13 14:14:19.4620 - Take the most significant 16 bytes to form the Kseed
2023-09-13 14:14:19.4630 - 	Kseed: 32715D02C2C83CC6872DD08F48488085
2023-09-13 14:14:19.4630 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2
2023-09-13 14:14:19.4640 - BACHandler - Getting initial challenge
2023-09-13 14:14:19.4640 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
2023-09-13 14:14:19.5150 - TagReader - Received response
2023-09-13 14:14:19.5160 - TagReader [unprotected] [0x68, 0x4c, 0x88, 0xd4, 0x7b, 0xb7, 0x84, 0xd3, ], sw1:0x90 sw2:0x00
2023-09-13 14:14:19.5170 - DATA - [104, 76, 136, 212, 123, 183, 132, 211]
2023-09-13 14:14:19.5170 - BACHandler - Doing mutual authentication
2023-09-13 14:14:19.5180 - Request an 8 byte random number from the MRTD's chip
2023-09-13 14:14:19.5180 - 	RND.ICC: 684C88D47BB784D3
2023-09-13 14:14:19.5180 - Generate an 8 byte random and a 16 byte random
2023-09-13 14:14:19.5190 - 	RND.IFD: 471E2CAAEE4004F5
2023-09-13 14:14:19.5190 - 	RND.Kifd: 090E534B29AD17334110CC09FBC5F290
2023-09-13 14:14:19.5190 - Concatenate RND.IFD, RND.ICC and Kifd
2023-09-13 14:14:19.5200 - 	S: 471E2CAAEE4004F5684C88D47BB784D3090E534B29AD17334110CC09FBC5F290
2023-09-13 14:14:19.5200 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2
2023-09-13 14:14:19.5210 - 	Eifd: FCF084949600A6D2428A73AD1F7621CA6D1FF2F9456CBC22E905595793611277
2023-09-13 14:14:19.5210 - Calc mac
2023-09-13 14:14:19.5210 - x0: FCF084949600A6D2
2023-09-13 14:14:19.5220 - y0: FB7BD855D2608B38
2023-09-13 14:14:19.5220 - x1: 428A73AD1F7621CA
2023-09-13 14:14:19.5230 - y1: 6E6D32D051DBFF6D
2023-09-13 14:14:19.5230 - x2: 6D1FF2F9456CBC22
2023-09-13 14:14:19.5240 - y2: 7FE68E8ABA5B91DC
2023-09-13 14:14:19.5240 - x3: E905595793611277
2023-09-13 14:14:19.5250 - y3: 1C5F83508934DC48
2023-09-13 14:14:19.5260 - x4: 8000000000000000
2023-09-13 14:14:19.5260 - y4: 0B321C5D4FB5DEF3
2023-09-13 14:14:19.5270 - y: 0B321C5D4FB5DEF3
2023-09-13 14:14:19.5270 - bkey: 78BDD6D40EC5DE92
2023-09-13 14:14:19.5270 - akey: 3A82E02BA91F161D
2023-09-13 14:14:19.5280 - b: BF249DB97A821115
2023-09-13 14:14:19.5290 - a: 20FF078225090983
2023-09-13 14:14:19.5290 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
2023-09-13 14:14:19.5290 - 	Mifd: 20FF078225090983
2023-09-13 14:14:19.5290 - Construct command data for MUTUAL AUTHENTICATE
2023-09-13 14:14:19.5300 - 	cmd_data: FCF084949600A6D2428A73AD1F7621CA6D1FF2F9456CBC22E90559579361127720FF078225090983
2023-09-13 14:14:19.5310 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0xFC, 0xF0, 0x84, 0x94, 0x96, 0x00, 0xA6, 0xD2, 0x42, 0x8A, 0x73, 0xAD, 0x1F, 0x76, 0x21, 0xCA, 0x6D, 0x1F, 0xF2, 0xF9, 0x45, 0x6C, 0xBC, 0x22, 0xE9, 0x05, 0x59, 0x57, 0x93, 0x61, 0x12, 0x77, 0x20, 0xFF, 0x07, 0x82, 0x25, 0x09, 0x09, 0x83, 0x00]
2023-09-13 14:14:19.6120 - TagReader - Received response
2023-09-13 14:14:19.6130 - TagReader [unprotected] [], sw1:0x6a sw2:0x88
2023-09-13 14:14:19.6130 - Error reading tag: sw1 - 0x6A, sw2 - 0x88
2023-09-13 14:14:19.6130 - reason: Referenced data not found
C
```

danydev, 2023-09-13:
Did you try the latest version of the library? @AndyQ solved several problems regarding this kind of errors

NirajAkratech, 2023-09-13:
Yes

@danydev, Thanks for the quick response!