AndyQ/NFCPassportReader #157: OpenSSL version issue

fowwest, 2022-08-22:
I am using NFCPassportReader via custom native module in my React Native app (v0.69.4) and RN framework involves OpenSSL-universal dependency v1.1.1100 while NFCPassportReader caps it down to v1.1.180 and therefore, it's causing a pod install error as there's a conflict in resolution of OpenSSL-universal pod.
Is it possible to update the podspec of NFCPassportReader to use OpenSSL-universal v1.1.1100?

AndyQ, 2022-08-27:
There should be no reason why you can't - I've not tested it against that yet, but I'll probably be updating it to use the latest OpenSSL version which I think is currently 1.1.1700

harouf, 2022-09-05:
Great, could you release a new version with OpenSSL dependency version upgraded to 1.1.1700?

AndyQ, 2022-12-18:
I've updated to Openssl 1.1.1900 in 2.0.1

closed 2022-12-18 by AndyQ