AndyQ/NFCPassportReader #163: Referenced data not found

gojogo88, 2022-10-19:
Hi... I'm trying to get the Example_spm project working.  I added a new masterlist.pem file in the NFCPassportReaderApp since the included one is already expired but everytime i try to scan a german passport (issued last year), i'm getting `Security status not satisfied` then it falls back to BAC then I get an error `referenced data not found`.  Any idea on what I'm doing wrong?

douglasg, 2022-10-19:
@gojogo88 This is solved on this commit (pull request)
https://github.com/AndyQ/NFCPassportReader/pull/153/commits

gojogo88, 2022-10-24:
So I made sure i see the changes in that commit but every time I  scan a German passport, I'm still getting
Error reading tag: sw1 - 0x6A, sw2 - 0x88
reason: Referenced data not found

danydev, 2022-11-28:
@AndyQ what do you think about the PR?

AndyQ, 2022-11-28:
It looks ok, but I have a feeling I may have started with -1 for length and had to change it to 256 as it caused an issue with some passports.   I'll go back throughout the history and check and if not then will merge in. Sadly may not be this week though as away with work.

AndyQ, 2022-12-18:
PR has been merged - sorry for delay!

closed 2022-12-18 by AndyQ