AndyQ/NFCPassportReader #164: Incorrect parameters P1-P2", 106, 134

febinfathah, 2022-11-08:
I am trying to scan the French ID card using the NFCPassportReader Demo application. But it was failing with an error "missing the required entitlement". After a bit of research, I found the application identifier for this card is "A0000001510000". 

Now, I can detect the tag, and while selecting the master file I am getting this error "Incorrect parameters P1-P2", 106, 134". But in our android app, we are using JMRTD to read the ID cards and can select the master file and select and read the EF.CardAccess file. I have compared both JMRTD and NFCPassportReader codes, and we have the same APDU commands and flow. 

Please help me with what I am missing and let me know if you need any further information. 

Thank you in advance.

TSkovsgaard, 2023-01-23:
Hey @febinfathah I end up with the same error 106, 134. Did you figure out what is wrong ? I suppose my real error is due to some authentication issue `sw1 - 0x69, sw2 - 0x85` indicates as I understand it an error in security

```
Error reading tag: sw1 - 0x69, sw2 - 0x85
reason: Conditions of use not satisfied
PACE Failed - falling back to BAC
Re-selecting eMRTD Application
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
BACHandler - Doing mutual authentication
Error reading tag: sw1 - 0x6A, sw2 - 0x86
reason: Incorrect parameters P1-P2
ResponseError("Incorrect parameters P1-P2", 106, 134)
```

danydev, 2023-02-04:
@TSkovsgaard  would you be able to test it with latest version? 

Also, can you post a log with debug enabled? See readme

TSkovsgaard, 2023-02-05:
Testet out the newest version, this is the ouput for verbose logging.

```
2023-02-05 10:34:51.9350 - tagReaderSessionDidBecomeActive
2023-02-05 10:34:52.7070 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x283c3cfc0>)
2023-02-05 10:34:52.7070 - tagReaderSession:connected to tag - starting authentication
2023-02-05 10:34:52.708625+0100 NFCReader[28116:7800131] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-02-05 10:34:52.7100 - TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
2023-02-05 10:34:52.7180 - TagReader - Received response
2023-02-05 10:34:52.7180 - TagReader [unprotected] [], sw1:0x69 sw2:0x85
2023-02-05 10:34:52.7190 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2023-02-05 10:34:52.7190 - reason: Conditions of use not satisfied
2023-02-05 10:34:52.7190 - PACE Failed - falling back to BAC
2023-02-05 10:34:52.7190 - Re-selecting eMRTD Application
2023-02-05 10:34:52.7190 - TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01]
2023-02-05 10:34:52.7350 - TagReader - Received response
2023-02-05 10:34:52.7350 - TagReader [unprotected] [], sw1:0x90 sw2:0x00
2023-02-05 10:34:52.7350 - Starting Basic Access Control (BAC)
2023-02-05 10:34:52.7350 - BACHandler - deriving Document Basic Access Keys
2023-02-05 10:34:52.7350 - Calculate the SHA-1 hash of MRZ_information
2023-02-05 10:34:52.7360 - 	MRZ KEY - XXXXXXXX<XXXXXXXXXXXXXXX
2023-02-05 10:34:52.7380 - 	sha1(MRZ_information): 2253C7489D1BFCDAC184BE179A6B57A4EF10D91B
2023-02-05 10:34:52.7390 - Take the most significant 16 bytes to form the Kseed
2023-02-05 10:34:52.7390 - 	Kseed: 2253C7489D1BFCDAC184BE179A6B57A4
2023-02-05 10:34:52.7390 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2
2023-02-05 10:34:52.7390 - BACHandler - Getting initial challenge
2023-02-05 10:34:52.7400 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
2023-02-05 10:34:52.7580 - TagReader - Received response
2023-02-05 10:34:52.7580 - TagReader [unprotected] [0x55, 0xfe, 0xab, 0xf1, 0xfd, 0x29, 0xe0, 0x6a, ], sw1:0x90 sw2:0x00
2023-02-05 10:34:52.7610 - DATA - [85, 254, 171, 241, 253, 41, 224, 106]
2023-02-05 10:34:52.7610 - BACHandler - Doing mutual authentication
2023-02-05 10:34:52.7610 - Request an 8 byte random number from the MRTD's chip
2023-02-05 10:34:52.7610 - 	RND.ICC: 55FEABF1FD29E06A
2023-02-05 10:34:52.7620 - Generate an 8 byte random and a 16 byte random
2023-02-05 10:34:52.7620 - 	RND.IFD: 58E6CF79D2BB0329
2023-02-05 10:34:52.7630 - 	RND.Kifd: 365A0E277E90C40BD14FD9DD00AB9CFF
2023-02-05 10:34:52.7630 - Concatenate RND.IFD, RND.ICC and Kifd
2023-02-05 10:34:52.7630 - 	S: 58E6CF79D2BB032955FEABF1FD29E06A365A0E277E90C40BD14FD9DD00AB9CFF
2023-02-05 10:34:52.7640 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2
2023-02-05 10:34:52.7640 - 	Eifd: 4930235258CD0EB4A3DE203D86B85C48BE7E62D5A7CDA6D7ACDF961CA899B140
2023-02-05 10:34:52.7650 - Calc mac
2023-02-05 10:34:52.7650 - x0: 4930235258CD0EB4
2023-02-05 10:34:52.7650 - y0: A90E14F0A6FAFDD8
2023-02-05 10:34:52.7650 - x1: A3DE203D86B85C48
2023-02-05 10:34:52.7650 - y1: DD5BE3A86DC7EFE7
2023-02-05 10:34:52.7660 - x2: BE7E62D5A7CDA6D7
2023-02-05 10:34:52.7660 - y2: 560F72561565747C
2023-02-05 10:34:52.7660 - x3: ACDF961CA899B140
2023-02-05 10:34:52.7660 - y3: 887707E0E2198FAD
2023-02-05 10:34:52.7660 - x4: 8000000000000000
2023-02-05 10:34:52.7660 - y4: 4A0411C141B2C644
2023-02-05 10:34:52.7670 - y: 4A0411C141B2C644
2023-02-05 10:34:52.7670 - bkey: 9D24C8A1C2720C30
2023-02-05 10:34:52.7670 - akey: FFFABE04400FFF31
2023-02-05 10:34:52.7670 - b: 309777B097151F2A
2023-02-05 10:34:52.7670 - a: 08088975951FD380
2023-02-05 10:34:52.7670 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
2023-02-05 10:34:52.7670 - 	Mifd: 08088975951FD380
2023-02-05 10:34:52.7670 - Construct command data for MUTUAL AUTHENTICATE
2023-02-05 10:34:52.7680 - 	cmd_data: 4930235258CD0EB4A3DE203D86B85C48BE7E62D5A7CDA6D7ACDF961CA899B14008088975951FD380
2023-02-05 10:34:52.7680 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x49, 0x30, 0x23, 0x52, 0x58, 0xCD, 0x0E, 0xB4, 0xA3, 0xDE, 0x20, 0x3D, 0x86, 0xB8, 0x5C, 0x48, 0xBE, 0x7E, 0x62, 0xD5, 0xA7, 0xCD, 0xA6, 0xD7, 0xAC, 0xDF, 0x96, 0x1C, 0xA8, 0x99, 0xB1, 0x40, 0x08, 0x08, 0x89, 0x75, 0x95, 0x1F, 0xD3, 0x80, 0x00]
2023-02-05 10:34:52.7790 - TagReader - Received response
2023-02-05 10:34:52.7790 - TagReader [unprotected] [], sw1:0x6a sw2:0x86
2023-02-05 10:34:52.7790 - Error reading tag: sw1 - 0x6A, sw2 - 0x86
2023-02-05 10:34:52.7790 - reason: Incorrect parameters P1-P2
ResponseError("Incorrect parameters P1-P2", 106, 134)
```

danydev, 2023-02-05:
ok coo, which country did issue the ID card?

TSkovsgaard, 2023-02-05:
Oman :)

danydev, 2023-02-05:
I see they issued 2 types of ID cards, one older (from 2006) and one newer (after 2016), are you dealing with the last type right?

TSkovsgaard, 2023-02-06:
Yes my test card is issued in 2019, it is expired though but that shouldn't be a problem ? I testet with an expired passport which worked flawless.

danydev, 2023-02-06:
I see, maybe @AndyQ could say something looking at your logs, let's wait for him.

AndyQ, 2023-02-06:
I can't see anything obvious, the only thing I can think of trying though is changing the expectedResponseLength from 256 to -1 (similar to a couple of other issues).  Not sure what affect (if any) this will have on existing code - would need to test that but would be interesting to see if this works for @TSkovsgaard.

So if you change the method TagReader:doMutualAuthentication( cmdData : Data )....
And change the expectedResponseLength to -1, and let me know if that gets any further.

TSkovsgaard, 2023-02-06:
Same error after changing to -1 
output:

```
2023-02-06 11:8:40.9910 - tagReaderSessionDidBecomeActive
2023-02-06 11:8:42.0480 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2822d8d20>)
2023-02-06 11:8:42.0480 - tagReaderSession:connected to tag - starting authentication
2023-02-06 11:08:42.049005+0100 NFCReader[29542:8451017] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-02-06 11:8:42.0510 - TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
2023-02-06 11:8:42.0600 - TagReader - Received response
2023-02-06 11:8:42.0610 - TagReader [unprotected] [], sw1:0x69 sw2:0x85
2023-02-06 11:8:42.0610 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2023-02-06 11:8:42.0610 - reason: Conditions of use not satisfied
2023-02-06 11:8:42.0610 - PACE Failed - falling back to BAC
2023-02-06 11:8:42.0610 - Re-selecting eMRTD Application
2023-02-06 11:8:42.0610 - TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01]
2023-02-06 11:8:42.0770 - TagReader - Received response
2023-02-06 11:8:42.0770 - TagReader [unprotected] [], sw1:0x90 sw2:0x00
2023-02-06 11:8:42.0770 - Starting Basic Access Control (BAC)
2023-02-06 11:8:42.0770 - BACHandler - deriving Document Basic Access Keys
2023-02-06 11:8:42.0770 - Calculate the SHA-1 hash of MRZ_information
2023-02-06 11:8:42.0770 - 	MRZ KEY - XXXXXXXX<XXXXXXXXXXXXXXX
2023-02-06 11:8:42.0800 - 	sha1(MRZ_information): 2253C7489D1BFCDAC184BE179A6B57A4EF10D91B
2023-02-06 11:8:42.0800 - Take the most significant 16 bytes to form the Kseed
2023-02-06 11:8:42.0810 - 	Kseed: 2253C7489D1BFCDAC184BE179A6B57A4
2023-02-06 11:8:42.0810 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2
2023-02-06 11:8:42.0810 - BACHandler - Getting initial challenge
2023-02-06 11:8:42.0810 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
2023-02-06 11:8:42.1020 - TagReader - Received response
2023-02-06 11:8:42.1020 - TagReader [unprotected] [0x01, 0xa9, 0x4f, 0xb1, 0x5e, 0xfa, 0xc2, 0x3c, ], sw1:0x90 sw2:0x00
2023-02-06 11:8:42.1040 - DATA - [1, 169, 79, 177, 94, 250, 194, 60]
2023-02-06 11:8:42.1040 - BACHandler - Doing mutual authentication
2023-02-06 11:8:42.1040 - Request an 8 byte random number from the MRTD's chip
2023-02-06 11:8:42.1040 - 	RND.ICC: 01A94FB15EFAC23C
2023-02-06 11:8:42.1050 - Generate an 8 byte random and a 16 byte random
2023-02-06 11:8:42.1050 - 	RND.IFD: 435CB698ABEA8DC3
2023-02-06 11:8:42.1050 - 	RND.Kifd: D5F67B507458D0F7382C00C4D76FB9BC
2023-02-06 11:8:42.1050 - Concatenate RND.IFD, RND.ICC and Kifd
2023-02-06 11:8:42.1060 - 	S: 435CB698ABEA8DC301A94FB15EFAC23CD5F67B507458D0F7382C00C4D76FB9BC
2023-02-06 11:8:42.1060 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2
2023-02-06 11:8:42.1070 - 	Eifd: 8987B9F46EE335F33E0622D8826462A14BB755D652302BE2D9109D334FAFD9F8
2023-02-06 11:8:42.1070 - Calc mac
2023-02-06 11:8:42.1070 - x0: 8987B9F46EE335F3
2023-02-06 11:8:42.1080 - y0: FB63F3140E515CB5
2023-02-06 11:8:42.1080 - x1: 3E0622D8826462A1
2023-02-06 11:8:42.1080 - y1: 329B592BDBA6E91D
2023-02-06 11:8:42.1080 - x2: 4BB755D652302BE2
2023-02-06 11:8:42.1080 - y2: 69126EB9369FE11A
2023-02-06 11:8:42.1090 - x3: D9109D334FAFD9F8
2023-02-06 11:8:42.1090 - y3: 25EE02C8440AC2E5
2023-02-06 11:8:42.1090 - x4: 8000000000000000
2023-02-06 11:8:42.1090 - y4: FDB0653EB3D71A29
2023-02-06 11:8:42.1090 - y: FDB0653EB3D71A29
2023-02-06 11:8:42.1090 - bkey: 9D24C8A1C2720C30
2023-02-06 11:8:42.1100 - akey: FFFABE04400FFF31
2023-02-06 11:8:42.1100 - b: 45E78DF4C00E19C2
2023-02-06 11:8:42.1100 - a: FD87638CC14078EA
2023-02-06 11:8:42.1100 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
2023-02-06 11:8:42.1100 - 	Mifd: FD87638CC14078EA
2023-02-06 11:8:42.1100 - Construct command data for MUTUAL AUTHENTICATE
2023-02-06 11:8:42.1100 - 	cmd_data: 8987B9F46EE335F33E0622D8826462A14BB755D652302BE2D9109D334FAFD9F8FD87638CC14078EA
2023-02-06 11:8:42.1110 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x89, 0x87, 0xB9, 0xF4, 0x6E, 0xE3, 0x35, 0xF3, 0x3E, 0x06, 0x22, 0xD8, 0x82, 0x64, 0x62, 0xA1, 0x4B, 0xB7, 0x55, 0xD6, 0x52, 0x30, 0x2B, 0xE2, 0xD9, 0x10, 0x9D, 0x33, 0x4F, 0xAF, 0xD9, 0xF8, 0xFD, 0x87, 0x63, 0x8C, 0xC1, 0x40, 0x78, 0xEA]
2023-02-06 11:8:42.1210 - TagReader - Received response
2023-02-06 11:8:42.1220 - TagReader [unprotected] [], sw1:0x6a sw2:0x86
2023-02-06 11:8:42.1220 - Error reading tag: sw1 - 0x6A, sw2 - 0x86
2023-02-06 11:8:42.1220 - reason: Incorrect parameters P1-P2
ResponseError("Incorrect parameters P1-P2", 106, 134)
```

Could it be something with the AID ? have the following AID's in my plist
```
00000000000000
A0000002471001
A0000002472001
```

AndyQ, 2023-02-06:
Don't think so, that's really down to detecting the nfc chip.

One other thing to try - does the ReadID app read the id card OK?

TSkovsgaard, 2023-02-06:
No the ReadID app also returns an error `Authentication failed`, and Regula Document Reader returns
`LAYER6: PWD Suspended 2` Found the string here https://docs.regulaforensics.com/develop/doc-reader-sdk/mobile/files/RegulaSDK.strings.txt I found the key in [source](https://github.com/regulaforensics/DocumentReader-Android/blob/abc2d0e309375de4046e6094717275dc487d8894/docs/com/regula/documentreader/api/enums/eRFID_ErrorCodes.html) with the value `Error of General Authenticate APDU-command execution when performing PACE.`

AndyQ, 2023-02-06:
It looking like it either doesn't support BAC properly or it has its own version.  Sadly, I don't think there is much I can do here esp as I don' have access to id cards!

renevdkooi, 2023-03-21:
It's happening on the latest version of Dutch passports as well. Exact same code for an older passport (eq. 2019) works, but a passport handed out in 2023 is not working. Same error code as above.
I've got a brand new masterList.pem from today too.

2023-03-21 15:1:57.3120 - Starting Basic Access Control (BAC)
2023-03-21 15:1:57.3120 - BACHandler - deriving Document Basic Access Keys
2023-03-21 15:1:57.3120 - BACHandler - Getting initial challenge
2023-03-21 15:1:57.3300 - BACHandler - Doing mutual authentication
2023-03-21 15:1:57.3460 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2023-03-21 15:1:57.3460 - reason: Conditions of use not satisfied

danydev, 2023-03-21:
@renevdkooi can you also try it with ReadID?

danydev, 2023-03-21:
can you post logs with [logging](https://github.com/AndyQ/NFCPassportReader#logging) enabled?

renevdkooi, 2023-03-21:
2023-03-21 15:10:02.8550 - tagReaderSessionDidBecomeActive
2023-03-21 15:10:04.3490 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2837ea3a0>)
2023-03-21 15:10:04.3500 - tagReaderSession:connected to tag - starting authentication
2023-03-21 15:10:04.351270+0700 [14354:4205489] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-03-21 15:10:04.3520 - Starting Basic Access Control (BAC)
2023-03-21 15:10:04.3520 - BACHandler - deriving Document Basic Access Keys
2023-03-21 15:10:04.3520 - Calculate the SHA-1 hash of MRZ_information
2023-03-21 15:10:04.3520 - 	MRZ KEY - NT82JRBC7407080852802080
2023-03-21 15:10:04.3550 - 	sha1(MRZ_information): D1AE4F690C9F49E1648BF8D0F79988DFCC6C5B5D
2023-03-21 15:10:04.3550 - Take the most significant 16 bytes to form the Kseed
2023-03-21 15:10:04.3550 - 	Kseed: D1AE4F690C9F49E1648BF8D0F79988DF
2023-03-21 15:10:04.3550 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2
2023-03-21 15:10:04.3560 - BACHandler - Getting initial challenge
2023-03-21 15:10:04.3560 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
2023-03-21 15:10:04.3750 - TagReader - Received response
2023-03-21 15:10:04.3760 - TagReader [unprotected] [0xae, 0x6e, 0x18, 0x29, 0x4a, 0x44, 0xf3, 0x1a, ], sw1:0x90 sw2:0x00
2023-03-21 15:10:04.3780 - DATA - [174, 110, 24, 41, 74, 68, 243, 26]
2023-03-21 15:10:04.3780 - BACHandler - Doing mutual authentication
2023-03-21 15:10:04.3780 - Request an 8 byte random number from the MRTD's chip
2023-03-21 15:10:04.3780 - 	RND.ICC: AE6E18294A44F31A
2023-03-21 15:10:04.3780 - Generate an 8 byte random and a 16 byte random
2023-03-21 15:10:04.3780 - 	RND.IFD: 84C40FC9C6B87287
2023-03-21 15:10:04.3790 - 	RND.Kifd: BDD263EEEEDA45314C963131622E224B
2023-03-21 15:10:04.3790 - Concatenate RND.IFD, RND.ICC and Kifd
2023-03-21 15:10:04.3790 - 	S: 84C40FC9C6B87287AE6E18294A44F31ABDD263EEEEDA45314C963131622E224B
2023-03-21 15:10:04.3790 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2
2023-03-21 15:10:04.3800 - 	Eifd: D0D0B2FA10260FF9D5F3CEF433B52BC84CB363D4126DBF3582B0B0BADAC1F606
2023-03-21 15:10:04.3800 - Calc mac
2023-03-21 15:10:04.3800 - x0: D0D0B2FA10260FF9
2023-03-21 15:10:04.3800 - y0: B77FC28A4D9F85F2
2023-03-21 15:10:04.3800 - x1: D5F3CEF433B52BC8
2023-03-21 15:10:04.3800 - y1: 68ABD9CC05F3F626
2023-03-21 15:10:04.3800 - x2: 4CB363D4126DBF35
2023-03-21 15:10:04.3810 - y2: E74C4D955C808A0E
2023-03-21 15:10:04.3810 - x3: 82B0B0BADAC1F606
2023-03-21 15:10:04.3810 - y3: AC3C55E6BB8F7D53
2023-03-21 15:10:04.3810 - x4: 8000000000000000
2023-03-21 15:10:04.3810 - y4: C1DC1C0190920C9B
2023-03-21 15:10:04.3810 - y: C1DC1C0190920C9B
2023-03-21 15:10:04.3810 - bkey: 4DA5DCF041E016F4
2023-03-21 15:10:04.3810 - akey: 8A815881EB3DE7BD
2023-03-21 15:10:04.3810 - b: 45ACFFA75B2C36AA
2023-03-21 15:10:04.3810 - a: 6DB37AA764E0BB29
2023-03-21 15:10:04.3810 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
2023-03-21 15:10:04.3810 - 	Mifd: 6DB37AA764E0BB29
2023-03-21 15:10:04.3810 - Construct command data for MUTUAL AUTHENTICATE
2023-03-21 15:10:04.3820 - 	cmd_data: D0D0B2FA10260FF9D5F3CEF433B52BC84CB363D4126DBF3582B0B0BADAC1F6066DB37AA764E0BB29
2023-03-21 15:10:04.3820 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0xD0, 0xD0, 0xB2, 0xFA, 0x10, 0x26, 0x0F, 0xF9, 0xD5, 0xF3, 0xCE, 0xF4, 0x33, 0xB5, 0x2B, 0xC8, 0x4C, 0xB3, 0x63, 0xD4, 0x12, 0x6D, 0xBF, 0x35, 0x82, 0xB0, 0xB0, 0xBA, 0xDA, 0xC1, 0xF6, 0x06, 0x6D, 0xB3, 0x7A, 0xA7, 0x64, 0xE0, 0xBB, 0x29, 0x00]
2023-03-21 15:10:04.3980 - TagReader - Received response
2023-03-21 15:10:04.3980 - TagReader [unprotected] [], sw1:0x69 sw2:0x85
2023-03-21 15:10:04.3980 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2023-03-21 15:10:04.3980 - reason: Conditions of use not satisfied
2023-03-21 15:10:07.2080 - tagReaderSession:didInvalidateWithError - Session invalidated by user

renevdkooi, 2023-03-21:
I think I'm being thick.. i put "skipPACE" on true. :(

danydev, 2023-03-21:
That's good news!

NirajAkratech, 2023-08-09:
Hi @TSkovsgaard 

I have the same issue with OMAN resident card, Do you find any solutions?

Thanks

danydev, 2023-08-09:
> Hi @TSkovsgaard
> 
> I have the same issue with OMAN resident card, Do you find any solutions?
> 
> Thanks

Can you post logs with [logging](https://github.com/AndyQ/NFCPassportReader#logging) set to debug?

NirajAkratech, 2023-08-09:
Hi @danydev ,
Thanks for the response 
Here is logs

2023-08-09 18:10:44.9990 - tagReaderSessionDidBecomeActive
2023-08-09 18:10:46.3420 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2804f9aa0>)
2023-08-09 18:10:46.3430 - tagReaderSession:connected to tag - starting authentication
2023-08-09 18:10:46.343863+0530 e-Passport[65893:3406517] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-08-09 18:10:46.3450 - TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
2023-08-09 18:10:46.3530 - TagReader - Received response
2023-08-09 18:10:46.3540 - TagReader [unprotected] [], sw1:0x69 sw2:0x85
2023-08-09 18:10:46.3540 - Error reading tag: sw1 - 0x69, sw2 - 0x85
2023-08-09 18:10:46.3540 - reason: Conditions of use not satisfied
2023-08-09 18:10:46.3540 - PACE Failed - falling back to BAC
2023-08-09 18:10:46.3540 - Re-selecting eMRTD Application
2023-08-09 18:10:46.3550 - TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01]
2023-08-09 18:10:46.3700 - TagReader - Received response
2023-08-09 18:10:46.3700 - TagReader [unprotected] [], sw1:0x90 sw2:0x00
2023-08-09 18:10:46.3710 - Starting Basic Access Control (BAC)
2023-08-09 18:10:46.3710 - BACHandler - deriving Document Basic Access Keys
2023-08-09 18:10:46.3710 - Calculate the SHA-1 hash of MRZ_information
2023-08-09 18:10:46.3710 - 	MRZ KEY - 113414631084011082001207
2023-08-09 18:10:46.3740 - 	sha1(MRZ_information): 87808A8999D1C15356A7069264416BB423B2B43C
2023-08-09 18:10:46.3740 - Take the most significant 16 bytes to form the Kseed
2023-08-09 18:10:46.3750 - 	Kseed: 87808A8999D1C15356A7069264416BB4
2023-08-09 18:10:46.3750 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2
2023-08-09 18:10:46.3750 - BACHandler - Getting initial challenge
2023-08-09 18:10:46.3750 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]
2023-08-09 18:10:46.3930 - TagReader - Received response
2023-08-09 18:10:46.3930 - TagReader [unprotected] [0xb8, 0x1a, 0xf0, 0x25, 0x50, 0x11, 0xed, 0xa4, ], sw1:0x90 sw2:0x00
2023-08-09 18:10:46.3970 - DATA - [184, 26, 240, 37, 80, 17, 237, 164]
2023-08-09 18:10:46.3970 - BACHandler - Doing mutual authentication
2023-08-09 18:10:46.3970 - Request an 8 byte random number from the MRTD's chip
2023-08-09 18:10:46.3970 - 	RND.ICC: B81AF0255011EDA4
2023-08-09 18:10:46.3980 - Generate an 8 byte random and a 16 byte random
2023-08-09 18:10:46.3980 - 	RND.IFD: 3673A351E5036F24
2023-08-09 18:10:46.3980 - 	RND.Kifd: 49440A3BC897596767A7AB56D76CA26A
2023-08-09 18:10:46.3990 - Concatenate RND.IFD, RND.ICC and Kifd
2023-08-09 18:10:46.3990 - 	S: 3673A351E5036F24B81AF0255011EDA449440A3BC897596767A7AB56D76CA26A
2023-08-09 18:10:46.4000 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2
2023-08-09 18:10:46.4000 - 	Eifd: 5358714B44C3CE8ABFB698FE3D8AD2C7B2CEC5BE786503AC26BA6259F5F593F1
2023-08-09 18:10:46.4000 - Calc mac
2023-08-09 18:10:46.4000 - x0: 5358714B44C3CE8A
2023-08-09 18:10:46.4010 - y0: 7D354539EECDD2FF
2023-08-09 18:10:46.4010 - x1: BFB698FE3D8AD2C7
2023-08-09 18:10:46.4010 - y1: 0374B03A1004C5AA
2023-08-09 18:10:46.4010 - x2: B2CEC5BE786503AC
2023-08-09 18:10:46.4020 - y2: A08A0EF6621FA7D0
2023-08-09 18:10:46.4020 - x3: 26BA6259F5F593F1
2023-08-09 18:10:46.4020 - y3: 3FA4B422DFE87AE5
2023-08-09 18:10:46.4020 - x4: 8000000000000000
2023-08-09 18:10:46.4020 - y4: D91CF63791600F4D
2023-08-09 18:10:46.4030 - y: D91CF63791600F4D
2023-08-09 18:10:46.4030 - bkey: CEB1D0F42582DAEB
2023-08-09 18:10:46.4030 - akey: BA7C2128541FF197
2023-08-09 18:10:46.4030 - b: DF380261D01A6B64
2023-08-09 18:10:46.4030 - a: 681B6FFF921FEB53
2023-08-09 18:10:46.4030 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
2023-08-09 18:10:46.4040 - 	Mifd: 681B6FFF921FEB53
2023-08-09 18:10:46.4040 - Construct command data for MUTUAL AUTHENTICATE
2023-08-09 18:10:46.4040 - 	cmd_data: 5358714B44C3CE8ABFB698FE3D8AD2C7B2CEC5BE786503AC26BA6259F5F593F1681B6FFF921FEB53
2023-08-09 18:10:49.4050 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x53, 0x58, 0x71, 0x4B, 0x44, 0xC3, 0xCE, 0x8A, 0xBF, 0xB6, 0x98, 0xFE, 0x3D, 0x8A, 0xD2, 0xC7, 0xB2, 0xCE, 0xC5, 0xBE, 0x78, 0x65, 0x03, 0xAC, 0x26, 0xBA, 0x62, 0x59, 0xF5, 0xF5, 0x93, 0xF1, 0x68, 0x1B, 0x6F, 0xFF, 0x92, 0x1F, 0xEB, 0x53, 0x00]
2023-08-09 18:10:49.4170 - TagReader - Received response
2023-08-09 18:10:49.4190 - TagReader [unprotected] [], sw1:0x6a sw2:0x86
2023-08-09 18:10:49.4190 - Error reading tag: sw1 - 0x6A, sw2 - 0x86
2023-08-09 18:10:49.4200 - reason: Incorrect parameters P1-P2

danydev, 2023-08-09:
Would you mind using the latest code from main branch? (so no 2.0.2 currently published, but really what's on `main` branch)
Does [ReadID Me](https://apps.apple.com/us/app/readid-me/id1463949991) works with that card?
Sorry for those questions, but it may help a little bit to gather info

NirajAkratech, 2023-08-09:
@danydev I am using this pod: ` pod 'NFCPassportReader', git:'https://github.com/AndyQ/NFCPassportReader.git'  `

And It's also not working [ReadID Me](https://apps.apple.com/us/app/readid-me/id1463949991) app as well

NirajAkratech, 2023-08-09:
@danydev Just for info, I testing with an expired Oman Resident card, but that shouldn't be a problem ?

danydev, 2023-08-09:
no it should work, that should not be a factor

NirajAkratech, 2023-08-09:
@danydev Just for Info I checked with android app which is use the `net.sf.scuba:scuba` libs, Same card working with android app

NirajAkratech, 2023-08-25:
Hi @AndyQ And @danydev ,
Now I am able to read data from Oman Resident card, Just I change the readCardAccess APDU,

 `let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 0x00, instructionCode: 0xA4, p1Parameter: 0x00, p2Parameter: 0x0C, data: Data([0xA0,0x00,0x00,0x00,0x18,0x52,0x4F,0x50,0x01,0x01]), expectedResponseLength: -1)`
 
And also read the DataGroups DG1, DG2, DG4, DG6, DG10, DG11, DG13 But I am not able to read DataGroups DG5, DG7, DG8, DG9, DG12, DG14 and DG16
 
Getting error:
2023-08-25 13:45:39.1460 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2023-08-25 13:45:39.1460 - reason: Security status not satisfied
2023-08-25 13:45:39.1460 - TagError reading tag - ResponseError("Security status not satisfied", 105, 130)
2023-08-25 13:45:39.1460 - ERROR - Security status not satisfied

Here is full logs for COM and DG5:

2023-08-25 15:39:05.580593+0530 DemoIDV[50141:2580417] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
SelectFile cmd: [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]
2023-08-25 15:39:05.6110 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-08-25 15:39:05.6130 - reason: File not found
2023-08-25 15:39:05.6130 - PACE Failed - falling back to BAC
2023-08-25 15:39:05.6350 - Starting Basic Access Control (BAC)
2023-08-25 15:39:05.7870 - Basic Access Control (BAC) - SUCCESS!
2023-08-25 15:39:05.791699+0530 DemoIDV[50141:2580139] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-08-25 15:39:05.7920 - Reading tag - COM
2023-08-25 15:39:05.793460+0530 DemoIDV[50141:2580139] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
SelectFile cmd: [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x02, 0x1E]
2023-08-25 15:39:05.8030 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-08-25 15:39:05.8030 - reason: File not found
2023-08-25 15:39:05.8090 - TagError reading tag - ResponseError("File not found", 106, 130)
2023-08-25 15:39:05.8100 - ERROR - File not found
2023-08-25 15:39:05.8100 - Starting Basic Access Control (BAC)
2023-08-25 15:39:05.9070 - Basic Access Control (BAC) - SUCCESS!
SelectFile cmd: [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x02, 0x1E]
2023-08-25 15:39:05.9170 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-08-25 15:39:05.9180 - reason: File not found
2023-08-25 15:39:05.9180 - TagError reading tag - ResponseError("File not found", 106, 130)
2023-08-25 15:39:05.9180 - ERROR - File not found
2023-08-25 15:39:05.918963+0530 DemoIDV[50141:2580139] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2023-08-25 15:39:05.9190 - Reading tag - DG5
2023-08-25 15:39:05.920454+0530 DemoIDV[50141:2580139] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
SelectFile cmd: [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x02, 0x05]
2023-08-25 15:39:05.9610 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2023-08-25 15:39:05.9610 - reason: Security status not satisfied
2023-08-25 15:39:05.9610 - TagError reading tag - ResponseError("Security status not satisfied", 105, 130)
2023-08-25 15:39:05.9610 - ERROR - Security status not satisfied
2023-08-25 15:39:05.9620 - Starting Basic Access Control (BAC)
2023-08-25 15:39:06.0640 - Basic Access Control (BAC) - SUCCESS!
SelectFile cmd: [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x02, 0x05]
2023-08-25 15:39:06.0980 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2023-08-25 15:39:06.0990 - reason: Security status not satisfied
2023-08-25 15:39:06.0990 - TagError reading tag - ResponseError("Security status not satisfied", 105, 130)
2023-08-25 15:39:06.0990 - ERROR - Security status not satisfied

vguerci, 2024-05-30:
Hello, we were facing the same issue, the explanation and fix are on iOS APIs, where PACE is sadly poorly documented.

Long story short, on iOS >= 16, to read NFC from _some_ PACE documents, you **must** use [`NFCTagReaderSession.PollingOption.pace`](https://developer.apple.com/documentation/corenfc/nfctagreadersession/pollingoption/3950980-pace), or you will get this `Incorrect parameters P1-P2` error. i.e. in `PassportReader.readPassport()`, something like:
```swift
let pollingOption: NFCTagReaderSession.PollingOption = if #available(iOS 16, *) { .pace } else { .iso14443 }
readerSession = NFCTagReaderSession(pollingOption: [pollingOption], delegate: self, queue: nil)
```

⚠️ With this polling option, passports are no longer detected, from Apple documentation (_"This is an exclusive value that cannot be combine with other NFCPollingOption values; this will override all other combinations."_), we can't combine both. This means you must know if your doc is capable of PACE before initiating the session.

⚠️ To be able to use this polling option, you will also need to add PACE to the `Near Field Communication Tag Reader Session Formats` key of your app entitlements, or iOS will fail the session with a "Missing entitlement" error:
```xml
<key>com.apple.developer.nfc.readersession.formats</key>
<array>
    <string>PACE</string>
    <string>TAG</string>
</array>
```

ℹ️  For french IDs, no need to add `A0000001510000` to iso 7816 identifiers with PACE polling.

danydev, 2024-05-30:
This is quite interesting, @AndyQ do you think we can incorporate somehow in the API?

I mean, it looks like we could at very least offer an configuration to enable this polling, that would  enable something like
Try with iso14443 first. At this point if it works, good, otherwise I can prompt the user to scan again, but this time I'll enable "PACE with polling" from code.

I just described my use case where I want to scan the maximum "kind" of documents, but I think that some flexibility here in the API would be appreciated

vguerci, 2024-05-30:
💯 Agree, maybe turning that `skipPACE` into an `usePACE` would make sense?

It is unclear to what extent this pace polling is needed, as _some_ PACE docs can be read w/o it. But OTOH, some docs, like 🇫🇷 ID cards it **must** be used...

AndyQ, 2024-05-31:
I've created a test branch - aa_test which makes the following changes:
- passportReader.readPassport now takes in an optional PassportReaderOptions struct with various configuration options
- Added option usePACEPolling - Switches to PACE Polling for detecting passports rather that iso14443
- Active Authentication no longer always uses extended read
- Instead, we first try with standard 256 byte read, and if fails, tries extended read
    - Note this is currently untested as I don't have any passports that require extended read
- Sample app updated to show new options

If anyone can test this branch to see if it works fine that would be great!

NezarAli, 2025-02-02:
Hey @NirajAkratech,
Can you please share more details on how you managed to read data from Omani IDs?
Changing the APDU was not enough