AndyQ/NFCPassportReader #165: Access to DataGroup3

dangary7, 2022-12-21:
I'm trying to access to the DataGroup3 to be able to see the fingerprint data stored there, I get an error of Expected SM data objects mossing (0x105, 0x135) is there any idea of how could I fix this?

Thanhdv2007, 2022-12-21:
same question

dangary7, 2022-12-21:
When debugging and trying to access to DataGroup3, I'm getting the following logs

2022-12-21 12:39:10.6120 - Reading tag - DG3
2022-12-21 12:39:10.612859-0600 NFCSelfie[32977:3375618] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
2022-12-21 12:39:10.6670 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2022-12-21 12:39:10.6680 - reason: Security status not satisfied
2022-12-21 12:39:10.6680 - TagError reading tag - ResponseError("Security status not satisfied", 105, 130)
2022-12-21 12:39:10.6680 - ERROR - Security status not satisfied
2022-12-21 12:39:10.6680 - Starting Basic Access Control (BAC)
2022-12-21 12:39:10.6690 - BACHandler - deriving Document Basic Access Keys
2022-12-21 12:39:10.6690 - BACHandler - Getting initial challenge
2022-12-21 12:39:10.6860 - Error reading tag: sw1 - 0x69, sw2 - 0x88
2022-12-21 12:39:10.6870 - reason: SM data objects incorrect
2022-12-21 12:39:13.4730 - tagReaderSession:didInvalidateWithError - Session invalidated by user

Any idea?

filom, 2022-12-21:
DG3 and DG4 contains biometric info and they require a higher level of security, only government entities terminals can provide it.

AndyQ, 2023-02-03:
I'm closing this as it's unlikely that I'll be able to support DG3 any time soon.