AndyQ/NFCPassportReader #166: Error timeout in Iphone 14 - iOS 16.2 and cannot detect in

Thanhdv2007, 2022-12-22:
Hi,
I'm testing on Iphone 14, iOS 16.2, the DataGroup reading is very slow, sometimes it can read it, sometimes it doesn't, most of the time it can't read the data in the chip, there is an error like the log below.
Log
`2022-12-22 17:26:40.3050 - Using version 1.1.4.1
2022-12-22 17:26:40.3890 - tagReaderSessionDidBecomeActive
2022-12-22 17:26:40.4640 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2814fedf0>)
2022-12-22 17:26:40.4640 - tagReaderSession:connected to tag - starting authentication
2022-12-22 17:26:40.4960 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2022-12-22 17:26:40.4960 - reason: Security status not satisfied
2022-12-22 17:26:40.4960 - PACE Failed - falling back to BAC
2022-12-22 17:26:40.4960 - Re-selecting eMRTD Application
2022-12-22 17:26:40.5120 - Starting Basic Access Control (BAC)
2022-12-22 17:26:40.5120 - BACHandler - deriving Document Basic Access Keys
2022-12-22 17:26:40.5120 - BACHandler - Getting initial challenge
2022-12-22 17:26:40.5250 - BACHandler - Doing mutual authentication
2022-12-22 17:26:40.5860 - DATA - [219, 144, 82, 25, 164, 157, 141, 187, 240, 190, 32, 102, 223, 23, 82, 187, 207, 86, 19, 71, 209, 216, 113, 233, 240, 130, 148, 253, 222, 166, 108, 132, 122, 249, 234, 124, 33, 242, 171, 43]
2022-12-22 17:26:40.5870 - BACHandler - complete
2022-12-22 17:26:40.5870 - Basic Access Control (BAC) - SUCCESS!
2022-12-22 17:26:40.5870 - Reading tag - COM
2022-12-22 17:26:40.6530 - TagReader - Number of data bytes to read - 24
2022-12-22 17:26:43.1500 - DG Found - ["DG1", "DG2", "DG3", "DG13", "DG15", "DG14"]
2022-12-22 17:26:43.1500 - Reading tag - DG14
2022-12-22 17:26:45.6950 - TagReader - Number of data bytes to read - 382
2022-12-22 17:26:45.8230 - Performing Chip Authentication - number of public keys found - 1
2022-12-22 17:26:46.0970 - Public Key successfully sent to passport!
2022-12-22 17:26:46.0980 - Restarting secure messaging using AES encryption
2022-12-22 17:26:46.0980 - Reading tag - SOD
2022-12-22 17:26:48.6270 - TagReader - Number of data bytes to read - 1765
2022-12-22 17:26:56.0960 - tagReaderSession:failed to connect to tag - Tag response error / no response
2022-12-22 17:26:58.8840 - tagReaderSession:didInvalidateWithError - Session invalidated by user`

Video

https://user-images.githubusercontent.com/89065538/209114421-9da02671-b768-46ad-b53c-5d4e7356ece6.MP4

Thanks

Toanpt15, 2022-12-25:
I have the same error. @Thanhdv2007 Did you fix it?

Toanpt15, 2022-12-25:
@AndyQ Pls help us for fix the error. It's very important for us.

AndyQ, 2022-12-27:
What is the make of the passport, and when was it issued?  There were some issues with some countries and older passports who didn't quite follow the specs correctly, or used slow chips.

Also, with some passports, the range of the chip is very limited or the power it can transmit is very weak meaning the phone reader has to be perfectly positioned (again mainly older passports though).

I'm only currently able to test with a variety of UK passports (some old - 5-10 years) and the newer ones, plus I had a sample passport (about 6 years old) but I've not had any issues reading them.

Toanpt15, 2022-12-28:
@AndyQ I use Vietnam ID Card. Your SDK works for my ID Card with iPhone 13. But it's fail with IP 14. Did you test UK Passport with IP 14?

Toanpt15, 2022-12-28:
@AndyQ I test my ID Cards by using iPhone 14 and ReadID Me app. It works well. So your SDK can read the ID Card as ReadID Me app?

danydev, 2023-01-25:
@Toanpt15 at the very least you should provide a full log with the higher level of debug
```
let reader = PassportReader(logLevel: .debug)
```

cyberneid, 2023-04-21:
Same issue here with iphone 12 Pro with iOS 16.3. The most of the time the reading fails with "no reponse"
It seems as there is a reading timeout somewhere that invalidate the session.

jjynus, 2023-08-22:
@AndyQ  
TagReader line 259
let (data, sw1, sw2) = try await tag.sendCommand(apdu: toSend)
예외처리가 필요해.. 일좀 해

vguerci, 2024-11-04:
Could this be related to, if not duplicating, #183 ?

And again a Pro model.

AndyQ, 2024-11-04:
Could you try turning off WiFi and seeing if that resolves the timeout issues? There have been a couple of comments that something has changed in the there that that may be interferring with the NFC.