AndyQ/NFCPassportReader #167: Reading Turkish eID problem in release version

aytacyaydem, 2022-12-25:
I have a React-Native application and I implemented a 3rd party eID NFC Reader SDK. This 3rd SDK is using NFCPassportReader library and CoreNFC package also. I'm trying to read Turkish Citizenship eID with this module. My problem this module is working properly in debug mode but It's crushing in release version. I've added a crush log related to this issue and I've added tagReaderSession implementation code. I've added my "com.apple.developer.nfc.readersession.iso7816.select-identifiers" configuration code also. Does anybody have an idea about that problem?

```
Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes: 0x0000000000000001, 0x0000000000000000
VM Region Info: 0 is not in any region.  Bytes before following region: 4340301824
      REGION TYPE                 START - END      [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                   102b3c000-10603c000 [ 53.0M] r-x/r-x SM=COW  ...ePieMobileDev
Termination Reason: SIGNAL 11 Segmentation fault: 11
Terminating Process: exc handler [9236]
```

```
Triggered by Thread:  8
Thread 8 name:   Dispatch queue: com.apple.corenfc.readersession.delegate
Thread 8 Crashed:
```
```
public func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {
        Log.debug( "tagReaderSession:didDetect - \(tags[0])" )
        if tags.count > 1 {
            Log.debug( "tagReaderSession:more than 1 tag detected! - \(tags)" )

            let errorMessage = NFCViewDisplayMessage.error(.MoreThanOneTagFound)
            self.invalidateSession(errorMessage: errorMessage, error: NFCPassportReaderError.MoreThanOneTagFound)
            return
        }

        let tag = tags.first!
        var passportTag: NFCISO7816Tag
        switch tags.first! {
        case let .iso7816(tag):
            passportTag = tag
        default:
            Log.debug( "tagReaderSession:invalid tag detected!!!" )

            let errorMessage = NFCViewDisplayMessage.error(NFCPassportReaderError.TagNotValid)
            self.invalidateSession(errorMessage:errorMessage, error: NFCPassportReaderError.TagNotValid)
            return
        }
                
        // Connect to tag
        Log.debug( "tagReaderSession:connecting to tag - \(tag)" )
        session.connect(to: tag) { [unowned self] (error: Error?) in
            if error != nil {
                Log.debug( "tagReaderSession:failed to connect to tag - \(error?.localizedDescription ?? "Unknown error")" )
                let errorMessage = NFCViewDisplayMessage.error(NFCPassportReaderError.ConnectionError)
                self.invalidateSession(errorMessage: errorMessage, error: NFCPassportReaderError.ConnectionError)
                return
            }
            
            Log.debug( "tagReaderSession:connected to tag - starting authentication" )
            self.updateReaderSessionMessage( alertMessage: NFCViewDisplayMessage.authenticatingWithPassport(0) )

            self.tagReader = TagReader(tag:passportTag)
            
            if let newAmount = self.dataAmountToReadOverride {
                self.tagReader?.overrideDataAmountToRead(newAmount: newAmount)
            }
            
            self.tagReader!.progress = { [unowned self] (progress) in
                if let dgId = self.currentlyReadingDataGroup {
                    self.updateReaderSessionMessage( alertMessage: NFCViewDisplayMessage.readingDataGroupProgress(dgId, progress) )
                } else {
                    self.updateReaderSessionMessage( alertMessage: NFCViewDisplayMessage.authenticatingWithPassport(progress) )
                }
            }

            DispatchQueue.global().async {
                self.startReading( )
            }
        }
    }
```
```
<key>NFCReaderUsageDescription</key>
<string>Read the NFC chip of ePassports</string>
<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
<array>
    <string>A0000002471001</string>
</array>
```
I can run the application in debug mode without any problems, but when I start reading nfc in release mode, the application crashes.

AndyQ, 2023-02-03:
Are you able to provide a stack trace showing where its crashing and also a debug log (set log level to debug)?

brkerturk, 2023-02-16:
@aytacyaydem I do not have any problems with my ios-native application reading Turkish e-ids. Inspecting the stack trace would be nice.

aytacyaydem, 2023-02-16:
This bug was related to useFlipper and OpenSSL. There is openssl in useFlipper on debug mode but not in release mode that was the cause of the error.
We added the following configurations for the release mode to fix it.
[image]
[image]

closed 2023-02-16 by aytacyaydem