AndyQ/NFCPassportReader #168: Crashed app while reading NFC using this library

boipqiod, 2022-12-27:
Hello

Our app is crashed while reading NFC using this library, 1.1.9 ver.
We found out the issue in firebase Crashlytics.

This is log.

```
Crashed: com.apple.corenfc.readersession.delegate
0  NFCPassportReader              0x575a0 $s17NFCPassportReader15SecureMessagingC9unprotect5rapduAA12ResponseAPDUVAG_tKF + 4420
1  NFCPassportReader              0x63058 $s17NFCPassportReader03TagB0C4send3cmd9completedySo14NFCISO7816APDUC_yAA08ResponseH0VSg_AA0aB5ErrorOSgtctFy10Foundation4DataV_s5UInt8VASs0J0_pSgtcfU_ + 988
2  NFCPassportReader              0xe040 $s10Foundation4DataVs5UInt8VAEs5Error_pSgIeggyyg_So6NSDataCA2ESo7NSErrorCSgIeyByyyy_TR + 124
3  CoreNFC                        0x119e4 __38-[NFCTag _sendAPDU:completionHandler:]_block_invoke + 464
4  libdispatch.dylib              0x1e68 _dispatch_call_block_and_release + 32
5  libdispatch.dylib              0x3a2c _dispatch_client_callout + 20
6  libdispatch.dylib              0xb124 _dispatch_lane_serial_drain + 668
7  libdispatch.dylib              0xbcb4 _dispatch_lane_invoke + 444
8  libdispatch.dylib              0x16500 _dispatch_workloop_worker_thread + 648
9  libsystem_pthread.dylib        0x10bc _pthread_wqthread + 288
10 libsystem_pthread.dylib        0xe5c start_wqthread + 8
```

However, we are unable to investigate the issue because we can’t cause the same issue to occur in our environment.

 Is there any way to solve it?
Thank you.

AndyQ, 2023-02-03:
Sorry, I'd need more information for this.  I'm closing this - if you can get more information please raise a new issue with the debug logs.

closed 2023-02-03 by AndyQ

boipqiod, 2023-02-06:
@AndyQ 

thank for answer