AndyQ/NFCPassportReader #169: Unable to read German passport - mutual authenticate issue

kirillivonin, 2023-01-11:
Hi,

I've recently updated to the newest version of NFCPassportReader and started having issues with German passports. Basically, I'm having the same issue as here: [Reference data not found](https://github.com/AndyQ/NFCPassportReader/issues/163), so it appears the issue wasn't solved :(

Here's the log of the event from start to error:
`2023-01-11 16:22:33.4820 - tagReaderSessionDidBecomeActive

2023-01-11 16:22:33.6260 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x282b22eb0>)

2023-01-11 16:22:33.6270 - tagReaderSession:connected to tag - starting authentication

2023-01-11 16:22:33.627622+0100 TestApp[7324:797609] [CoreNFC] 00000002 817b1170 -[NFCTagReaderSession setAlertMessage:]:90  (null)

2023-01-11 16:22:33.6280 - Starting Basic Access Control (BAC)

2023-01-11 16:22:33.6280 - BACHandler - deriving Document Basic Access Keys

2023-01-11 16:22:33.6280 - Calculate the SHA-1 hash of MRZ_information

2023-01-11 16:22:33.6280 - 	MRZ KEY - C15NRGH4X481031693204245

2023-01-11 16:22:33.6300 - 	sha1(MRZ_information): 04B8B3B52D8210189D2F0E9FF1BE9CBA7193C1A0

2023-01-11 16:22:33.6300 - Take the most significant 16 bytes to form the Kseed

2023-01-11 16:22:33.6300 - 	Kseed: 04B8B3B52D8210189D2F0E9FF1BE9CBA

2023-01-11 16:22:33.6300 - Calculate the Basic Access Keys (Kenc and Kmac) using TR-SAC 1.01, 4.2

2023-01-11 16:22:33.6310 - BACHandler - Getting initial challenge

2023-01-11 16:22:33.6310 - TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]

2023-01-11 16:22:33.6790 - TagReader - Received response

2023-01-11 16:22:33.6790 - TagReader [unprotected] [0x0d, 0x8e, 0x79, 0x87, 0xd7, 0x19, 0xc5, 0x40, ], sw1:0x90 sw2:0x00

2023-01-11 16:22:33.6800 - DATA - [13, 142, 121, 135, 215, 25, 197, 64]

2023-01-11 16:22:33.6800 - BACHandler - Doing mutual authentication

2023-01-11 16:22:33.6800 - Request an 8 byte random number from the MRTD's chip

2023-01-11 16:22:33.6800 - 	RND.ICC: 0D8E7987D719C540

2023-01-11 16:22:33.6800 - Generate an 8 byte random and a 16 byte random

2023-01-11 16:22:33.6800 - 	RND.IFD: DB36560D477D3564

2023-01-11 16:22:33.6800 - 	RND.Kifd: AFD1D04E48B6CF693FBBBEDE5D1A30F8

2023-01-11 16:22:33.6800 - Concatenate RND.IFD, RND.ICC and Kifd

2023-01-11 16:22:33.6810 - 	S: DB36560D477D35640D8E7987D719C540AFD1D04E48B6CF693FBBBEDE5D1A30F8

2023-01-11 16:22:33.6810 - Encrypt S with TDES key Kenc as calculated in Appendix 5.2

2023-01-11 16:22:33.6820 - 	Eifd: 75D4D946BA168A9D461E40943F2EC26F820620324D3D9492534F54DC880694BC

2023-01-11 16:22:33.6820 - Calc mac

2023-01-11 16:22:33.6820 - x0: 75D4D946BA168A9D

2023-01-11 16:22:33.6820 - y0: 16B7947C7F42FA86

2023-01-11 16:22:33.6820 - x1: 461E40943F2EC26F

2023-01-11 16:22:33.6820 - y1: 02F65983ACF65715

2023-01-11 16:22:33.6820 - x2: 820620324D3D9492

2023-01-11 16:22:33.6830 - y2: 72F11D3375CFDD59

2023-01-11 16:22:33.6830 - x3: 534F54DC880694BC

2023-01-11 16:22:33.6830 - y3: 9794054327BC9B49

2023-01-11 16:22:33.6830 - x4: 8000000000000000

2023-01-11 16:22:33.6830 - y4: 7DB41E0D07AED8C8

2023-01-11 16:22:33.6840 - y: 7DB41E0D07AED8C8

2023-01-11 16:22:33.6840 - bkey: DE6C7FA82991F89C

2023-01-11 16:22:33.6840 - akey: 99ED5FAEEEE601D5

2023-01-11 16:22:33.6840 - b: 836776C237C70800

2023-01-11 16:22:33.6840 - a: 6DDE4E3848A2FDE7

2023-01-11 16:22:33.6850 - Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2

2023-01-11 16:22:33.6850 - 	Mifd: 6DDE4E3848A2FDE7

2023-01-11 16:22:33.6850 - Construct command data for MUTUAL AUTHENTICATE

2023-01-11 16:22:33.6850 - 	cmd_data: 75D4D946BA168A9D461E40943F2EC26F820620324D3D9492534F54DC880694BC6DDE4E3848A2FDE7

2023-01-11 16:22:33.6860 - TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x75, 0xD4, 0xD9, 0x46, 0xBA, 0x16, 0x8A, 0x9D, 0x46, 0x1E, 0x40, 0x94, 0x3F, 0x2E, 0xC2, 0x6F, 0x82, 0x06, 0x20, 0x32, 0x4D, 0x3D, 0x94, 0x92, 0x53, 0x4F, 0x54, 0xDC, 0x88, 0x06, 0x94, 0xBC, 0x6D, 0xDE, 0x4E, 0x38, 0x48, 0xA2, 0xFD, 0xE7, 0x00]

2023-01-11 16:22:33.7840 - TagReader - Received response

2023-01-11 16:22:33.7850 - TagReader [unprotected] [], sw1:0x6a sw2:0x88

2023-01-11 16:22:33.7850 - Error reading tag: sw1 - 0x6A, sw2 - 0x88

2023-01-11 16:22:33.7860 - reason: Referenced data not found`

danydev, 2023-01-25:
@kirillivonin did you try reverting the fix? did it work for you?
Since it looks like some passports fails with the former and some with the latter, we may try doing first a try with 256 and then fallback to -1.

AndyQ, 2023-01-25:
So it sounds like it's older passports that have issues with a -1 length. Could you let me know the expiry date of any passports that fail with the -1?  

Will have a think how best to approach this. Not sure yet if I can assume that if the issue date is < xxx then default to 256 otherwise use -1.

It will be a bit tricky for me to test as all my available passports work fine with both -1 and 256!  Also, I'm not yet sure if when this fails, I'd need to totally restart the connection (right from initial passport detection) as with some passports if they get an error then they just won't process anything else on that session.

Will have a think.

Thormeard, 2023-01-26:
Hi,
I'm also have the same problem with a German passport. I have tried both 1.1.9 version (the one im using im my app) and your example app (latest version, 2.0.1).
Would be happy to debug/help if needed
Passport will expire in 2032 (I think it was issued in 2022)

Here is .debug logs:
```tagReaderSessionDidBecomeActive
tagReaderSession: didDetect - iso7816(<NFCIS07816Tag: 0x282a296b0>)
tagReaderSession: connecting to tag tagReaderSession: connected to tag -iso7816(<NFCIS07816Tag: 0x282a296b0>)
tagReaderSession: connected to tag - starting authentication
[CoreNFC] - [NFCTagReaderSession setAlertMessage:]:101 (null)
Error reading tag: sw1 - 0x67, sw2 - 0x00 reason: Wrong length
Re-selecting eMRTD Application
Error reading tag: sw1 - 0x67, sw2 - 0x00 reason: Wrong length
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
BACHandler - Doing mutual authentication
Error reading tag: sw1 - 0x6A, Sw2 - 0x88
reason: Referenced data not found
ERROR - Referenced data not found
BAC Failed
Error: Optional (NFCPassportReader.NCPassportReaderError.ResponseError("Referenceddatanotfound", 106, 136))
```

danydev, 2023-02-01:
So for German passport it doesn't work either with 1.1.9 (where we had -1) and with 2.0.1 (where we have 256). That's different from what @fuggly reported.

@fuggly can you tell us who issued your documents along with date? Thanks

AndyQ, 2023-02-01:
@danydev @Thormeard, so for the 1st try (1.1.9), that is failing BAC - its not even attempting PACE as the first step fails, but not quite sure why (not enough info).  Can you increase the logging level to debug?

The 2nd try (2.0.1), PACE succeeds, but it looks like it fails to read the COM data group - file not found! (which is where the supported Datagroups are held).  I thought that COM was a mandatory data group.

Then, it retries BAC but I suspect at that point the passport has decided enough is enough and drops its keys and connection resulting in that failing then.  Again, not enough info though.

@Thormeard Can you try the ReadID app and see if that can read it OK? and if so, what Datagroups does it find?

Thormeard, 2023-02-03:
I also found this closed merge request https://github.com/AndyQ/NFCPassportReader/pull/153
And this is the same fix I did to make German passport work.
There was 3 changes to -1, but in the last commit there was only 2 https://github.com/AndyQ/NFCPassportReader/commit/43892bbd3efd46daae97fe844d1b28baeb8e8208
I think we just missing the 3rd one to make German passports work

danydev, 2023-02-03:
that's cool! Maybe there is a reason, let's wait for @AndyQ

AndyQ, 2023-02-03:
@Thormeard Good spot!  I must have missed that.  I'll fix that and release a new version shortly!

AndyQ, 2023-02-03:
OK v2.0.2 is up - with that fix in.  I've tested it against all the passports I have and all worked fine (selectApplication, PACE, etc).  @fuggly @Thormeard, are you able to test this and confirm this fixes the issue?

Thanks very much!

fuggly, 2023-02-03:
@AndyQ I have to tell you next week (currently not working on this), but I will let you know with he passports I have to use

Thormeard, 2023-02-03:
@AndyQ Just tested with the sample app, works fine (for all 3 passports I have)! Thank you!

AndyQ, 2023-02-03:
I'll close this now - please reopen if it there is still an issue

closed 2023-02-03 by AndyQ

emirbeytekin, 2023-09-27:
Hello again,
i read all comments but i have an error scan to German passport. I can't change my min target (ios13.1) and error is continue to me. It's my TagReader.swift file, what can i need to change ?

`
//
//  TagHandler.swift
//  NFCTest
//
//  Created by Andy Qua on 09/06/2019.
//  Copyright © 2019 Andy Qua. All rights reserved.
//

import Foundation

#if !os(macOS)
import CoreNFC

@available(iOS 13, *)
public class TagReader {
    var tag : NFCISO7816Tag
    var secureMessaging : SecureMessaging?
    var maxDataLengthToRead : Int = 0xA0  // Should be able to use 256 to read arbitrary amounts of data at full speed BUT this isn't supported across all passports so for reliability just use the smaller amount.

    var progress : ((Int)->())?

    init( tag: NFCISO7816Tag ) {
        self.tag = tag
    }
    
    func overrideDataAmountToRead( newAmount : Int ) {
        maxDataLengthToRead = newAmount
    }
    
    func reduceDataReadingAmount() {
        if maxDataLengthToRead > 0xA0 {
            maxDataLengthToRead = 0xA0
        }
    }

    func readDataGroup( dataGroup: DataGroupId, completed: @escaping ([UInt8]?, NFCPassportReaderError?)->() )  {
        guard let tag = dataGroup.getFileIDTag() else {
            completed(nil, NFCPassportReaderError.UnsupportedDataGroup)
            return
        }
        
        selectFileAndRead(tag: tag, completed:completed )
    }
    
    func getChallenge( completed: @escaping (ResponseAPDU?, NFCPassportReaderError?)->() ) {
        let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x84, p1Parameter: 0, p2Parameter: 0, data: Data(), expectedResponseLength: 8)
        
        send( cmd: cmd, completed: completed )
    }
    
    func doInternalAuthentication( challenge: [UInt8], completed: @escaping (ResponseAPDU?, NFCPassportReaderError?)->() ) {
        let randNonce = Data(challenge)
        
        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 256)

        send( cmd: cmd, completed: completed )
    }

    func doMutualAuthentication( cmdData : Data, completed: @escaping (ResponseAPDU?, NFCPassportReaderError?)->() ) {
        let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x82, p1Parameter: 0, p2Parameter: 0, data: cmdData, expectedResponseLength: 256)

        send( cmd: cmd, completed: completed )
    }
    
    /// The MSE KAT APDU, see EAC 1.11 spec, Section B.1.
    /// This command is sent in the "DESede" case.
[... 340 lines cut ...]
            0x6D: [0x00:"Instruction code not supported or invalid"],
            0x6E: [0x00:"Class not supported"],
            0x6F: [0x00:"No precise diagnosis"],
            0x90: [0x00:"Success"] //No further qualification
        ]
        
        // Special cases - where sw2 isn't an error but contains a value
        if sw1 == 0x61 {
            return "SW2 indicates the number of response bytes still available - (\(sw2) bytes still available)"
        } else if sw1 == 0x64 {
            return "State of non-volatile memory unchanged (SW2=00, other values are RFU)"
        } else if sw1 == 0x6C {
            return "Wrong length Le: SW2 indicates the exact length - (exact length :\(sw2))"
        }

        if let dict = errors[sw1], let errorMsg = dict[sw2] {
            return errorMsg
        }
        
        return "Unknown error - sw1: 0x\(binToHexRep(sw1)), sw2 - 0x\(binToHexRep(sw2)) "
    }
}

#endif
`