AndyQ/NFCPassportReader #170: DO99 Fatal error: Array index is out of range

andycg0327, 2023-01-12:
Excuse me, the same problem still happen.
[](https://github.com/AndyQ/NFCPassportReader/issues/86)
But I do not know how to re-open issue, so I add a new one.

It been a while since the last check, unfortunately, today we tried in iPhone (XR, SE, 12, 13), the same issue still happen and just only in iPhone12...

Can you please help for this problem again? Thank you so much.

danydev, 2023-01-25:
Try with the latest version, and provide a full log with debug enabled. It may differ from what you already posted, since that was from a previous version

andycg0327, 2023-01-31:
The following is Log using iPhone 12 with NFCPassportReader 2.0.1 (Pods project which update to the newest version).
Thank you so much.

2023-01-31 9:49:55.5000 - Using version 1.1.4.1
2023-01-31 9:49:55.6090 - tagReaderSessionDidBecomeActive
2023-01-31 9:49:55.6760 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x283eed560>)
2023-01-31 9:49:55.6760 - tagReaderSession:connected to tag - starting authentication
2023-01-31 9:49:55.6930 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-01-31 9:49:55.6930 - reason: File not found
2023-01-31 9:49:55.6940 - PACE Failed - falling back to BAC
2023-01-31 9:49:55.6940 - Re-selecting eMRTD Application
2023-01-31 9:49:55.7030 - Starting Basic Access Control (BAC)
2023-01-31 9:49:55.7030 - BACHandler - deriving Document Basic Access Keys
2023-01-31 9:49:55.7040 - BACHandler - Getting initial challenge
2023-01-31 9:49:55.7130 - BACHandler - Doing mutual authentication
2023-01-31 9:49:55.8490 - DATA - [81, 16, 238, 251, 66, 250, 232, 189, 88, 150, 255, 153, 91, 179, 58, 31, 48, 226, 87, 131, 146, 12, 247, 169, 101, 239, 28, 75, 88, 4, 8, 171, 141, 96, 53, 108, 160, 93, 13, 93]
2023-01-31 9:49:55.8500 - BACHandler - complete
2023-01-31 9:49:55.8500 - Basic Access Control (BAC) - SUCCESS!
2023-01-31 9:49:55.8540 - Reading tag - COM
2023-01-31 9:49:55.8860 - TagReader - Number of data bytes to read - 23
2023-01-31 9:49:55.9040 - DG Found - ["DG1", "DG2", "DG11", "DG12", "DG13"]
2023-01-31 9:49:55.9060 - Reading tag - SOD
2023-01-31 9:49:55.9460 - TagReader - Number of data bytes to read - 2099
Swift/Array.swift:405: Fatal error: Array index is out of range
(lldb)

AndyQ, 2023-02-03:
Could you please provide a full stack trace?  and ideally a full debug log (see logging to debug)?

boipqiod, 2023-02-06:
I hope this message finds you well. I noticed that we are encountering a similar issue with the iPhone 14pro, where it occurs occasionally.
We have conducted tests with a Korean passport and it appears that this issue is persistent for those who use our service and is not related to the passport itself.

Our investigation has revealed that this issue was present in version 1.1.9 and still occurs after the recent update of the module.

Unfortunately, we do not have a complete log available, however, I am kindly attaching some logs and the relevant code for your convenience and further analysis.

`2023-02-01 16:8:30.4590 - Restarting secure messaging using DESede encryption
2023-02-01 16:8:30.4600 - Reading tag - SOD
2023-02-01 16:8:30.5050 - TagReader - Number of data bytes to read - 1847
2023-02-01 16:8:30.8920 - Reading tag - DG1
2023-02-01 16:8:30.9240 - TagReader - Number of data bytes to read - 91
2023-02-01 16:8:30.9500 - Reading tag - DG2
2023-02-01 16:8:31.0020 - TagReader - Number of data bytes to read - 11668
Swift/Array.swift:405: Fatal error: Array index is out of range
Printing description of rapduBin:
▿ 2 elements
  - 0 : 144
  - 1 : 0`
  
  `        //DO'99'
        // Mandatory, only absent if SM error occurs
        do99 = [UInt8](rapduBin[offset..<offset+4])
        let sw1 = rapduBin[offset+2]
        let sw2 = rapduBin[offset+3]
        offset += 4
        needCC = true`

danydev, 2023-02-06:
To be clear, when you say `and still occurs after the recent update of the module` you actually mean [2.0.2](https://github.com/AndyQ/NFCPassportReader/releases/tag/2.0.2) released 2 days ago, right? Because there was a change that could be somehow relevant.

boipqiod, 2023-02-06:
I apologize for the confusion. I was unaware of the recent update from two days ago and am currently using version 2.0.1, which appears to be causing an issue.

May I kindly confirm if my understanding is correct, that using the latest released version of 2.0.2 may resolve the issue at hand?

danydev, 2023-02-06:
We can't tell for sure, but there is a change, that may be relevant and solved a similar (but not equal) problem.

boipqiod, 2023-02-06:
Thank you for your quick reply.

After updating to version 2.0.2, we will test it with the passport and device that had the problem.

andycg0327, 2023-02-07:
I already set the log level to .debug, but still only these logs below.
BTW I change to 2.0.2, it still happen.

2023-02-07 14:39:21.4900 - tagReaderSessionDidBecomeActive
2023-02-07 14:39:21.5390 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x281081d40>)
2023-02-07 14:39:21.5400 - tagReaderSession:connected to tag - starting authentication
2023-02-07 14:39:21.541142+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.5550 - Error reading tag: sw1 - 0x6A, sw2 - 0x82
2023-02-07 14:39:21.5560 - reason: File not found
2023-02-07 14:39:21.5560 - PACE Failed - falling back to BAC
2023-02-07 14:39:21.5560 - Re-selecting eMRTD Application
2023-02-07 14:39:21.5630 - Starting Basic Access Control (BAC)
2023-02-07 14:39:21.5640 - BACHandler - deriving Document Basic Access Keys
2023-02-07 14:39:21.5650 - BACHandler - Getting initial challenge
2023-02-07 14:39:21.5730 - BACHandler - Doing mutual authentication
2023-02-07 14:39:21.6310 - DATA - [63, 94, 78, 234, 231, 212, 210, 39, 89, 168, 160, 170, 226, 228, 132, 153, 124, 212, 240, 144, 189, 152, 157, 37, 35, 208, 66, 246, 0, 167, 46, 235, 48, 44, 32, 12, 202, 112, 21, 141]
2023-02-07 14:39:21.6310 - BACHandler - complete
2023-02-07 14:39:21.6310 - Basic Access Control (BAC) - SUCCESS!
2023-02-07 14:39:21.634977+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.6350 - Reading tag - COM
2023-02-07 14:39:21.636001+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.6650 - TagReader - Number of data bytes to read - 23
2023-02-07 14:39:21.665841+0800 Demo[400:10466] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.6810 - DG Found - ["DG1", "DG2", "DG11", "DG12", "DG13"]
2023-02-07 14:39:21.682996+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.6830 - Reading tag - DG2
2023-02-07 14:39:21.684073+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.7140 - TagReader - Number of data bytes to read - 20339
2023-02-07 14:39:21.715069+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.743915+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.772655+0800 Demo[400:10466] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.801360+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.834324+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2023-02-07 14:39:21.863661+0800 Demo[400:10393] [CoreNFC] 00000002 82c343f0 -[NFCTagReaderSession setAlertMessage:]:93  (null)
Swift/Array.swift:405: Fatal error: Array index is out of range
2023-02-07 14:39:21.891999+0800 Demo[400:10466] Swift/Array.swift:405: Fatal error: Array index is out of range
(lldb)

atatlisu, 2023-06-04:
> Thank you for your quick reply.
> 
> After updating to version 2.0.2, we will test it with the passport and device that had the problem.

hello, was upgrading version solved your problem ?

atatlisu, 2023-06-04:
> I already set the log level to .debug, but still only these logs below. BTW I change to 2.0.2, it still happen.
>
> 2023-02-07 14:39:21.4900 - tagReaderSessionDidBecomeActive 2023-02-07 14:39:21.5390 - [...]

Hello, did you find a solution to this problem ?

jjynus, 2023-06-21:
//DO'99'
 // Mandatory, only absent if SM error occurs

add ----------------------------------------------------------------------------------------------------------------
line 154

guard rapduBin.count >= offset + 5 else {
       print("size error")
       let returnSw1 = (rapduBin.count >= offset+3) ? rapduBin[offset+2] : 0;
       let returnSw2 = (rapduBin.count >= offset+4) ? rapduBin[offset+3] : 0;
       return ResponseAPDU(data: [], sw1: returnSw1, sw2: returnSw2);
}
---------------------------------------------------------------------------------------------------------------- end
do99 = [UInt8](rapduBin[offset..<offset+4])

danydev, 2023-06-21:
@AndyQ that sounds promising, what do you think?

boipqiod, 2023-06-21:
> > Thank you for your quick reply.
> > After updating to version 2.0.2, we will test it with the passport and device that had the problem.
> 
> hello, was upgrading version solved your problem ?

Actually, it doesn't seem to be solved yet. Our app doesn't use nfc in earnest, so I'm waiting for it to be solved.

I just hope the great idea up there will solve it

danydev, 2023-06-21:
> Actually, it doesn't seem to be solved yet. Our app doesn't use nfc in earnest, so I'm waiting for it to be solved.
>
> I just hope the great idea up there will solve it

If you can give a try to the changes @jjynus proposed, that would be actually useful

AndyQ, 2023-06-21:
Looks interesting - will give it a test out.

AndyQ, 2023-06-21:
I've pushed up the change (just to main for the moment) - if you could check and let me know if it resolves the issue then I'll release 2.0.3 with it.

Thanks @jjynus

danydev, 2023-08-09:
@jjynus did you have a chance to test it?

Since it's been released, I suppose can be closed?

jjynus, 2023-08-09:
@danydev 
sure!!

closed 2023-10-05 by AndyQ