AndyQ/NFCPassportReader #174: Active Authentication fails - SW2 indicates 27 bytes still available

rbrouwer, 2023-02-28:
Some passports (I know of Finland) with active authentication using long m1 and different RSA SHA hashes will have response exceeding 256 bytes when doing tagReader.doInternalAuthentication.

This will result in response containing sw1 - 0x61, sw2 - 0x1B, which causes a NFCPasswordReaderError.ResponseError.

Increasing exceptedResponseLength in doInternalAuthentication removes this exception, but also only return 256 bytes of the response, which is then invalid for verifyActiveAuthentication (given part of the hash and the trailing bytes will be gone).

Possibly command chaining is required in those cases?

rbrouwer, 2023-02-28:
In this case doing a binary read (like TagReader in selectFileAndRead) gets the remaining data. So I guess either PassportReader or TagReader would need to handle that. It would also be nice if in case of a NFCPasswordReaderError.ResponseError the data would get passed along, so the first 256 do not have to be re-read using a binary read.

rbrouwer, 2023-02-28:
Never mind seems that this is simply a case of ignoring the sw1 - 0x61, sw2 - 0x1B response of this document is providing for no reason or increasing expectedResponseLength. Let's do some more digging into why the RSA decrypt seems to mess up. It does decrypt, it just does not seem like it is what I expect.

kirillivonin, 2023-04-05:
In my case, there seems to be a problem with Netherlands National ID where AA response is too short - 14 bytes vs 80 in JMRTD

rbrouwer, 2023-04-06:
> In my case, there seems to be a problem with Netherlands National ID where AA response is too short - 14 bytes vs 80 in JMRTD

I have not see any issues with Netherlands National ID (other than iPhones not having the strongest RFID antennas). What model is it precisely?/when was that ID issued? And it possibly might be a different issue or do you also get a identical response object?

kirillivonin, 2023-04-06:
might be - NFCPassportReader verifies the signature and greenlights it; however, our backend which uses same code as JMRTD to verify the signature says it's incorrect. Which is a bit weird as NFCPassportReader is based on JMRTD :/

ID is issued in 2021

rbrouwer, 2023-04-06:
> @rbrouwer
> Is there any reference on what iPhone models might experience this? It would be very helpful to know which models are "problematic"

iPhone 14 Pro really do not like the latest NLD passports (2021 and later).

kirillivonin, 2023-04-06:
New French passport is also finicky on iPhone 14 Pro (iPhone 13 works fine)