AndyQ/NFCPassportReader #180: Change in APDU data & instructions due to some passports are giving error while scanning

kmgoyani, 2023-05-03:
I'm scanning the passport of Denmark.  It is giving me timeout error while NFC reading. I have solved that issue by changing APDU data & instructions in the `readCardAccess()` method as below.

`let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 0x00, instructionCode: 0xA4, p1Parameter: 0x00, p2Parameter: 0x0C, data: Data([0x3f, 0x00, 0x30, 0x04]), expectedResponseLength: 256)`

It works for me.

danydev, 2023-09-05:
@AndyQ it sounds like a low hanging fruit, what do you think?

AndyQ, 2023-09-05:
I'll take a look hopefully in the next day or so!

AndyQ, 2023-09-08:
Took a look this morning and unfortunately that breaks PACE on all my test passports.
It doesn't like the extra parameters - 0x30, 0x04.

I get 0x67 0x00 - Wrong length response

Going back a few PRs - we had a discussion about this when I was implementing PACE:
https://github.com/AndyQ/NFCPassportReader/issues/44#issuecomment-602803793

danydev, 2023-09-08:
Those cases where something work in a document and fail in another are kind of scary,

I'm wondering, in those cases, is there the ability to try one way, and then fallback to the other in case of failure?

If this is possible, we could aim on safely\easily increasing the supported documents over time.

AndyQ, 2023-09-11:
I need to do some more testing but from previous experience, I typically, when you get a failure in a service, the passport often kills the connection and you have to restart then whole discovery phase again! (which is somewhat annoying).

Now I'm hoping that in this case, just trying to selecting the Master Card index with additional options (which may fail) may not do this but we'll see!

danydev, 2023-09-11:
I see! In those other cases, where you have to discover the whole discovery phase, you also mean that he needs to actually do again the step where the passport is actually read, right? Damn that's pity.

Maybe (just maybe) we could try to give an API where this is exposed somehow, so that a developer that know the "type" (aka issuer nationality) may try to set this on the options to be able to read that kind of document (I'm thinking on cases where people already do ocr for the MRZ, they could extract the issuer as well)