AndyQ/NFCPassportReader #183: tagReaderSession:failed to connect to tag - Tag response error / no response

cyberneid, 2023-06-11:
Starting from iOS 16, on iPhone 12, 13, and 14, we are experiencing a big issue with NFC. We are using it with Italian electronic identity cards, Azerbaijan electronic identity cards as well as other eid from other countries and passports as well.

With iPhone X, XR, and 7 it works like a charm

With iPhone 12, 13, and 14, the NFC reading starts but after BAC, during the reading of DGs, and very often during the reading of DG2, the NFC stops reading and returns this error:

[CoreNFC] -[NFCTagReaderSession transceive:tagUpdate:error:]:879 Error Domain=NFCError Code=102 "Tag response error / no response" UserInfo={NSLocalizedDescription=Tag response error / no response, NSUnderlyingError=0x281e81200 {Error Domain=nfcd Code=29 "Tag Error" UserInfo={NSLocalizedDescription=Tag Error, NSUnderlyingError=0x281e81d70 {Error Domain=com.apple.nfstack Code=20 "No response from tag" UserInfo={NSLocalizedDescription=No response from tag}}}}}

it is not a systematic error. Sometimes it works and the app reads all DGs, but very often we get that error.

We tried everything to understand the issue and to find a solution but with no luck.

Any suggestion?

AndyQ, 2023-08-30:
I've seem a couple of reports that some phones/passports are more fussy with iphone NFC location and require a more accurate or closer position to the chip (no idea why), which seemed to start with iOS 16, and can lose connection with the chip on longer data reads.  I can't replicate it here though so not really sure what I can do.

VincentLaula, 2023-09-27:
I am also looking at similar reports but without having been able to reproduce yet.
I am interested in all reports, findings, or detailed test reports. (Note that enabling the iOS NFC wallet profile can provide more detailed and low-level iOS traces.)

I have experienced some frame delay timing violations on some iPhones in the past.
Can the issue be related to a specific NFC chip variant in the phone?

Was the issue reported in the Apple Feedback system?

AndyQ, 2023-10-05:
I've noticed this happen with my iPhone 15 Pro on one of my test passports.  All other passports work fine except this one.  However, the same passport works fine with an iPhone 11.

Note - The ReadID also shows the same issues with my iPhone 15 Pro so suspect either a CoreNFC issue of something odd with the NFC Chip!  Will be raising a feedback with Apple.

RtifactApps, 2023-10-24:
Same here!
I have an iPhone 14 pro and I keep getting connection lost on `SOD` and `DG2`. But it works fine on my old iPhone 7+ and on the iPhone 11 from my work.

jmecom, 2024-08-23:
Hey @AndyQ -- I've seen similar problems on a different project, where NFC performs well on old iPhones, like the 11, but new ones fail.

Did you hear anything back from Apple?

phisakel, 2024-09-10:
It fails to read Greek ID cards with the same message

mnive93, 2024-10-07:
I am facing the same issue. It almost fails everytime on iPhone 15 Pro max with the following error. What's the solution for this? 

` Error Domain=NFCError Code=102 "Tag response error / no response" UserInfo={NSLocalizedDescription=Tag response error / no response, NSUnderlyingError=0x303bd6af0 {Error Domain=nfcd Code=29 "Tag Error" UserInfo={NSLocalizedDescription=Tag Error, NSUnderlyingError=0x303bd6910 {Error Domain=com.apple.nfstack Code=20 "No response from tag" UserInfo={NSLocalizedDescription=No response from tag}}}}}`

vguerci, 2024-10-08:
Just facing the same behaviour with a brand new iPhone 16 Pro / iOS 18.0.1, but I don't have any issues with an iPhone 14 Pro. That happens on all my documents, ReadID failing similarly on one, but performing better on one of them 🤷

AndyQ, 2024-10-08:
For info, I'm seeing similar things on my 16 Pro.  No ideas at the moment sadly

mnive93, 2024-10-10:
We see that if the Wifi is turned off in these models scanning works as expected. [[Reference]https://genk.vn/da-tim-ra-cach-sua-loi-quet-cccd-bang-nfc-cho-nguoi-dung-iphone-20240626182838168.chn)]

tastydev, 2024-10-10:
> We see that if the Wifi is turned off in these models scanning works as expected. [Reference](https://genk.vn/da-tim-ra-cach-sua-loi-quet-cccd-bang-nfc-cho-nguoi-dung-iphone-20240626182838168.chn)

I can confirm it with my iPhone 16 Pro... wtf. The iOS NFC API returns quite often with the tagResponseError. As soon as i turn my WIFI off my success rate is increased drastically! In the worst case it is an hardware issue i.e wifi electromagnetic waves are interfering the nfc field communication... The best case is an software issue inside ios 18.x.x. I will also contact apple with some statistics + proofs.

vguerci, 2024-10-14:
I confirm as well, but turning off wifi isn't enough for me, I need to switch on Airplane mode...

I doubt this is an iOS 18 issue, as this issue is older than it, sounds more like a device-specific issue, like you may be unlucky and get a bad NFC reader in your device, maybe interfering with wifi?

May try to go to the Genius bar and see if I can get and 16 Pro replacement that works, I have a coworker with a 16 Pro max that supposed to work 🤷

@AndyQ Does it fails on all of your test documents or just some?

AndyQ, 2024-10-14:
Really annoyingly - today - I haven't managed to get a single fail!  all my reads are working.

Was trying to replicate this today so I can update a feedback with a sysdiagnose and pointing to this issue (really annoying as last week I was getting frequent timeouts!).

vguerci, 2024-10-22:
Just upgraded my 16 Pro to iOS 18.1 RC, and I can now read my docs...
Not completely flawlessly, I am still having some disconnections, but that works while it was systematically failing with 18.0.1.