AndyQ/NFCPassportReader #184: Failure to read a Hungarian passport

TomerPickerT, 2023-06-12:
Hello everyone, I'm having trouble reading the Hungarian passport. The logs are attached - any ideas would be appreciated.
I know this passport supports PACE authentication, and I saw it failed on step4. 

Using version 1.0.1
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x281668090>)
"TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x90 sw2:0x00"
"TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x90 sw2:0x00"
"TagReader - sending [0x00, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x04]"
"TagReader - Received response"
"TagReader [unprotected] [0x31, 0x14, 0x30, 0x12, ], sw1:0x90 sw2:0x00"
TagReader - Number of data bytes to read - 20
"TagReader - sending [0x00, 0xB0, 0x00, 0x02, 0x14]"
"TagReader - Received response"
"TagReader [unprotected] [0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x04, 0x02, 0x02, 0x02, 0x01, 0x02, 0x02, 0x01, 0x0d, ], sw1:0x90 sw2:0x00"
Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-128
doPace - inpit parameters
paceOID - 0.4.0.127.0.7.2.2.4.2.2
parameterSpec - 927
mappingType - GM
agreementAlg - ECDH
cipherAlg - AES
digestAlg - SHA-1
keyLength - 128
keyLength - BS5962035385120393012039
paceKey - [0x6f, 0x6e, 0x90, 0xe2, 0x70, 0x78, 0x57, 0x06, 0xcc, 0xd7, 0xb7, 0x99, 0x10, 0x3b, 0x9c, 0x97, ]
"TagReader - sending [0x00, 0x22, 0xC1, 0xA4, 0x0F, 0x80, 0x0A, 0x04, 0x00, 0x7F, 0x00, 0x07, 0x02, 0x02, 0x04, 0x02, 0x02, 0x83, 0x01, 0x01]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x90 sw2:0x00"
Doing PACE Step1...
"TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x02, 0x7C, 0x00, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [0x7c, 0x12, 0x80, 0x10, 0x6d, 0x67, 0xfb, 0xce, 0x73, 0x17, 0x16, 0x1c, 0x34, 0xff, 0x7a, 0xe9, 0xb6, 0xa0, 0xbe, 0xb2, ], sw1:0x90 sw2:0x00"
Doing PACE Step2...
   Using General Mapping (GM)...
Generating ECDH mapping keys from parameterSpec - 927
Sending public mapping key to passport..
"TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x45, 0x7C, 0x43, 0x81, 0x41, 0x04, 0x22, 0xAA, 0x69, 0x40, 0x7F, 0x4E, 0x7E, 0x4F, 0x7C, 0x4A, 0x3A, 0x3F, 0xC3, 0x03, 0xBF, 0x57, 0x1C, 0x17, 0x14, 0x4E, 0x59, 0x14, 0xF8, 0x8E, 0x74, 0x77, 0x5A, 0xB4, 0xBF, 0xBA, 0x7E, 0x50, 0x3E, 0xE4, 0xD9, 0x3A, 0x16, 0x0E, 0x05, 0x54, 0xAC, 0xEB, 0xDC, 0x6F, 0x8F, 0xBF, 0x47, 0xD1, 0x72, 0x46, 0xBA, 0x79, 0xC2, 0xCF, 0x82, 0x4E, 0xCD, 0x28, 0x45, 0xE4, 0x87, 0x1C, 0xE2, 0xD9, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [0x7c, 0x43, 0x82, 0x41, 0x04, 0x4a, 0x2d, 0x45, 0xa9, 0x26, 0x73, 0xc0, 0x90, 0xf6, 0x48, 0x76, 0xb0, 0x03, 0x27, 0x27, 0x97, 0x1e, 0x52, 0x9c, 0xdc, 0x78, 0x0f, 0x81, 0x32, 0xb3, 0x63, 0xcd, 0xfa, 0x3e, 0x3b, 0xbd, 0x01, 0x3c, 0xb0, 0xa4, 0x1f, 0x19, 0x3f, 0x37, 0xed, 0x78, 0x9a, 0xf3, 0x56, 0x77, 0x04, 0x39, 0x15, 0xd9, 0x3e, 0x15, 0xcb, 0xf0, 0x9e, 0x63, 0xd9, 0xc0, 0xb7, 0xba, 0x6c, 0x6b, 0xc8, 0xc0, 0xe8, ], sw1:0x90 sw2:0x00"
Received passports public mapping key
Doing ECDH Mapping agreement
Doing PACE Step3 - Key Exchange
Generated Ephemeral key pair
Sending ephemeral public key to passport
"TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x45, 0x7C, 0x43, 0x83, 0x41, 0x04, 0x9D, 0x35, 0x13, 0xF0, 0x61, 0x5E, 0x0E, 0x47, 0xE2, 0x17, 0x5E, 0x81, 0x06, 0x6A, 0x55, 0x3C, 0xDF, 0x57, 0xFE, 0x39, 0x36, 0x4B, 0x3D, 0xC1, 0x3B, 0xF9, 0x5D, 0x5F, 0x70, 0xF6, 0x84, 0x3A, 0x25, 0xB0, 0x82, 0xE0, 0x37, 0x35, 0xC5, 0xC4, 0xFD, 0xE9, 0xF8, 0x31, 0xFF, 0xB9, 0xCE, 0x79, 0xB7, 0x99, 0xFA, 0x5B, 0xC3, 0x2A, 0x03, 0xC3, 0x2B, 0xBF, 0xCE, 0x17, 0x68, 0x38, 0x37, 0x33, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [0x7c, 0x43, 0x84, 0x41, 0x04, 0x3c, 0xdc, 0x36, 0x0d, 0xaf, 0xab, 0x08, 0x98, 0x6a, 0x85, 0x49, 0x01, 0x0d, 0xda, 0x2d, 0xbd, 0x8d, 0x86, 0x03, 0x56, 0xe0, 0x7e, 0x6f, 0x0b, 0x54, 0x28, 0xbe, 0xf7, 0x78, 0x44, 0x9f, 0x93, 0x52, 0x10, 0x77, 0x74, 0xed, 0x68, 0x91, 0x76, 0x65, 0x50, 0x3c, 0x0b, 0x81, 0xd7, 0x82, 0x14, 0x6f, 0x71, 0x81, 0x51, 0xb0, 0x80, 0x3a, 0x5b, 0x8c, 0x7b, 0xb2, 0x51, 0x1b, 0x76, 0x10, 0x81, ], sw1:0x90 sw2:0x00"
Doing PACE Step4 Key Agreement...
Computing shared secret...
Deriving ksEnc and ksMac keys from shared secret
Generating authentication token
Sending auth token to passport
"TagReader - sending [0x00, 0x86, 0x00, 0x00, 0x0C, 0x7C, 0x0A, 0x85, 0x08, 0x80, 0xF9, 0x25, 0x06, 0x65, 0x66, 0x31, 0xD3, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x63 sw2:0x00"
PACE Failed - falling back to BAC
Re-selecting eMRTD Application
"TagReader - sending [0x00, 0xA4, 0x04, 0x0C, 0x07, 0xA0, 0x00, 0x00, 0x02, 0x47, 0x10, 0x01]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x90 sw2:0x00"
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
"TagReader - sending [0x00, 0x84, 0x00, 0x00, 0x08]"
"TagReader - Received response"
"TagReader [unprotected] [0x7a, 0xfc, 0x1e, 0xd6, 0x9b, 0xbf, 0xb8, 0x4b, ], sw1:0x90 sw2:0x00"
BACHandler - Doing mutual authentication
"TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x8E, 0x98, 0xB4, 0x81, 0xFF, 0x70, 0x09, 0xE6, 0xB5, 0x32, 0xCE, 0x86, 0x0B, 0x9C, 0x78, 0x26, 0x5F, 0x38, 0x9E, 0x91, 0xB9, 0x9F, 0xA7, 0xB9, 0x83, 0xA8, 0xD3, 0xFB, 0x4D, 0x13, 0xFC, 0xEB, 0x01, 0x37, 0x09, 0x3E, 0x0A, 0xEB, 0x54, 0x24, 0x00]"
"TagReader - Received response"
"TagReader [unprotected] [], sw1:0x63 sw2:0x00"
DATA - []

pawisoon, 2023-06-13:
What was the problem? Is it working now? I see you closed the issue

TomerPickerT, 2023-06-13:
Thanks for reaching out!
The problem was on my side, it is working perfectly now.
And thanks for this amazing package!