AndyQ/NFCPassportReader #187: Error reading tag: sw1 - 0x69, sw2 - 0x82

leonacky, 2023-07-07:
I tried to read ID Card of Vietnam and get this error
Error reading tag: sw1 - 0x69, sw2 - 0x82
Some application in appstore to read id card of vietnam. I see that they don't need mrzkey to decode data, they just need id card number to read data.
Anyone can explain me how do they do?

LuuNguyen0811, 2023-07-17:
In Vietnamese card, you can decode from ID number to same data inside mrzkey

tungtomvn, 2023-07-21:
> 

can you give me the source code? Thanks!

tungtomvn, 2023-07-21:
> In Vietnamese card, you can decode from ID number to same data inside mrzkey

can you give me the source code? Thanks!

tungtomvn, 2023-07-21:
> I tried to read Vietnamese ID Card and get this error `Error reading tag: sw1 - 0x69, sw2 - 0x82` Some application in appstore to read Vietnamese id card. I see that they don't need mrzkey to decode data, they just need id card number to read data. Anyone can explain me how do they do?
> 
> My logs:
> 
> ```
> 2023-07-07 12:8:50.4460 - Using version 1.1.4.1
> 2023-07-07 12:8:50.5440 - tagReaderSessionDidBecomeActive
> 2023-07-07 12:8:50.5890 - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x28206ca50>)
> 2023-07-07 12:8:50.5900 - tagReaderSession:connected to tag - starting authentication
> 2023-07-07 12:8:50.6150 - Error reading tag: sw1 - 0x69, sw2 - 0x82
> 2023-07-07 12:8:50.6150 - reason: Security status not satisfied
> 2023-07-07 12:8:50.6160 - PACE Failed - falling back to BAC
> 2023-07-07 12:8:50.6160 - Re-selecting eMRTD Application
> 2023-07-07 12:8:50.6310 - Starting Basic Access Control (BAC)
> 2023-07-07 12:8:50.6310 - BACHandler - deriving Document Basic Access Keys
> 2023-07-07 12:8:50.6320 - BACHandler - Getting initial challenge
> 2023-07-07 12:8:50.6420 - BACHandler - Doing mutual authentication
> 2023-07-07 12:8:50.6660 - DATA - [129, 1, 155, 56, 243, 115, 97, 81, 77, 145, 227, 90, 35, 80, 91, 24, 112, 231, 59, 228, 178, 140, 249, 227, 124, 142, 90, 106, 170, 192, 236, 145, 115, 214, 221, 74, 237, 23, 169, 147]
> 2023-07-07 12:8:50.6670 - BACHandler - complete
> 2023-07-07 12:8:50.6670 - Basic Access Control (BAC) - SUCCESS!
> 2023-07-07 12:8:50.6730 - Reading tag - COM
> 2023-07-07 12:8:50.7050 - TagReader - Number of data bytes to read - 24
> 2023-07-07 12:8:50.7250 - DG Found - ["DG1", "DG2", "DG3", "DG13", "DG15", "DG14"]
> 2023-07-07 12:8:50.7270 - Reading tag - DG14
> 2023-07-07 12:8:50.7580 - TagReader - Number of data bytes to read - 382
> 2023-07-07 12:8:50.8600 - Performing Chip Authentication - number of public keys found - 1
> 2023-07-07 12:8:50.9850 - Public Key successfully sent to passport!
> 2023-07-07 12:8:50.9880 - Restarting secure messaging using AES encryption
> 2023-07-07 12:8:50.9890 - Reading tag - SOD
> 2023-07-07 12:8:51.0200 - TagReader - Number of data bytes to read - 1568
> 2023-07-07 12:8:51.3420 - Reading tag - DG1
> 2023-07-07 12:8:51.3740 - TagReader - Number of data bytes to read - 93
> 2023-07-07 12:8:51.3980 - Reading tag - DG2
> 2023-07-07 12:8:51.4320 - TagReader - Number of data bytes to read - 13194
> 2023-07-07 12:8:53.9890 - Reading tag - DG13
> 2023-07-07 12:8:54.0220 - TagReader - Number of data bytes to read - 422
> 2023-07-07 12:8:54.1120 - Reading tag - DG15
> 2023-07-07 12:8:54.1440 - TagReader - Number of data bytes to read - 294
> 2023-07-07 12:8:54.2050 - Performing Active Authentication
> 2023-07-07 12:8:54.5380 - tagReaderSession:failed to connect to tag - Tag response error / no response
> 2023-07-07 12:8:57.3050 - tagReaderSession:didInvalidateWithError - Session invalidated by user
> ```

For ICAO, with PACE auth, just need ID card number but BAC need id card, date of birth, date of expired

Niraj91, 2023-08-07:
Hi @tungtomvn @LuuNguyen0811 @leonacky ,

I have the same issue with OMAN resident card, Do guys find any solutions?

Thanks

leonacky, 2023-08-18:
A recommend for Vietnamese guy. Try to read with MRZ and read DG13

duyanhv, 2023-08-20:
imo, they use the api given by RAR service to query mrz data from the card's ID -> no need for OCR or input by hand. VND 3000 / request

duyanhv, 2023-08-20:
> A recommend for Vietnamese guy. Try to read with MRZ and read DG13

do you have trouble decoding DG13? i tried decoding it but the format is a mess

AndyQ, 2023-08-30:
There is a PR for authenticating with CAN that may be of use, I've not merged it as I have no way of testing this as none of my available passports support this.

Would be interested in knowing is this works for you.

leonacky, 2023-08-30:
> > A recommend for Vietnamese guy. Try to read with MRZ and read DG13
> 
> do you have trouble decoding DG13? i tried decoding it but the return value is a mess

Try to decode it with ASN1

leonacky, 2023-08-30:
> Hi @tungtomvn @LuuNguyen0811 @leonacky ,
> 
> I have the same issue with OMAN resident card, Do guys find any solutions?
> 
> Thanks

I don't know about OMAN ID Card

leonacky, 2023-08-30:
I have resolved this issue

closed 2023-08-30 by leonacky

TranHoaiHung, 2023-09-19:
@leonacky Hello bro, how to read it?

starptit, 2023-10-03:
@leonacky  Was you able to do passive auth? With the generated pem file?

kevinarch, 2024-02-23:
How did you solve this  **Security status not satisfied** error ?

AVGThor, 2024-03-07:
> I have resolved this issue

@leonacky how to fix it, can you show me please, i got same problem

longcon9999, 2024-04-24:
> I have resolved this issue

Hi @leonacky, Could you use PACE with Vietnamese ID Card?

vijaysahani4115, 2024-06-21:
While doing Canada passport read via BAC , showing Not genuine , failing Country signing certificate.
[image]
[image]

Any specific reason ?

alien142, 2024-07-03:
I got same issue: Error reading tag: sw1 - 0x69, sw2 - 0x82
DataGroup is a uint8 list like [109, 130, 1, 174, 48, 130...]. Please give me some hint to decode it?

HieeusNguyen, 2024-12-19:
Hey bro @leonacky, how did you fix it?