AndyQ/NFCPassportReader #192: Passive Authentication not working with generated PEM file

starptit, 2023-10-03:
I'm trying to do passive authentication with the generated PEM file, but it does not work.

The error from Open SSL show that:
`error 20 at 0 depth lookup:unable to get local issuer certificate`

I've prepared:
- Latest ICAO PKD file from official website
- PEM file that is generated by script (after fixing existing bug)

AndyQ, 2023-10-05:
Are you sure that you have the right master list loaded?  I've been testing with some passports and can't replicate this.

starptit, 2023-10-06:
I've figured out that the public ICAO PKD file `icaopkd-002-complete-000236.ldif` does not contain the CA of my country (Vietnam), hence the PEM file will not work (even when my country joined the ICAO in 2021).

In addition, I learn that we can also get the `master list` from BSI German, which I investigate that it contains the certificate of my country. However, the master list file is encoded to extension .ml as describe in this discussion https://github.com/AndyQ/NFCPassportReader/issues/117.

Note that, BSI German might not be official source to any countries, then if you're doing government related application, try to contact ICAO instead, to other stuff, it's good to go.

My business does not require to do passive authentication anymore, then I'm closing this discussion.

To summarize:
- The error shows when the generated .PEM file do not contain the CA of scanning country.
- Look inside the .ldif file to locate if your country code is contained in the list.
- If the ICAO .ldif does not work, try to do with master list file from BSI German.
- Note that BSI German might not be official documents to your application.