AndyQ/NFCPassportReader #194: Error identifying Active Authentication Hash Algorithm - Australian ePassport

harry-anderson, 2023-11-03:
I used the sample application (SPM) to scan an Australian ePassported Issued 2023 and found Passive Authentication succeeds while Active Authentication fails with the following error printed in the session output:
```
...
2023-11-02 8:29:10.8570 - Reading tag - DG15
2023-11-02 8:29:10.9570 - TagReader - Number of data bytes to read - 294
2023-11-02 8:29:11.0460 - Performing Active Authentication
2023-11-02 8:29:11.8720 - Error identifying Active Authentication RSA message digest hash algorithm
2023-11-02 8:29:11.9670 - Passport passed SOD Verification
2023-11-02 8:29:11.9790 - Parse SOD - Using Algo - SHA256
2023-11-02 8:29:11.9800 -       - Hashes     - [NFCPassportReader.DataGroupId.DG15: "F3FB8A3D6346B37293BBF5939ED7C04369B7F8B642A27B6BAFA979CABEF561B9", NFCPassportReader.DataGroupId.DG1: "2C3BFF5F63EBE095A5F9E16C23C1D3792F7571635A56F2C1AB693E2D051E63DA", NFCPassportReader.DataGroupId.DG2: "73A88BDC89B7215D824174D67C53DF6B86E32AC9D262E67EFB2CA9F99A2F3F29", NFCPassportReader.DataGroupId.DG14: "06CAA3C87D4CF49923140662BC4565E0A578DA2F5717363869D30B9E9C7FB80F", NFCPassportReader.DataGroupId.DG7: "47636EFC50B4AEDC7D74C1DBFDA03F5B860EE107D0A4718BEA248C99AE843D42"]
2023-11-02 8:29:11.9800 - Passport passed Datagroup Tampering check
```
Any idea what is happening here? 

Thanks 🙏

rbrouwer, 2023-11-22:
DG15 ends with a trailer. Using that trailer the correct RSA-HASH type is selected ([here](https://github.com/AndyQ/NFCPassportReader/blob/c1d8dab53bc7af2d579e846d9e48acc1883661a2/Sources/NFCPassportReader/NFCPassportModel.swift#L310)).

Apparently that trailer did not make sense. So unless you know what bytes you received and what the hashing algorithm should have been, it will be difficult to say anything about it.

AndyQ, 2024-02-13:
Hi @harry-anderson, if you can let me know what the hash type byte data is then I can look to see what I'm missing.

e.g. change NFCPassportModel : verifyActiveAuthentication (around line 347) to:
                        Logger.passportReader.error( "Error identifying Active Authentication RSA message digest hash algorithm - '(hashTypeByte)" )

harry-anderson, 2024-02-14:
Hi @AndyQ ,

Thanks for you message, I made the change to log the hash type byte as you mention
```swift
 Log.error( "Error identifying Active Authentication RSA message digest hash algorithm - hashTypeByte = \(hashTypeByte)" )
```

here is what I am getting:
```
2024-02-14 19:39:30.5330 - Error identifying Active Authentication RSA message digest hash algorithm - hashTypeByte = 237
```
 Let me know if there is anything else you need 👍

harry-anderson, 2024-02-14:
@AndyQ Here you go:
```
2024-02-14 20:17:49.1470 - Performing Active Authentication
2024-02-14 20:17:49.5510 - Active Authentication
2024-02-14 20:17:49.5510 -    challange - E9EBB944E5E5C143
2024-02-14 20:17:49.5540 -    signature - 8C694DE474D9598998B771D32EE0D043030423C8F900AB2DF2D0F12B513C3D7D3989712CE40648DF23F5242737C74EC98965C4D8DF76A8F16E81D8BC9B00F95B2D6DA27742391CCCF22D46D3BAB9F32023C3C7B393CDAF7F3CC013620EC3A131244F277C0CA81E577681043FCBE8340DF4B191CCD73018643BE3EE434DDBDE8ED47F2481F1029FC3F91CF001D9F8E98FDF62479982BC5E885FF6EFC9733748F77ABC2FA7425486C71A155312E8FDFEABA3F8B0C5AE8399704DA6C6D74FEEFA47CA99C3986FC7FFCBADB73FA1B2C5A57EC59A1271830A0834A7CA7542A7F595
2024-02-14 20:17:49.5550 - AA Signature decrypted
2024-02-14 20:17:49.5550 - JfqkhM+yOITe1ffEXsMXEZK7QY8bpvSwLSR4DubFcnP+VmUqrjc2Mcd5EIHvnBiCVZLTgn6+ezQqW1w8nSAmKNg+R3OW6F76axPWX38LAVxF/sM+dZ6g8ZN+KJA3tLoFO/V0lmx6XyU9bWR7zyGvH/vrSRUOoGvJ8c5L1BQoDMZOhbyZnuBLtRYN9fip0XjQqiUloKjgG/ZzB1FphZ6A77KyZ20McKkK12BqrEdcBIhB8mJMYWXsoRhypaLrqDcOK3SoU+wDk3SJlze7o3/YPMlkMw2K2q0ZDmOBi92ca1f8ZXqINSLgMx0OkNmfFDaYirj7VxnOWcH4OFm4mkijlw==
2024-02-14 20:17:49.5560 - Error identifying Active Authentication RSA message digest hash algorithm - hashTypeByte = 151 97
2024-02-14 20:17:49.6470 - Passport passed SOD Verification
2024-02-14 20:17:49.6540 - Parse SOD - Using Algo - SHA256
2024-02-14 20:17:49.6540 -       - Hashes     - [NFCPassportReader.DataGroupId.DG15: "F3FB8A3D6346B37293BBF5939ED7C04369B7F8B642A27B6BAFA979CABEF561B9", NFCPassportReader.DataGroupId.DG1: "2C3BFF5F63EBE095A5F9E16C23C1D3792F7571635A56F2C1AB693E2D051E63DA", NFCPassportReader.DataGroupId.DG7: "47636EFC50B4AEDC7D74C1DBFDA03F5B860EE107D0A4718BEA248C99AE843D42", NFCPassportReader.DataGroupId.DG14: "06CAA3C87D4CF49923140662BC4565E0A578DA2F5717363869D30B9E9C7FB80F", NFCPassportReader.DataGroupId.DG2: "73A88BDC89B7215D824174D67C53DF6B86E32AC9D262E67EFB2CA9F99A2F3F29"]
2024-02-14 20:17:49.6540 - Passport passed Datagroup Tampering check
```

Thanks in advance

AndyQ, 2024-02-15:
Hmmm I'm a little stumped here!  Going through docs and other code, unless I'm missing something I have no idea what Algorithm is being used here.

I did see some mention of it possibly now also using rsspss but as yet no other details (https://crypto.stackexchange.com/questions/75058/using-epassport-with-active-chip-authentication-to-sign-documents)

One thing to possibly also try, can you try the ReadID app and see if that passes Active Authentication?

harry-anderson, 2024-02-15:
No worries, thanks for looking into it. 

I tried ReadID on iOS and it does seem to pass. Its a new passport from 2023 so maybe something has been introduced.

> Active Authentication SUCCEEDED Signature Checked

petteri-huusko-op, 2024-02-15:
Just my 2 cents, https://github.com/AndyQ/NFCPassportReader/issues/174 (fixed in 2.1.1) also presented itself in similar way. The active authentication algorithm was misidentified because the correct byte that is supposed to select the algorithm was not actually read.

AndyQ, 2024-02-15:
Interesting - I wonder if this is also the case - @harry-anderson could you try turning on full logging and seeing if we get a similar code 0x61 when reading the AA signature?

Also, are you using 2.1.1?

rbrouwer, 2024-02-15:
I doubt it would be #174. That would cause "Error reading tag: sw1 - 0x61, sw2 - 0x" to appear, which does not in this case. ([TagReader.swift#L284](https://github.com/AndyQ/NFCPassportReader/blob/84e11ec67a5c0d5ff54f918182fd4658801cc7f2/Sources/NFCPassportReader/TagReader.swift#L284C38-L284C113))

In this case the response seems to have received the signature "OK". Therefore we reach [NFCPassportModel.swift#L305 ](https://github.com/AndyQ/NFCPassportReader/blob/84e11ec67a5c0d5ff54f918182fd4658801cc7f2/Sources/NFCPassportReader/NFCPassportModel.swift#L305 ), which you can see in supplied logs. It seems very likely that the response is indeed not the full response and we therefore do not see the expected trailer. In which case the expectedResponseLength on [TagReader.swift#L55](https://github.com/AndyQ/NFCPassportReader/blob/84e11ec67a5c0d5ff54f918182fd4658801cc7f2/Sources/NFCPassportReader/TagReader.swift#L55) is causing the response to be cut short.
Possibly a "-1" instead of 256 might solve it then?

AndyQ, 2024-02-15:
I was kind of leaning in that direction but wasn't quite sure what affect that would have on existing reads. I'll give it a go on my test passports see what happens. @harry-anderson if you could also try that and see if it gives you more data (you'll need to be on latest version though for it to try to automatically read the extra data).

harry-anderson, 2024-02-15:
Ok I switch to the `main` branch, previously was on version `2.0.2` and using the SPM App example for testing.
```
git log
commit 84e11ec67a5c0d5ff54f918182fd4658801cc7f2 (HEAD -> main, tag: 2.1.1, origin/main, origin/HEAD)
Author: Andy Qua <andy.qua@gmail.com>
Date:   Tue Feb 13 12:19:53 2024 +0000

    Updated to 2.1.1
```

Sorry I am not sure how to turn on full logging, can you point me to the file to change this?

Here is what I am getting:
```
Performing Active Authentication
Generated Active Authentication challange - 9BAC8602638B5B80
TagReader - sending [0x00, 0x88, 0x00, 0x00, 0x08, 0x9B, 0xAC, 0x86, 0x02, 0x63, 0x8B, 0x5B, 0x80, 0x00]
		SSC: 00000000000001C0
	Increment SSC with 1
		SSC: 00000000000001C1
Mask class byte and pad command header
	CmdHeader: 0C880000800000000000000000000000
Pad data
	Data: 9BAC8602638B5B808000000000000000
Encrypt data with KSenc
	EncryptedData: 90EF88F65FECAC6A417EDABABF059EA8
Build DO'87
	DO87: 87110190EF88F65FECAC6A417EDABABF059EA8
Build DO'97
	DO97: [151, 1, 0]
Concatenate CmdHeader and DO87 and DO97
	M: 0C88000080000000000000000000000087110190EF88F65FECAC6A417EDABABF059EA8970100
Compute MAC of M
	Concatenate SSC and M and add padding
		N: 000000000000000000000000000001C10C88000080000000000000000000000087110190EF88F65FECAC6A417EDABABF059EA897010080000000000000000000
aesMac - mac - 9D0036E64B06B281AFE80F223DCB1D4C00000000000000000000000000000000
	Compute MAC over N with KSmac
		CC: 9D0036E64B06B281
Build DO'8E
	DO8E: 8E089D0036E64B06B281
Construct and send protected APDU
	ProtectedAPDU: 0C8800002087110190EF88F65FECAC6A417EDABABF059EA89701008E089D0036E64B06B28100
TagReader - [SM] [0x0C, 0x88, 0x00, 0x00, 0x20, 0x87, 0x11, 0x01, 0x90, 0xEF, 0x88, 0xF6, 0x5F, 0xEC, 0xAC, 0x6A, 0x41, 0x7E, 0xDA, 0xBA, 0xBF, 0x05, 0x9E, 0xA8, 0x97, 0x01, 0x00, 0x8E, 0x08, 0x9D, 0x00, 0x36, 0xE6, 0x4B, 0x06, 0xB2, 0x81, 0x00]
TagReader - Received response, size 242b
	Increment SSC with 1
		SSC: 00000000000001C2
Receive response APDU of MRTD's chip
	RAPDU: 8781E101E1F46B42768F31B3111A3E6438A1C623C8BBBB86F984EF52865FC7396F4BB884BCED68BFB6FD61FF9C93FC802C146FFC6F205D37F6398A30AA7B66E4BB9A5141D419436E7EA62118A3B1584F35F3C110424E0D0B57D646766F1D70D9BBBEADCA3878522B3B49CDBD2A5E958C149F765EA26DD0EC9C40CAA0636755FEF6C955D400EFA19FB6B60C064FD99D8B70FA6F75BEF09716DB7E402E8A9758087C816804EA8E66B14C96ECFA0B1554419E258BE81481A781359CE003D06CD94D5B8CC9A4F92D2AD4BBB99DA0630380647E173B7725DB65F8216E38BEB73B7C09FC551E98990290008E0864E5E66223930B249000
Verify RAPDU CC by computing MAC of  DO'87 DO'99
	Concatenate SSC and  DO'87 DO'99 and add padding
		K: 000000000000000000000000000001C28781E101E1F46B42768F31B3111A3E6438A1C623C8BBBB86F984EF52865FC7396F4BB884BCED68BFB6FD61FF9C93FC802C146FFC6F205D37F6398A30AA7B66E4BB9A5141D419436E7EA62118A3B1584F35F3C110424E0D0B57D646766F1D70D9BBBEADCA3878522B3B49CDBD2A5E958C149F765EA26DD0EC9C40CAA0636755FEF6C955D400EFA19FB6B60C064FD99D8B70FA6F75BEF09716DB7E402E8A9758087C816804EA8E66B14C96ECFA0B1554419E258BE81481A781359CE003D06CD94D5B8CC9A4F92D2AD4BBB99DA0630380647E173B7725DB65F8216E38BEB73B7C09FC551E98990290008000000000000000
	Compute MAC with KSmac
aesMac - mac - 64E5E66223930B24AD03DE8D9197C3C800000000000000000000000000000000
		CC: 64E5E66223930B24
	Compare CC with data of DO'8E of RAPDU
		64E5E66223930B24  == 64E5E66223930B24 ? true
Decrypt data of DO'87 with KSenc
	DecryptedData: 10566A7DCF25D4DCBD8AB84BDFEF195C44701DA173EDC31AC2593BF47E4D499A7BED43437B8AEFE822C4A3BE41241E5427221F9759C7E4B86DFD8CABF7DA8F83252700B2D6BDC1F0847A127727C0D3BE4FE6427A61B6314477CCFA212517E065BC661876C1344C9202FE52B43F5FAAE530B9272A63DC4A28FD5A197EB884A407DD3417788FF3B9EE7A290D532F42AAB8977D2C2A8320BA88A9355094CE1DA376759F0C8BC903819D436D6889B95379B8CCF01E50D277D571FEA02C8CD419E88EEBE3CAA29130B7B2562CB089C7D95B3D636D0B11A2B501242EE2AFD83DE73D
Unprotected APDU: [10566A7DCF25D4DCBD8AB84BDFEF195C44701DA173EDC31AC2593BF47E4D499A7BED43437B8AEFE822C4A3BE41241E5427221F9759C7E4B86DFD8CABF7DA8F83252700B2D6BDC1F0847A127727C0D3BE4FE6427A61B6314477CCFA212517E065BC661876C1344C9202FE52B43F5FAAE530B9272A63DC4A28FD5A197EB884A407DD3417788FF3B9EE7A290D532F42AAB8977D2C2A8320BA88A9355094CE1DA376759F0C8BC903819D436D6889B95379B8CCF01E50D277D571FEA02C8CD419E88EEBE3CAA29130B7B2562CB089C7D95B3D636D0B11A2B501242EE2AFD83DE73D] 90 00
TagReader [SM - unprotected] [0x10, 0x56, 0x6a, 0x7d, 0xcf, 0x25, 0xd4, 0xdc, 0xbd, 0x8a, 0xb8, 0x4b, 0xdf, 0xef, 0x19, 0x5c, 0x44, 0x70, 0x1d, 0xa1, 0x73, 0xed, 0xc3, 0x1a, 0xc2, 0x59, 0x3b, 0xf4, 0x7e, 0x4d, 0x49, 0x9a, 0x7b, 0xed, 0x43, 0x43, 0x7b, 0x8a, 0xef, 0xe8, 0x22, 0xc4, 0xa3, 0xbe, 0x41, 0x24, 0x1e, 0x54, 0x27, 0x22, 0x1f, 0x97, 0x59, 0xc7, 0xe4, 0xb8, 0x6d, 0xfd, 0x8c, 0xab, 0xf7, 0xda, 0x8f, 0x83, 0x25, 0x27, 0x00, 0xb2, 0xd6, 0xbd, 0xc1, 0xf0, 0x84, 0x7a, 0x12, 0x77, 0x27, 0xc0, 0xd3, 0xbe, 0x4f, 0xe6, 0x42, 0x7a, 0x61, 0xb6, 0x31, 0x44, 0x77, 0xcc, 0xfa, 0x21, 0x25, 0x17, 0xe0, 0x65, 0xbc, 0x66, 0x18, 0x76, 0xc1, 0x34, 0x4c, 0x92, 0x02, 0xfe, 0x52, 0xb4, 0x3f, 0x5f, 0xaa, 0xe5, 0x30, 0xb9, 0x27, 0x2a, 0x63, 0xdc, 0x4a, 0x28, 0xfd, 0x5a, 0x19, 0x7e, 0xb8, 0x84, 0xa4, 0x07, 0xdd, 0x34, 0x17, 0x78, 0x8f, 0xf3, 0xb9, 0xee, 0x7a, 0x29, 0x0d, 0x53, 0x2f, 0x42, 0xaa, 0xb8, 0x97, 0x7d, 0x2c, 0x2a, 0x83, 0x20, 0xba, 0x88, 0xa9, 0x35, 0x50, 0x94, 0xce, 0x1d, 0xa3, 0x76, 0x75, 0x9f, 0x0c, 0x8b, 0xc9, 0x03, 0x81, 0x9d, 0x43, 0x6d, 0x68, 0x89, 0xb9, 0x53, 0x79, 0xb8, 0xcc, 0xf0, 0x1e, 0x50, 0xd2, 0x77, 0xd5, 0x71, 0xfe, 0xa0, 0x2c, 0x8c, 0xd4, 0x19, 0xe8, 0x8e, 0xeb, 0xe3, 0xca, 0xa [... cut ...]
      - Hashes     - [NFCPassportReader.DataGroupId.DG1: "2C3BFF5F63EBE095A5F9E16C23C1D3792F7571635A56F2C1AB693E2D051E63DA", NFCPassportReader.DataGroupId.DG14: "06CAA3C87D4CF49923140662BC4565E0A578DA2F5717363869D30B9E9C7FB80F", NFCPassportReader.DataGroupId.DG2: "73A88BDC89B7215D824174D67C53DF6B86E32AC9D262E67EFB2CA9F99A2F3F29", NFCPassportReader.DataGroupId.DG7: "47636EFC50B4AEDC7D74C1DBFDA03F5B860EE107D0A4718BEA248C99AE843D42", NFCPassportReader.DataGroupId.DG15: "F3FB8A3D6346B37293BBF5939ED7C04369B7F8B642A27B6BAFA979CABEF561B9"]
Passport passed Datagroup Tampering check

```

rbrouwer, 2024-02-16:
@harry-anderson, on [TagReader.swift#L55](https://github.com/AndyQ/NFCPassportReader/blob/84e11ec67a5c0d5ff54f918182fd4658801cc7f2/Sources/NFCPassportReader/TagReader.swift#L55) did you try to change expectedResponseLength to -1? Or did you not do that in the log above?

harry-anderson, 2024-02-16:
@rbrouwer Above I had not made the change. I changed this line as you mention:
```
 let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x22, p1Parameter: 0x41, p2Parameter: 0xA6, data: data, expectedResponseLength: -1)
````
Here is the log with that change:
```
Performing Active Authentication
Generated Active Authentication challange - 1F30EFE7EBEA8705
TagReader - sending [0x00, 0x88, 0x00, 0x00, 0x08, 0x1F, 0x30, 0xEF, 0xE7, 0xEB, 0xEA, 0x87, 0x05, 0x00]
		SSC: 00000000000001C0
	Increment SSC with 1
		SSC: 00000000000001C1
Mask class byte and pad command header
	CmdHeader: 0C880000800000000000000000000000
Pad data
	Data: 1F30EFE7EBEA87058000000000000000
Encrypt data with KSenc
	EncryptedData: E2F6419149F7D093BBCEB58749477363
Build DO'87
	DO87: 871101E2F6419149F7D093BBCEB58749477363
Build DO'97
	DO97: [151, 1, 0]
Concatenate CmdHeader and DO87 and DO97
	M: 0C880000800000000000000000000000871101E2F6419149F7D093BBCEB58749477363970100
Compute MAC of M
	Concatenate SSC and M and add padding
		N: 000000000000000000000000000001C10C880000800000000000000000000000871101E2F6419149F7D093BBCEB5874947736397010080000000000000000000
aesMac - mac - 99C0697B6806753B0ED3097A517FF63700000000000000000000000000000000
	Compute MAC over N with KSmac
		CC: 99C0697B6806753B
Build DO'8E
	DO8E: 8E0899C0697B6806753B
Construct and send protected APDU
	ProtectedAPDU: 0C88000020871101E2F6419149F7D093BBCEB587494773639701008E0899C0697B6806753B00
TagReader - [SM] [0x0C, 0x88, 0x00, 0x00, 0x20, 0x87, 0x11, 0x01, 0xE2, 0xF6, 0x41, 0x91, 0x49, 0xF7, 0xD0, 0x93, 0xBB, 0xCE, 0xB5, 0x87, 0x49, 0x47, 0x73, 0x63, 0x97, 0x01, 0x00, 0x8E, 0x08, 0x99, 0xC0, 0x69, 0x7B, 0x68, 0x06, 0x75, 0x3B, 0x00]
TagReader - Received response, size 242b
	Increment SSC with 1
		SSC: 00000000000001C2
Receive response APDU of MRTD's chip
	RAPDU: 8781E1015230E4E1E8A6FD2F772BCF3CA602D2762D13690AC01AF125468389C75D2892F7E950B01A074346A1991DD4DDB8CED242CAD822A8DBAF8B63DC700DC8B808DA9F8091351D2C21D9FA7C322DD1E7DBF0A3D715EFC25A3C6E7A2DCE48D193C591DC51CBB6639323B6C977E8D6FB1E0C5C4388E411DA9DF43DA09EC46F8BED81E6EAD14521329A8381531788497CDF11A0498D173257E76A6123977E458596384A491F02711B0FB31E0B3F5369CA74D5AD40FB1C08FAD078FE5FA6955D6388580CF36DC28838408FDA088B5A478A6B2CFA841B91F71BB345DCB1179E39CE1FC2630E990290008E0897BEEB7A317DEDEF9000
Verify RAPDU CC by computing MAC of  DO'87 DO'99
	Concatenate SSC and  DO'87 DO'99 and add padding
		K: 000000000000000000000000000001C28781E1015230E4E1E8A6FD2F772BCF3CA602D2762D13690AC01AF125468389C75D2892F7E950B01A074346A1991DD4DDB8CED242CAD822A8DBAF8B63DC700DC8B808DA9F8091351D2C21D9FA7C322DD1E7DBF0A3D715EFC25A3C6E7A2DCE48D193C591DC51CBB6639323B6C977E8D6FB1E0C5C4388E411DA9DF43DA09EC46F8BED81E6EAD14521329A8381531788497CDF11A0498D173257E76A6123977E458596384A491F02711B0FB31E0B3F5369CA74D5AD40FB1C08FAD078FE5FA6955D6388580CF36DC28838408FDA088B5A478A6B2CFA841B91F71BB345DCB1179E39CE1FC2630E990290008000000000000000
	Compute MAC with KSmac
aesMac - mac - 97BEEB7A317DEDEF489B9F39FE8E3E9B00000000000000000000000000000000
		CC: 97BEEB7A317DEDEF
	Compare CC with data of DO'8E of RAPDU
		97BEEB7A317DEDEF  == 97BEEB7A317DEDEF ? true
Decrypt data of DO'87 with KSenc
	DecryptedData: 4FE982991C670D20A01AAC7F8AA592E686CC4079BA2799082451CB34D20D4DA9A40325CA921FAB10604DD3F2BAC7C164A125D854BEB2DC19E2690435A2FC79262C8C891026235052F349BCA33F6E2A9055799EAA0A560BEBE85DBFDAF304E44C98375743578D971570D3D0FB2BB783D3AAFED4823D43E69F6DE99D41A734773B40B537AA17E003F9A80F44D931BE8036E5BB228061735A7B5464FF663FBBB08705C7DC51B91D7E59F9EB6A387FBA74BCD9C89EFEEC969A77A13420233EB0E550A0A1DDAFE77A602F02F62D261FAF9F94C1D5AA833C8DDF9795DBCD573EC14A
Unprotected APDU: [4FE982991C670D20A01AAC7F8AA592E686CC4079BA2799082451CB34D20D4DA9A40325CA921FAB10604DD3F2BAC7C164A125D854BEB2DC19E2690435A2FC79262C8C891026235052F349BCA33F6E2A9055799EAA0A560BEBE85DBFDAF304E44C98375743578D971570D3D0FB2BB783D3AAFED4823D43E69F6DE99D41A734773B40B537AA17E003F9A80F44D931BE8036E5BB228061735A7B5464FF663FBBB08705C7DC51B91D7E59F9EB6A387FBA74BCD9C89EFEEC969A77A13420233EB0E550A0A1DDAFE77A602F02F62D261FAF9F94C1D5AA833C8DDF9795DBCD573EC14A] 90 00
TagReader [SM - unprotected] [0x4f, 0xe9, 0x82, 0x99, 0x1c, 0x67, 0x0d, 0x20, 0xa0, 0x1a, 0xac, 0x7f, 0x8a, 0xa5, 0x92, 0xe6, 0x86, 0xcc, 0x40, 0x79, 0xba, 0x27, 0x99, 0x08, 0x24, 0x51, 0xcb, 0x34, 0xd2, 0x0d, 0x4d, 0xa9, 0xa4, 0x03, 0x25, 0xca, 0x92, 0x1f, 0xab, 0x10, 0x60, 0x4d, 0xd3, 0xf2, 0xba, 0xc7, 0xc1, 0x64, 0xa1, 0x25, 0xd8, 0x54, 0xbe, 0xb2, 0xdc, 0x19, 0xe2, 0x69, 0x04, 0x35, 0xa2, 0xfc, 0x79, 0x26, 0x2c, 0x8c, 0x89, 0x10, 0x26, 0x23, 0x50, 0x52, 0xf3, 0x49, 0xbc, 0xa3, 0x3f, 0x6e, 0x2a, 0x90, 0x55, 0x79, 0x9e, 0xaa, 0x0a, 0x56, 0x0b, 0xeb, 0xe8, 0x5d, 0xbf, 0xda, 0xf3, 0x04, 0xe4, 0x4c, 0x98, 0x37, 0x57, 0x43, 0x57, 0x8d, 0x97, 0x15, 0x70, 0xd3, 0xd0, 0xfb, 0x2b, 0xb7, 0x83, 0xd3, 0xaa, 0xfe, 0xd4, 0x82, 0x3d, 0x43, 0xe6, 0x9f, 0x6d, 0xe9, 0x9d, 0x41, 0xa7, 0x34, 0x77, 0x3b, 0x40, 0xb5, 0x37, 0xaa, 0x17, 0xe0, 0x03, 0xf9, 0xa8, 0x0f, 0x44, 0xd9, 0x31, 0xbe, 0x80, 0x36, 0xe5, 0xbb, 0x22, 0x80, 0x61, 0x73, 0x5a, 0x7b, 0x54, 0x64, 0xff, 0x66, 0x3f, 0xbb, 0xb0, 0x87, 0x05, 0xc7, 0xdc, 0x51, 0xb9, 0x1d, 0x7e, 0x59, 0xf9, 0xeb, 0x6a, 0x38, 0x7f, 0xba, 0x74, 0xbc, 0xd9, 0xc8, 0x9e, 0xfe, 0xec, 0x96, 0x9a, 0x77, 0xa1, 0x34, 0x20, 0x23, 0x3e, 0xb0, 0xe5, 0x50, 0xa0, 0xa1, 0xdd, 0xaf, 0xe7, 0x7a, 0x60, 0x2f, 0x02, 0xf6, 0x2d, 0x26, 0x1f, 0xaf, 0x9f, 0x94, 0xc1, 0xd5, 0xaa, 0x83, 0x3c, 0x8d, 0xdf, 0x97, 0x95, 0xdb, 0xcd, 0x57, 0x3e, 0xc1, 0x4 [... cut ...]
   signature - 4FE982991C670D20A01AAC7F8AA592E686CC4079BA2799082451CB34D20D4DA9A40325CA921FAB10604DD3F2BAC7C164A125D854BEB2DC19E2690435A2FC79262C8C891026235052F349BCA33F6E2A9055799EAA0A560BEBE85DBFDAF304E44C98375743578D971570D3D0FB2BB783D3AAFED4823D43E69F6DE99D41A734773B40B537AA17E003F9A80F44D931BE8036E5BB228061735A7B5464FF663FBBB08705C7DC51B91D7E59F9EB6A387FBA74BCD9C89EFEEC969A77A13420233EB0E550A0A1DDAFE77A602F02F62D261FAF9F94C1D5AA833C8DDF9795DBCD573EC14A
Error identifying Active Authentication RSA message digest hash algorithm
-[NFCTagReaderSession setAlertMessage:]:101 (null)
Parse SOD - Using Algo - SHA256
```
Thanks

AndyQ, 2024-02-29:
@harry-anderson  sorry for the delay - that looks like you changed the wrong line - looks like uyou changed the on in the sendMSEKAT function - rather than the one in the doInternalAuthentication function.

Could you check?

harry-anderson, 2024-03-01:
@AndyQ all good, and sorry, I had indeed changed the wrong line. I fixed it now, changed `doInternalAuthentication` instead :
```swift
 func doInternalAuthentication( challenge: [UInt8] ) async throws -> ResponseAPDU {
        let randNonce = Data(challenge)
        
        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: -1)

        return try await send( cmd: cmd )
    }
```
Here is the log:
```
Performing Active Authentication
Generated Active Authentication challange - 2BBD76A0B051337C
TagReader - sending [0x00, 0x88, 0x00, 0x00, 0x08, 0x2B, 0xBD, 0x76, 0xA0, 0xB0, 0x51, 0x33, 0x7C]
		SSC: 00000000000001C0
	Increment SSC with 1
		SSC: 00000000000001C1
Mask class byte and pad command header
	CmdHeader: 0C880000800000000000000000000000
Pad data
	Data: 2BBD76A0B051337C8000000000000000
Encrypt data with KSenc
	EncryptedData: 6DCAD52F11E0488F2075CDA2D0FE5536
Build DO'87
	DO87: 8711016DCAD52F11E0488F2075CDA2D0FE5536
Concatenate CmdHeader and DO87
	M: 0C8800008000000000000000000000008711016DCAD52F11E0488F2075CDA2D0FE5536
Compute MAC of M
	Concatenate SSC and M and add padding
		N: 000000000000000000000000000001C10C8800008000000000000000000000008711016DCAD52F11E0488F2075CDA2D0FE553680000000000000000000000000
aesMac - mac - 39BB3FF3B246C5323CB91DA4018A39EC00000000000000000000000000000000
	Compute MAC over N with KSmac
		CC: 39BB3FF3B246C532
Build DO'8E
	DO8E: 8E0839BB3FF3B246C532
Construct and send protected APDU
	ProtectedAPDU: 0C8800001D8711016DCAD52F11E0488F2075CDA2D0FE55368E0839BB3FF3B246C53200
TagReader - [SM] [0x0C, 0x88, 0x00, 0x00, 0x1D, 0x87, 0x11, 0x01, 0x6D, 0xCA, 0xD5, 0x2F, 0x11, 0xE0, 0x48, 0x8F, 0x20, 0x75, 0xCD, 0xA2, 0xD0, 0xFE, 0x55, 0x36, 0x8E, 0x08, 0x39, 0xBB, 0x3F, 0xF3, 0xB2, 0x46, 0xC5, 0x32, 0x00]
TagReader - Received response, size 14b
	Increment SSC with 1
		SSC: 00000000000001C2
Receive response APDU of MRTD's chip
	RAPDU: 990290008E08C63EC17BE403C8DF9000
Verify RAPDU CC by computing MAC of  DO'99
	Concatenate SSC and  DO'99 and add padding
		K: 000000000000000000000000000001C299029000800000000000000000000000
	Compute MAC with KSmac
aesMac - mac - C63EC17BE403C8DFDE259D37D1A1214100000000000000000000000000000000
		CC: C63EC17BE403C8DF
	Compare CC with data of DO'8E of RAPDU
		C63EC17BE403C8DF  == C63EC17BE403C8DF ? true
Unprotected APDU: [] 90 00
TagReader [SM - unprotected] [], sw1:0x90 sw2:0x00
Active Authentication
   challange - 2BBD76A0B051337C
   signature -
Error identifying Active Authentication RSA message digest hash algorithm
-[NFCTagReaderSession setAlertMessage:]:101 (null)
Parse SOD - Using Algo - SHA256
```

AndyQ, 2024-03-04:
OK, so thats out then, looks like it didn't even get the signature back this time (which is odd esp as no error).

Am a little stumped at the moment.

tomgi, 2024-04-06:
Hi @AndyQ @harry-anderson,

We're recently run into the same issue and together with @donbobka we've made some discoveries that we hope will help in solving this issue 🙂 

Australian R series passport:
- uses a 256 byte RSA key for AA, so the expected AA signature length is also exactly 256 bytes
- we're only getting 231 bytes back in AA response after unwrapping from the Secure Messaging envelope
- the chip doesn't seem to support response chaining - we're getting `0x90 0x00` status, which indicates success and no remaining data available - the [fix](https://github.com/AndyQ/NFCPassportReader/pull/201) for Finnish passports doesn't apply here - apparently the Australian chip just silently truncates the response here
- `OpenSSLUtils.decryptRSASignature` called on truncated data is effectively random bytes, so we fail to identify the digest algorithm based on decrypted signature trailer
- we've managed to get AA working by forcing extended length APDU for AA request (both nested and Secure Messaging envelope) and response:
```diff
diff --git a/Sources/NFCPassportReader/SecureMessaging.swift b/Sources/NFCPassportReader/SecureMessaging.swift
index 575185a..765745b 100644
--- a/Sources/NFCPassportReader/SecureMessaging.swift
+++ b/Sources/NFCPassportReader/SecureMessaging.swift
@@ -88,7 +88,7 @@ public class SecureMessaging {
         // otherwise its a single byte of size
         let size = do87.count + do97.count + do8e.count
         var dataSize: [UInt8]
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 256 { // possibly should be apdu.expectedResponseLength > 231 to account for SM envelope
             dataSize = [0x00] + intToBin(size, pad: 4)
         } else {
             dataSize = intToBin(size)
@@ -98,7 +98,7 @@ public class SecureMessaging {
 
         // If the data is more that 255, specify the we are using extended length (0x00, 0x00)
         // Thanks to @filom for the fix!
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 256 { // possibly should be apdu.expectedResponseLength > 231 to account for SM envelope
             protectedAPDU += [0x00,0x00]
         } else {
             protectedAPDU += [0x00]
diff --git a/Sources/NFCPassportReader/TagReader.swift b/Sources/NFCPassportReader/TagReader.swift
index 8791622..ac8366b 100644
--- a/Sources/NFCPassportReader/TagReader.swift
+++ b/Sources/NFCPassportReader/TagReader.swift
@@ -52,7 +52,7 @@ public class TagReader {
     func doInternalAuthentication( challenge: [UInt8] ) async throws -> ResponseAPDU {
         let randNonce = Data(challenge)
 
-        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 256)
+        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 65536)
 
         return try await send( cmd: cmd )
     }
```

- notice that we had to make the change in both places inside `protect`, despite the fact that we're not sending more than 256 bytes, but only expecting to receive more than 256 bytes back (our guess is that we must use extended APDU format for both request and response together)
- our fix breaks AA for some older Russian passport that don't support extended length APDU at all, so we're not going to submit that PoC as a PR 🙂 

We used to have the same issue on android using jmrtd library and it was just recently fixed in https://sourceforge.net/p/jmrtd/code/1878/ - seems like they try to predict the AA signature size based on the AA public key from DG15.

We might submit a PR with a proper fix based on jmrtd's approach in the following days/weeks, but we would also be very happy for you to do it, since we don't have access to too many passport to test against and our understanding of the ICAO standards and NFC tech is also quite limited 😅

harry-anderson, 2024-04-12:
Hi @tomgi and cc @AndyQ 

I tried out the changes mentioned by @tomgi and AA check now succeeds! 🥳 

Changes:
```diff
index 575185a..958b391 100644
--- a/Sources/NFCPassportReader/SecureMessaging.swift
+++ b/Sources/NFCPassportReader/SecureMessaging.swift
@@ -88,7 +88,7 @@ public class SecureMessaging {
         // otherwise its a single byte of size
         let size = do87.count + do97.count + do8e.count
         var dataSize: [UInt8]
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 256 {
             dataSize = [0x00] + intToBin(size, pad: 4)
         } else {
             dataSize = intToBin(size)
@@ -98,7 +98,7 @@ public class SecureMessaging {
             
         // If the data is more that 255, specify the we are using extended length (0x00, 0x00)
         // Thanks to @filom for the fix!
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 256 {
             protectedAPDU += [0x00,0x00]
         } else {
             protectedAPDU += [0x00]
diff --git a/Sources/NFCPassportReader/TagReader.swift b/Sources/NFCPassportReader/TagReader.swift
index 8791622..2212a41 100644
--- a/Sources/NFCPassportReader/TagReader.swift
+++ b/Sources/NFCPassportReader/TagReader.swift
@@ -52,7 +52,7 @@ public class TagReader {
     func doInternalAuthentication( challenge: [UInt8] ) async throws -> ResponseAPDU {
         let randNonce = Data(challenge)
         
-        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 256)
+        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 65536)
 
         return try await send( cmd: cmd )
     }
```

Snippet from the logs:
```
...
Unprotected APDU: [494C295336C612703DFD55D58C5581CF897466B8566C7D16272D91177421BF39E4FF9DE5E43C4DB33A94515C7CCBD653BEAA8ACF93A90EE8F12993025E9128D03B850A15D8D93D59126FCFD829D9961401A88558ABBF35C91BB9B65F28D74B9D067BB4190DAA6B4F421050ED566A67A9DADE0D865BF77A73A8279D0C72D904419334087079B8AD83DF27D9A4F7E4F4452A7CC31E5C1962635BF8C95081FEB892A18CE940B7C1A43FADA3FD712A5EEA9474B455E8CE3DFD8E6C07B31529F8B59F9DF6B02D0363C9BCD067F6648A1F35061A75D372C75A25962E48B7D8B99F35045128B8BC55C8B4328AF3FA047BA4D8C4112609A2854A100DC865FBE6790DA175] 90 00
TagReader [SM - unprotected] [0x49, 0x4c, 0x29, 0x53, 0x36, 0xc6, 0x12, 0x70, 0x3d, 0xfd, 0x55, 0xd5, 0x8c, 0x55, 0x81, 0xcf, 0x89, 0x74, 0x66, 0xb8, 0x56, 0x6c, 0x7d, 0x16, 0x27, 0x2d, 0x91, 0x17, 0x74, 0x21, 0xbf, 0x39, 0xe4, 0xff, 0x9d, 0xe5, 0xe4, 0x3c, 0x4d, 0xb3, 0x3a, 0x94, 0x51, 0x5c, 0x7c, 0xcb, 0xd6, 0x53, 0xbe, 0xaa, 0x8a, 0xcf, 0x93, 0xa9, 0x0e, 0xe8, 0xf1, 0x29, 0x93, 0x02, 0x5e, 0x91, 0x28, 0xd0, 0x3b, 0x85, 0x0a, 0x15, 0xd8, 0xd9, 0x3d, 0x59, 0x12, 0x6f, 0xcf, 0xd8, 0x29, 0xd9, 0x96, 0x14, 0x01, 0xa8, 0x85, 0x58, 0xab, 0xbf, 0x35, 0xc9, 0x1b, 0xb9, 0xb6, 0x5f, 0x28, 0xd7, 0x4b, 0x9d, 0x06, 0x7b, 0xb4, 0x19, 0x0d, 0xaa, 0x6b, 0x4f, 0x42, 0x10, 0x50, 0xed, 0x56, 0x6a, 0x67, 0xa9, 0xda, 0xde, 0x0d, 0x86, 0x5b, 0xf7, 0x7a, 0x73, 0xa8, 0x27, 0x9d, 0x0c, 0x72, 0xd9, 0x04, 0x41, 0x93, 0x34, 0x08, 0x70, 0x79, 0xb8, 0xad, 0x83, 0xdf, 0x27, 0xd9, 0xa4, 0xf7, 0xe4, 0xf4, 0x45, 0x2a, 0x7c, 0xc3, 0x1e, 0x5c, 0x19, 0x62, 0x63, 0x5b, 0xf8, 0xc9, 0x50, 0x81, 0xfe, 0xb8, 0x92, 0xa1, 0x8c, 0xe9, 0x40, 0xb7, 0xc1, 0xa4, 0x3f, 0xad, 0xa3, 0xfd, 0x71, 0x2a, 0x5e, 0xea, 0x94, 0x74, 0xb4, 0x55, 0xe8, 0xce, 0x3d, 0xfd, 0x8e, 0x6c, 0x07, 0xb3, 0x15, 0x29, 0xf8, 0xb5, 0x9f, 0x9d, 0xf6, 0xb0, 0x2d, 0x03, 0x63, 0xc9, 0xbc, 0xd0, 0x67, 0xf6, 0x64, 0x8a, 0x1f, 0x35, 0x06, 0x1a, 0x75, 0xd3, 0x72, 0xc7, 0x5a, 0x25, 0x96, 0x2e, 0x48, 0xb7, 0xd8, 0xb9, 0x9f, 0x35, 0x04, 0x51, 0x28, 0xb8, 0xbc, 0x55, 0xc8, 0xb4, 0x32, 0x8a, 0xf3, 0xfa, 0x04, 0x7b, 0xa4, 0xd8, 0xc4, 0x11, 0x26, 0x09, 0xa2, 0x85, 0x4a, 0x10, 0x0d, 0xc8, 0x65, 0xfb, 0xe6, 0x79, 0x0d, 0xa1, 0x75, ], sw1:0x90 sw2:0x00
Active Authentication
   challange - 18FE8850E870EB88
   signature - 494C295336C612703DFD55D58C5581CF897466B8566C7D16272D91177421BF39E4FF9DE5E43C4DB33A94515C7CCBD653BEAA8ACF93A90EE8F12993025E9128D03B850A15D8D93D59126FCFD829D9961401A88558ABBF35C91BB9B65F28D74B9D067BB4190DAA6B4F421050ED566A67A9DADE0D865BF77A73A8279D0C72D904419334087079B8AD83DF27D9A4F7E4F4452A7CC31E5C1962635BF8C95081FEB892A18CE940B7C1A43FADA3FD712A5EEA9474B455E8CE3DFD8E6C07B31529F8B59F9DF6B02D0363C9BCD067F6648A1F35061A75D372C75A25962E48B7D8B99F35045128B8BC55C8B4328AF3FA047BA4D8C4112609A2854A100DC865FBE6790DA175
Active Authentication (RSA) successful

```

harry-anderson, 2024-04-12:
@tomgi for completeness I also tried setting `apdu.expectedResponseLength > 231`. Which also works!

```diff
--- a/Sources/NFCPassportReader/SecureMessaging.swift
+++ b/Sources/NFCPassportReader/SecureMessaging.swift
@@ -88,7 +88,7 @@ public class SecureMessaging {
         // otherwise its a single byte of size
         let size = do87.count + do97.count + do8e.count
         var dataSize: [UInt8]
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 231 {
             dataSize = [0x00] + intToBin(size, pad: 4)
         } else {
             dataSize = intToBin(size)
@@ -98,7 +98,7 @@ public class SecureMessaging {
             
         // If the data is more that 255, specify the we are using extended length (0x00, 0x00)
         // Thanks to @filom for the fix!
-        if size > 255 {
+        if size > 255 || apdu.expectedResponseLength > 231 {
             protectedAPDU += [0x00,0x00]
         } else {
             protectedAPDU += [0x00]
diff --git a/Sources/NFCPassportReader/TagReader.swift b/Sources/NFCPassportReader/TagReader.swift
index 8791622..2212a41 100644
--- a/Sources/NFCPassportReader/TagReader.swift
+++ b/Sources/NFCPassportReader/TagReader.swift
@@ -52,7 +52,7 @@ public class TagReader {
     func doInternalAuthentication( challenge: [UInt8] ) async throws -> ResponseAPDU {
         let randNonce = Data(challenge)
         
-        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 256)
+        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 65536)
 
         return try await send( cmd: cmd )
     }
```
Logs:
```
Active Authentication
   challange - FA339125D84D5277
   signature - 4E068EF76444389FE16AFF1A465049986DCE8FB41034C1441633D219C90405ADDA7F4F4393C09DC349E86C2F2D1ABF6E3D6C433F44F17D2B96B64FC998CD5EFF59254E1C2BBDFF498FDBCE952AE1302B3669C3F0A13D64719A048B4F05DB807E500263440D39878BB38A65FA78E787140AEB9866100B9242172F8A23DB2E289B408A06F9F3B8B80329AF1F1B7589F6F5C13D50B350BCECBA88F6D4826D4F29474F8A885877F760D71D4FB6EEC6CF45093336A03F2598FF76C966CE3101D39A866B4181CE596C1D69DD38AE96064EFBC67D0B87843BB1F5583082D46D0F971375E483C8B3A4E5FBD9D6A568142BAC4B5AD7F3287CDFC800D5B02593E0521EB0E3
Active Authentication (RSA) successful
```
But If I remove the `|| apdu.expectedResponseLength > 231`, and only increase the `expectedResponseLength`, it fails e.g:

```diff
--- a/Sources/NFCPassportReader/TagReader.swift
+++ b/Sources/NFCPassportReader/TagReader.swift
@@ -52,7 +52,7 @@ public class TagReader {
     func doInternalAuthentication( challenge: [UInt8] ) async throws -> ResponseAPDU {
         let randNonce = Data(challenge)
         
-        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 256)
+        let cmd = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 65536)
 
         return try await send( cmd: cmd )
     }
```
Logs:
```
Active Authentication
   challange - 30E854AC488CE6E2
   signature - 17AB497356761EBAF3B48681FBFECD12E9502364802AD7B10AAA9B6086AA7A18EC6D6DFEFCB6CC29C16EC872B8EBB4A3B75C3F5C2299C0041F4A02238EC31DB8421FC1C8BCC868489946A452638DE1FA680ECAB4B27EE2A0BA5D86C3377865E72EBA662F62E3B8CCDCA94C79C39F80F867D449A3953F1F0101F033CD3F503D76CE5B1649B92FED52BA1A63C14BEB61E3659C4ACF16EC03963D52F08F165FBE879772C771004CA4B00614D259F37DD21C9EA115F6DE50D41514F783EF3B5EC54ACF6C26FA3A747841167E98768F571F840C3BB6616432496D15B7D1820AD7DB
Error identifying Active Authentication RSA message digest hash algorithm
```

Hope that helps

AndyQ, 2024-04-12:
Thats great news, I'll test out across my dev passports hopefully today (work has got in the way this week!) and see if thats a compatible change.

AndyQ, 2024-04-12:
That tested fine on all my test passports (going back to expiring in 2015).  I've committed the change to main - but not yet done a release.  If you could please check out the main branch and confirm it still works for you that would be great.

I think I'm willing to push this up even with the slight risk it breaks with old passports.

Thanks @tomgi, @donbobka, @harry-anderson for the fix and testing

harry-anderson, 2024-04-13:
I tested it out with the latest commit `ce3a9d436` on the main branch and AA check is still successful with my passport. Thanks again for your help @AndyQ 🙏 !!

rbrouwer, 2024-05-23:
Sadly the current fix does break AA for Chinese passports (recent models). I believe a solution such as in JMRTD might be needed given you cannot just expect extended format to just work for all.

AndyQ, 2024-05-29:
Re-opening this - will take a look when I get a chance

rbrouwer, 2024-05-30:
> Re-opening this - will take a look when I get a chance

I had made #223 for that. Feel free to close either. Duplicates are not helpful.

AndyQ, 2024-05-31:
OK - will move this over to 223

AndyQ, 2024-07-03:
One thing, I've just change things slightly so that extended mode isn't the default as this breaks other passports. 

I've still some more investigation to do but for the moment I've added a new flag to PassportReader - useExtendedMode. This defaults to false if not specified. 

Currently in main branch only at the moment.

closed 2026-11-09 by AndyQ