AndyQ/NFCPassportReader #197: German passport BAC failed. Referenced data not found

nicop2000, 2023-12-14:
I tried your sample app and it won't work on my German passport no matter which settings I use.

Without Passive Authentication:
2023-12-14 21:38:09.709123+0100 NFCPassportReaderApp[3571:1650949] [ShareSheet] connection invalidated
2023-12-14 21:38:10.532383+0100 NFCPassportReaderApp[3571:1648763] [app] Using version 1.0.1
2023-12-14 21:38:13.632541+0100 NFCPassportReaderApp[3571:1651244] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
Error reading tag: sw1 - 0x67, sw2 - 0x00
reason: Wrong length
Error reading tag: sw1 - 0x67, sw2 - 0x00
reason: Wrong length
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 0x6A, sw2 - 0x88
reason: Referenced data not found
ERROR - Referenced data not found
BAC Failed

With Passive Authentication:
2023-12-14 21:39:04.408148+0100 NFCPassportReaderApp[3571:1648763] [app] Using version 1.0.1
2023-12-14 21:39:05.866404+0100 NFCPassportReaderApp[3571:1651951] [CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
Error reading tag: sw1 - 0x67, sw2 - 0x00
reason: Wrong length
Error reading tag: sw1 - 0x67, sw2 - 0x00
reason: Wrong length
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 0x6A, sw2 - 0x88
reason: Referenced data not found
ERROR - Referenced data not found
BAC Failed

smlu, 2023-12-14:
The error suggests that passport doesn't support BAC protocol. This is usually the case with German passports.
Try using PACE protocol.

nicop2000, 2023-12-14:
Okay, so you have a feature in your sample app? If not I'll just need to know, so I'll write my one ^^

nicop2000, 2023-12-14:
But according to the federal office for information security BAC is used in German e-passports.

PACE is only used for German ID cards.

nicop2000, 2023-12-21:
Seems to be an iPhone issue, since it now works with no changes made

closed 2023-12-21 by nicop2000