AndyQ/NFCPassportReader #199: Verification failure with .pem list

nicop2000, 2023-12-21:
I downloaded the icaopkd-002-complete-000244.ldif from icao, used the extract.py script on it, copied the list into my project and wanted to verify a passport.
This is when I got an 'Verification of certificate failed' with error code 25 which is X509_V_ERR_PATH_LENGTH_EXCEEDED.
Any idea why this comes up or how to fix it? I'm not an expert with certs.
Thanks in advance :)

trungnl08, 2024-01-30:
Did you fix this issue ? I have the same problem but cannot handle after many days :)

AndyQ, 2024-02-13:
Interesting. I just downloaded the latest icaopkd-002-complete-000252.ldif and ran that through and it completed fine with no errors.

Would you be able to either send me a copy of the 244.ldif file (and I'll check it) or try the latest version from the icao site?

nicop2000, 2024-02-14:
@AndyQ 
We're talking about this site, right? [ICAO](https://download.pkd.icao.int/download)

I downloaded this file icaopkd-002-delta-000252.ldif under "The latest collection of CSCA Master Lists." with 7.7 MB size. This still fails. I set the Master List URL and the verification failed: OpenSSLError
  - UnableToVerifyX509CertificateForSOD : "Verification of certificate failed - errorCode 20"

nicop2000, 2024-02-26:
A fresh installation of python3 and resetting all Xcode caches solved it

closed 2024-02-26 by nicop2000