AndyQ/NFCPassportReader #202: Fails on chinese passport

iAbdul, 2024-02-01:
Hi, 
Code works good with Canadian and usa passport however it fails with Chinese passport and shows alert "unsupported data group". Can you guide me how to make it work on Chinese passport. 
Thanks!

TerjeLon, 2024-04-09:
Not all passports follow the ICAO standard regarding listing Datagroups in the COM's 5C tag. 
This should perhaps not give an error, as reading the passport itself should be fine.

Perhaps a solution here could be to return the PassportModel in the callback instead of an error, and add a "warning/errors" parameter to the Model so consumers can be notified of the deviations that has occurd.

TerjeLon, 2024-04-09:
> I have noticed this too. There can be 19 DataGroups but the app only supports to 16. Try and add DG17-19 with no parsing.

Theres no official documentation on those datagroups.

i will try to create a pr for this when i have the time

TerjeLon, 2024-04-10:
@iAbdul could you try again with the branch on the PR above and see if this fixes it for you?
@advatar did you have any documents you could try on? Testing is much appreciated as i dont have any documents on hand.

Please comment in the PR so that if/when @AndyQ eventually gets around to it he can see that the fix is verified.

iAbdul, 2024-09-21:
> @iAbdul could you try again with the branch on the PR above and see if this fixes it for you? @advatar did you have any documents you could try on? Testing is much appreciated as i dont have any [...]

@TerjeLon @AndyQ it seems working with skip tag. Thanks!!

PR #209 (2025-02-18): Skip unknown tags
Should resolve #202.

## Identified two possibilities as to why this may occur:

### Trying to read a non-supported (Unknown) datagroup.
I am struggling to see how this could even be possible as COM parse function already filters these out:

```swift
for v in vals {
    if let index = DataGroupParser.tags.firstIndex(of: v) {
        dataGroupsPresent.append( DataGroupParser.dataGroupNames[index] )
    }
}
```

Regardless, in such a scenario the reading should not fail, but just skip the datagroup and carry on. So i added a safeguard where this should now be the behaviour goin [... cut]
closed 2025-02-18 by AndyQ