AndyQ/NFCPassportReader #203: extract.py: Fails for `.ml` files

matthiasgeihs, 2024-02-15:
First of all: Great project!

I stumbled across a minor bug when reading `.ml` files with the `extract.py` script.

https://github.com/AndyQ/NFCPassportReader/blob/84e11ec67a5c0d5ff54f918182fd4658801cc7f2/scripts/extract.py#L29-L42

For `.ml` files, the variable `cns` is not set, and hence the script fails at line 37.

matthiasgeihs, 2024-02-29:
If I see it correctly, this fails for all `.ml` files. Concretely, I was testing with the [German CSCA Master List](https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/ElekAusweise/CSCA/GermanMasterList.html).

AndyQ, 2024-04-23:
I've just fixed this - its in main and will be in the next release

matthiasgeihs, 2024-04-24:
Feel free to close the issue whenever you feel like it. 🙂