AndyQ/NFCPassportReader #205: File not found ID Card

nicop2000, 2024-02-26:
When scanning ID cards which are older than 2022 I get a Referenced data not found but in the middle of the stacktrace is a file not found error. Funnily id cards from 2022 and 2023 work fine as well as passports.

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - found [CoreNFC.NFCTag.iso7816(<NFCISO7816Tag: 0x282f34940>)]
tagReaderSession:connected to tag - starting authentication
45[NFCTagReaderSession setAlertMessage:]:101
TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
TagReader - Received response, size 0b
TagReader [unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]
TagReader - Received response, size 0b
TagReader [unprotected] [], sw1:0x90 sw2:0x00
TagReader - sending [0x00, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x04]
TagReader - Received response, size 4b
TagReader [unprotected] [0x31, 0x81, 0xc1, 0x30, ], sw1:0x90 sw2:0x00
TagReader - Number of data bytes to read - 193
45[NFCTagReaderSession setAlertMessage:]:101
TagReader - data bytes remaining: 193, will read : 160
TagReader - sending [0x00, 0xB0, 0x00, 0x03, 0xA0]
TagReader - Received response, size 160b
TagReader [unprotected] [0x30, 0x0d, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x02, 0x02, 0x01, 0x02, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x02, 0x02, 0x01, 0x02, 0x02, 0x01, 0x41, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x04, 0x02, 0x02, 0x02, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x30, 0x1c, 0x06, 0x09, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x30, 0x0c, 0x06, 0x07, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x02, 0x01, 0x41, 0x30, 0x2a, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x06, 0x16, 0x1e, 0x68, 0x74, 0x74, 0x70, 0x3a, 0x2f, 0x2f, 0x62, 0x73, 0x69, 0x2e, 0x62, 0x75, 0x6e, 0x64, 0x2e, 0x64, 0x65, 0x2f, 0x63, 0x69, 0x66, 0x2f, 0x6e, 0x70, 0x61, 0x2e, 0x78, 0x6d, 0x6c, 0x30, 0x3e, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x08, 0x31, 0x32, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x02, 0x02, 0x01, 0x02, ], sw1:0x90 sw2:0x00
TagReader - got resp - [0x30, 0x0d, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x02, 0x02, 0x01, 0x02, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x02, 0x02, 0x01, 0x02, 0x02, 0x01, 0x41, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x04, 0x02, 0x02, 0x02, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x30, 0x1c, 0x06, 0x09, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x30, 0x0c, 0x06, 0x07, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x02, 0x01, 0x41, 0x30, 0x2a, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x06, 0x16, 0x1e, 0x68, 0x74, 0x74, 0x70, 0x3a, 0x2f, 0x2f, 0x62, 0x73, 0x69, 0x2e, 0x62, 0x75, 0x6e, 0x64, 0x2e, 0x64, 0x65, 0x2f, 0x63, 0x69, 0x66, 0x2f, 0x6e, 0x70, 0x61, 0x2e, 0x78, 0x6d, 0x6c, 0x30, 0x3e, 0x06, 0x08, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x08, 0x31, 0x32, 0x30, 0x12, 0x06, 0x0a, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x02, 0x02, 0x01, 0x02, ], sw1 : 144, sw2 : 0
TagReader - Amount of data left to read - 33
45[NFCTagReaderSession setAlertMessage:]:101
TagReader - data bytes remaining: 33, will read : 33
TagReader - sending [0x00, 0xB0, 0x00, 0xA3, 0x21]
TagReader - Received response, size 33b
TagReader [unprotected] [0x02, 0x01, 0x45, 0x30, 0x1c, 0x06, 0x09, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x30, 0x0c, 0x06, 0x07, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x02, 0x01, 0x45, ], sw1:0x90 sw2:0x00
TagReader - got resp - [0x02, 0x01, 0x45, 0x30, 0x1c, 0x06, 0x09, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x02, 0x02, 0x03, 0x02, 0x30, 0x0c, 0x06, 0x07, 0x04, 0x00, 0x7f, 0x00, 0x07, 0x01, 0x02, 0x02, 0x01, 0x0d, 0x02, 0x01, 0x45, ], sw1 : 144, sw2 : 0
TagReader - Amount of data left to read - 0
Read CardAccess - data 3181C1300D060804007F00070202020201023012060A04007F000702020302020201020201413012060A04007F0007020204020202010202010D301C060904007F000702020302300C060704007F0007010202010D020141302A060804007F0007020206161E687474703A2F2F6273692E62756E642E64652F6369662F6E70612E786D6C303E060804007F000702020831323012060A04007F00070202030202020102020145301C060904007F000702020302300C060704007F0007010202010D020145
Starting Password Authenticated Connection Establishment (PACE)
Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-128
doPace - inpit parameters
paceOID - 0.4.0.127.0.7.2.2.4.2.2
parameterSpec - 927
mappingType - Generic Mapping
agreementAlg - ECDH
cipherAlg - AES
digestAlg - SHA-1
keyLength - 128
keyLength - L1FHWZ7Y1702010472505026
paceKey - [0x90, 0xb3, 0xdb, 0x43, 0x96, 0xbe, 0x65, 0x67, 0x52, 0xa8, 0xdd, 0xc1, 0x60, 0x42, 0xce, 0x62, ]
TagReader - sending [0x00, 0x22, 0xC1, 0xA4, 0x0F, 0x80, 0x0A, 0x04, 0x00, 0x7F, 0x00, 0x07, 0x02, 0x02, 0x04, 0x02, 0x02, 0x83, 0x01, 0x01]
TagReader - Received response, size 0b
TagReader [unprotected] [], sw1:0x90 sw2:0x00
Doing PACE Step1...
TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x02, 0x7C, 0x00, 0x00]
TagReader - Received response, size 20b
TagReader [unprotected] [0x7c, 0x12, 0x80, 0x10, 0x4b, 0xa1, 0x19, 0x32, 0x51, 0xe3, 0xaf, 0x6c, 0x46, 0x7c, 0x40, 0x0a, 0x03, 0x7d, 0x2e, 0xd9, ], sw1:0x90 sw2:0x00
Encrypted nonce - [0x4b, 0xa1, 0x19, 0x32, 0x51, 0xe3, 0xaf, 0x6c, 0x46, 0x7c, 0x40, 0x0a, 0x03, 0x7d, 0x2e, 0xd9, ]
Decrypted nonce - [0x89, 0xdf, 0xd3, 0xab, 0xd4, 0xae, 0x58, 0xf4, 0x41, 0x89, 0xb1, 0x61, 0x0e, 0x5f, 0x3f, 0xac, ]
Doing PACE Step2...
   Using General Mapping (GM)...
Generating ECDH mapping keys from parameterSpec - 927
public mapping key - [0x04, 0x15, 0x94, 0xbb, 0x88, 0xf2, 0xff, 0xae, 0x92, 0x7c, 0x88, 0x88, 0xd1, 0x73, 0x07, 0x95, 0x0b, 0xba, 0x7d, 0xba, 0xf3, 0x1d, 0xb7, 0xe3, 0xf8, 0xa6, 0x48, 0x83, 0xb1, 0x49, 0xdd, 0x19, 0x9a, 0x02, 0x5a, 0x86, 0xcd, 0x81, 0x82, 0x71, 0x0a, 0x79, 0xa3, 0x55, 0x0f, 0x96, 0xaa, 0x71, 0x6a, 0xd8, 0x8f, 0xf1, 0x34, 0x90, 0x56, 0xaf, 0x08, 0x3f, 0xa6, 0x6c, 0x3a, 0x5a, 0xea, 0xf7, 0x92, ]
Sending public mapping key to passport..
TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x45, 0x7C, 0x43, 0x81, 0x41, 0x04, 0x15, 0x94, 0xBB, 0x88, 0xF2, 0xFF, 0xAE, 0x92, 0x7C, 0x88, 0x88, 0xD1, 0x73, 0x07, 0x95, 0x0B, 0xBA, 0x7D, 0xBA, 0xF3, 0x1D, 0xB7, 0xE3, 0xF8, 0xA6, 0x48, 0x83, 0xB1, 0x49, 0xDD, 0x19, 0x9A, 0x02, 0x5A, 0x86, 0xCD, 0x81, 0x82, 0x71, 0x0A, 0x79, 0xA3, 0x55, 0x0F, 0x96, 0xAA, 0x71, 0x6A, 0xD8, 0x8F, 0xF1, 0x34, 0x90, 0x56, 0xAF, 0x08, 0x3F, 0xA6, 0x6C, 0x3A, 0x5A, 0xEA, 0xF7, 0x92, 0x00]
TagReader - Received response, size 69b
TagReader [unprotected] [0x7c, 0x43, 0x82, 0x41, 0x04, 0x13, 0xd8, 0xaf, 0xd8, 0x16, 0x6e, 0x3d, 0xc1, 0x3e, 0x87, 0x51, 0xa2, 0xf5, 0x43, 0xe7, 0x45, 0xb6, 0xda, 0xf6, 0xea, 0x53, 0xd4, 0xe0, 0xf8, 0xc7, 0xc9, 0x35, 0x9e, 0x31, 0x5e, 0x84, 0x08, 0xa9, 0x27, 0x0a, 0xc6, 0x8e, 0x17, 0x4b, 0xe8, 0xc6, 0xb2, 0x7b, 0xb9, 0x5b, 0x9a, 0x53, 0x47, 0x88, 0x2a, 0x74, 0x0d, 0xf1, 0xff, 0x2a, 0x45, 0xae, 0x6a, 0x72, 0x5f, 0xaf, 0x54, 0x13, 0x93, ], sw1:0x90 sw2:0x00
Received passports public mapping key
   public mapping key - [0x04, 0x13, 0xd8, 0xaf, 0xd8, 0x16, 0x6e, 0x3d, 0xc1, 0x3e, 0x87, 0x51, 0xa2, 0xf5, 0x43, 0xe7, 0x45, 0xb6, 0xda, 0xf6, 0xea, 0x53, 0xd4, 0xe0, 0xf8, 0xc7, 0xc9, 0x35, 0x9e, 0x31, 0x5e, 0x84, 0x08, 0xa9, 0x27, 0x0a, 0xc6, 0x8e, 0x17, 0x4b, 0xe8, 0xc6, 0xb2, 0x7b, 0xb9, 0x5b, 0x9a, 0x53, 0x47, 0x88, 0x2a, 0x74, 0x0d, 0xf1, 0xff, 0x2a, 0x45, 0xae, 0x6a, 0x72, 0x5f, 0xaf, 0x54, 0x13, 0x93, ]
Doing ECDH Mapping agreement
Doing PACE Step3 - Key Exchange
Generated Ephemeral key pair
Ephemeral public key - [0x04, 0x97, 0xd1, 0x11, 0x28, 0xf6, 0xaf, 0x02, 0xee, 0xcc, 0xdd, 0x43, 0x7b, 0x9b, 0x5a, 0xf3, 0x6a, 0x7d, 0x5d, 0xbe, 0x41, 0x42, 0xe7, 0xda, 0x4f, 0xdd, 0x4f, 0xa8, 0xe8, 0x3b, 0xde, 0x55, 0xce, 0x65, 0x1f, 0x45, 0x2f, 0x1b, 0x9c, 0xe0, 0x3a, 0x6c, 0x7c, 0x55, 0x8f, 0x79, 0x57, 0xac, 0x38, 0x9c, 0x49, 0x5e, 0x5d, 0xf8, 0x29, 0x46, 0xbe, 0x38, 0x23, 0xb8, 0xba, 0xe1, 0x92, 0x20, 0x36, ]
Sending ephemeral public key to passport
TagReader - sending [0x10, 0x86, 0x00, 0x00, 0x45, 0x7C, 0x43, 0x83, 0x41, 0x04, 0x97, 0xD1, 0x11, 0x28, 0xF6, 0xAF, 0x02, 0xEE, 0xCC, 0xDD, 0x43, 0x7B, 0x9B, 0x5A, 0xF3, 0x6A, 0x7D, 0x5D, 0xBE, 0x41, 0x42, 0xE7, 0xDA, 0x4F, 0xDD, 0x4F, 0xA8, 0xE8, 0x3B, 0xDE, 0x55, 0xCE, 0x65, 0x1F, 0x45, 0x2F, 0x1B, 0x9C, 0xE0, 0x3A, 0x6C, 0x7C, 0x55, 0x8F, 0x79, 0x57, 0xAC, 0x38, 0x9C, 0x49, 0x5E, 0x5D, 0xF8, 0x29, 0x46, 0xBE, 0x38,  [... cut ...]
TagReader - sending [0x00, 0x82, 0x00, 0x00, 0x28, 0x03, 0x35, 0x1A, 0x06, 0x1D, 0xFB, 0xA3, 0x6E, 0x1E, 0x25, 0x7A, 0x6A, 0x86, 0xF4, 0x28, 0x9E, 0x44, 0xA8, 0x42, 0x5C, 0x5D, 0x9D, 0xBF, 0x90, 0x98, 0x7F, 0x83, 0x21, 0x67, 0xA5, 0x22, 0x9A, 0xD9, 0x50, 0x83, 0xD8, 0xB5, 0x2F, 0x7C, 0xAC, 0x00]
TagReader - Received response, size 0b
TagReader [unprotected] [], sw1:0x6a sw2:0x88
Error reading tag: sw1 - 0x6A, sw2 - 0x88
reason: Referenced data not found
tagReaderSession:didInvalidateWithError - Session invalidated by user

nicop2000, 2024-03-05:
Maybe it is the LDS version since it is used for finding data/files?
The working passport has 1.8, the others one I dont know

rbrouwer, 2024-04-07:
Is it by any chance a German ID? The IDs without epassport-logo do not actually implement ICAO, but have their own standard. On those you can do PACE, but "COM" is missing on them. "SOD" will give "Access denied". Any other DG1, DG2 and DG3 will also given "Access denied". All other DGs will be missing.

Nothing more you can do with those IDs. Possibly they open right up when doing Extended Access Control (EAC) / Terminal authentication (TA).

explit28, 2025-04-26:
Hm... My both German Passports are quit new. ID Card and Passport issued in July 2024....