AndyQ/NFCPassportReader #208: Sample app fails to read German passport

NickAtGit, 2024-04-08:
Hello, unfortunately I'm not able to read my passport. The MRZ is correct and passport works with other apps from the AppStore.

```
2024-04-08 3:56:24.2070 - Error reading tag: sw1 - 0x69, sw2 - 0x82
2024-04-08 3:56:24.2100 - reason: Security status not satisfied
2024-04-08 3:56:24.2100 - PACE Failed - falling back to BAC
2024-04-08 3:56:24.2250 - Starting Basic Access Control (BAC)
2024-04-08 3:56:24.3770 - Basic Access Control (BAC) - SUCCESS!
2024-04-08 3:56:24.3780 - Reading tag - COM
2024-04-08 3:56:24.4660 - Reading tag - DG14
2024-04-08 3:56:24.7650 - Performing Chip Authentication - number of public keys found - 2
2024-04-08 3:56:25.0790 - Restarting secure messaging using DESede encryption
2024-04-08 3:56:25.0810 - Reading tag - SOD
2024-04-08 3:56:25.6780 - Reading tag - DG1
2024-04-08 3:56:25.7660 - Reading tag - DG2
-[NFCTagReaderSession transceive:tagUpdate:error:]:893 Error Domain=NFCError Code=102 "Tag response error / no response" UserInfo={NSLocalizedDescription=Tag response error / no response, NSUnderlyingError=0x3032974b0 {Error Domain=nfcd Code=29 "Tag Error" UserInfo={NSLocalizedDescription=Tag Error, NSUnderlyingError=0x303297a80 {Error Domain=com.apple.nfstack Code=20 "No response from tag" UserInfo={NSLocalizedDescription=No response from tag}}}}}
```

danydev, 2024-04-08:
So, if I understand correctly, by just retrying multiple times you are able to make it work?.

Can you put logs with debug logs enabled? `let reader = PassportReader(logLevel: .debug)`

danydev, 2024-04-08:
you're right that's the old way to enable log level, that's not needed right now, you should already see\filter all kind of logs in the IDE.

@AndyQ do you find this kind of issue interesting? I mean, if ReadId is able to not fail, it probably does something slightly different? What it could be?
btw looks like the same issue reported here https://github.com/AndyQ/NFCPassportReader/issues/166

AndyQ, 2024-04-09:
It is really interesting actually, and a little odd that it can read some of the items (so security and comms channels are setup correctly).  The only thing I can really think of is maybe reading too quickly?

I haven't yet though come across a passport that has this behaviour so I'm unable to test this out though.

danydev, 2024-04-09:
@NickAtGit as you can see it's really hard to try debug this remotely.
Can you scan like 6/7 times and post the full logs in separated files? It's a a bit a shot in the dark, but maybe, Andy can see something in the logs that could shed some ideas about tweaks to test out.