AndyQ/NFCPassportReader #213: Library evolution issue

rounaq-jahan, 2024-04-30:
I am trying to add this package into my sdk and using it into another app but it is giving warning of Library Evolution while I am importing NFCPassportReader and app is not running, can you please look into. It is working fine when I am adding to the app directly though.

AndyQ, 2024-05-04:
Sorry without more information I have no idea what the issue could be.

rounaq-jahan, 2024-05-04:
It is probably because of OpenSSL version everything works fine but when I try to import OpennSSL it is throwing this error.

AndyQ, 2024-05-08:
What version of OpenSSL are you using?

rounaq-jahan, 2024-05-08:
Anything about 3.0

AndyQ, 2024-05-31:
Closing this as this is really related to https://github.com/AndyQ/NFCPassportReader/issues/217

closed 2024-05-31 by AndyQ