AndyQ/NFCPassportReader #214: DG11 return in reading nfc Vietnam Id Card

buivanhieu2105, 2024-05-03:
Hi @AndyQ ,

Could u please support for reading DG11?

I have built the your example but can not get data dg 11

My log:
tagReaderSession:didDetect - found [CoreNFC.NFCTag.iso7816(<NFCISO7816Tag: 0x281f1c6f0>)]
16:11:28.034143+0700 NFCPassportReaderApp[7701:2242661] [passportReader] DG Found - ["DG1", "DG2", "DG3", "DG13", "DG15", "DG14"]

[image]

(lldb) po dataGroupsRead[.DG15] as? DataGroup15
▿ Optional<DataGroup15>
  ▿ some : <DataGroup15: 0x28047ca80>
(lldb) po dataGroupsRead[.DG11] as? DataGroup11
nil
(lldb) po dataGroupsRead[.DG1] as? DataGroup1
▿ Optional<DataGroup1>
  ▿ some : <DataGroup1: 0x281f725e0>
  
  
  Thanks!

rbrouwer, 2024-05-06:
Are you certain that that document actually have DG11?

Because that line "DG Found -" is not mentioning DG11 and that is basically coming from the passports COM-file, which contains which DataGroups are contained within that document. Given THAT does not contain DG11, it is likely that the document simply does not have a DG11. That is optional to have!

buivanhieu2105, 2024-05-07:
Hi @rbrouwer,

Thank you for your answer, 
I'm researching for nfc reader of ID Citizen Card, so If you have any document for ID Citizen Card, please help to sharing it
Thanks a lot!

rbrouwer, 2024-05-07:
I do not have such a document. But the logs you shared indicate that the document you used to scan that document does not have it.

A France passport I do have, does have DG1, 2, 3, 11, 12, 13 and 14. Those those 3 and 13 are skipped (DG 3 - fingerprints would not open as that would require a terminal certificate and DG 13 is defined as "Data Elements combining to form Data Group 13 (DG13) are at the discretion of the issuing State or organization", which therefore not supported by this lib.)
DG11 reads just fine in that case! So make sure the document you scan actually contains DG11 and then it should work.

buivanhieu-bstar, 2024-05-08:
> I do not have such a document. But the logs you shared indicate that the document you used to scan that document does not have it.
> 
> A France passport I do have, does have DG1, 2, 3, 11, 12, 13 and 14. Of those DG 3 and DG 13 are skipped (DG 3 - fingerprints would not open as that would require a terminal certificate and DG 13 is defined as "Data Elements combining to form Data Group 13 (DG13) are at the discretion of the issuing State or organization", which therefore not supported by this lib.) DG11 reads just fine in that case! So make sure the document you scan actually contains DG11 and then it should work.

Thank you for your useful information, so I can understand more about the data groups that NFC supports

closed 2024-05-08 by buivanhieu-bstar