AndyQ/NFCPassportReader #218: Some Portuguese and Spanish passports will not have image from DG2

rbrouwer, 2024-05-06:
The images in Portuguese and Spanish passports have images encode in JPEG2000 format with JPEG2000 features that iOS does not support.

Therefore:
https://github.com/AndyQ/NFCPassportReader/blob/306eac9e0de491128f7d8d595cc10ca114acea88/Sources/NFCPassportReader/DataGroups/DataGroup2.swift#L46
and
https://github.com/AndyQ/NFCPassportReader/blob/306eac9e0de491128f7d8d595cc10ca114acea88/Sources/NFCPassportReader/DataGroups/DataGroup7.swift#L30

will return nil, because UIImage is unable the decode the imagedata which is correctly read by the library.

Workaround would be to integrate libopenjp2 (or any other jpeg2000 lib) to decode/transcode the image into something UIImage would understand or hope iOS would complete their JPEG2000 integration within iOS/MacOS.

AndyQ, 2024-05-08:
I did have a little look at JPEG2000, but at the moment, its not something I'm going to add support for (unless for some reason it suddenly become really popular) as it seems to be really uncommon - I haven't yet seen one.  However, I do provide access to the raw image data so you could add support if it was really important for you.

closed 2024-05-08 by AndyQ

rbrouwer, 2024-05-08:
It was more of a documentation case. I already have an extension for NfcPassportModel giving me a UIImage in those cases.

Although I am not fully certain, I am fairly certain images, which start with jpeg2000CodestreamBitmapHeader, will very likely not work using that case. All you could consider is to handle (UnsupportedImageFormat error or such) and/or document it. That way other devs will not have to dig as deep to figure out while that specific document does properly have DG 2, but not have any image show up.

AndyQ, 2024-05-08:
Thanks for the comment - I'll add this to the docs!

jonathanellis, 2024-08-15:
If anyone wants to add support for JPEG2000, I have a detailed write-up of how to do so, here - using libopenjp2: https://github.com/jonathanellis/ios-jpeg2000

masadchattha, 2025-04-16:
> It was more of a documentation case. I already have an extension for NfcPassportModel giving me a UIImage in those cases.
> 
> Although I am not fully certain, I am fairly certain images, which start with jpeg2000CodestreamBitmapHeader, will very likely not work using that case. All you could consider is to handle (UnsupportedImageFormat error or such) and/or document it. That way other devs will not have to dig as deep to figure out why that specific document does properly have DG 2, but not have any image show up.

@rbrouwer Could you please share how you were able to successfully get a UIImage in this case? I’m currently facing the same issue — the image data is present, but UIImage(data:) returns nil.

masadchattha, 2025-04-17:
I converted the image data into a JP2 file and exported it, then attempted to process it using @jonathanellis 's [iOS-JPEG2000](https://github.com/jonathanellis/ios-jpeg2000) code. However, the app crashed during the process.

The exported JP2 file could not be opened on iOS/macOS, so I used the [Convertio](https://convertio.co/) API as a temporary workaround to convert it into a PNG for use in the project. However, I’m not satisfied with this solution and am seeking a more robust alternative.

This issue specifically affects Pakistan’s e-Passport, whereas the Hungarian card processes without any problems.

rbrouwer, 2025-04-17:
I would not know if my current solution would work for the Pakistan e-Passport.

I do not happen to have that one over here to test. Given we are talking about actual photographs I doubt any sharing of the data would happen. I could drop you an email, if you would be open share.

I do know after performing that write-up, I did indeed need to do additional tweaking to handle all my known cases.

masadchattha, 2025-04-18:
Hi @rbrouwer 
Just a quick note to let you know I’ve sent you an email at **r.brouwer@idcheckcenter.com** with a face.jp2 file attached. Looking forward to hearing if your workaround helps in this case. Thanks!