AndyQ/NFCPassportReader #219: Can not select Master File to read EF.CardAccess (for PACE) with Vietnam ID card (error: 6D 00)

longcon9999, 2024-05-06:
I always got error 6D 00 (Instruction code not supported or invalid) when select master file, so I can not using PACE. Please help me, @AndyQ. Thank you

daniel-e-gonzalez, 2024-05-08:
Hey @AndyQ . this issue is a little bit different than the exact error messages i've seen in other open/closed issues, but i wanted to let you know i found something interesting with master file selection for PACE (that might also fix this one). 

see 
https://github.com/AndyQ/NFCPassportReader/issues/156
https://github.com/AndyQ/NFCPassportReader/issues/164
https://github.com/AndyQ/NFCPassportReader/issues/116

there might be more too, but i think this is a good sampling.

what i found was that if i use 0x02 for the p1 parameter in readCardAccess in TagReader, the card access is able to be selected. i am not nearly as well versed in this as you are, so my terminology might be a bit wrong, but i think for some nfc chips the card access file might actually live in the applet rather than in the master file. i think by sending 0x02, we are just going and selecting the EF file directly. see 3.6.2 EF Selection Using the SELECT Command in ICAO 9303. 

I have a Latvian ID that was giving the exact same error as in https://github.com/AndyQ/NFCPassportReader/issues/116, and when I made this change it fixed PACE. It also seems like using it as a fallback does work, at least for the Latvian ID. it doesn't make me rescan or anything when I try a different p1 param.

longcon9999, 2024-05-09:
> Can you please supply a log? And are you sure your passport supports PACE?

Here is log:
2024-05-09 5:15:21.6550 PM - tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2836f8a80>)
2024-05-09 5:15:21.6550 PM - tagReaderSession:connected to tag - starting authentication
2024-05-09 17:15:21.655426+0700 NFCReaderSwift[60133:1676843] [CoreNFC] 00000002 80a90b40 -[NFCTagReaderSession setAlertMessage:]:93  (null)
2024-05-09 5:15:21.6560 PM - TagReader - sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00]
2024-05-09 5:15:21.6630 PM - TagReader - Received response size 0b
2024-05-09 5:15:21.6630 PM - TagReader [unprotected] [], sw1:0x6d sw2:0x00
2024-05-09 5:15:21.6640 PM - TagReader - sending [0x00, 0xA4, 0x02, 0x0C, 0x02, 0x01, 0x1C]
2024-05-09 5:15:21.6690 PM - TagReader - Received response size 0b
2024-05-09 5:15:21.6690 PM - TagReader [unprotected] [], sw1:0x69 sw2:0x82
2024-05-09 5:15:21.6690 PM - PACE Failed - falling back to BAC

I received [0x6d sw2:0x00](sw1:0x6d sw2:0x00) when call tagReader.readCardAccess()
sending [0x00, 0xA4, 0x00, 0x0C, 0x02, 0x3F, 0x00] is select master file, because of select master file 6d 00 (Instruction code not supported or invalid), when code read EC.CardAccess got error  sw1:0x69 sw2:0x82 (Security status not satisfied)

Please help me, I really need using PACE.
My ID card can PACE using jmrtd of android

daniel-e-gonzalez, 2024-05-10:
@longcon9999 it isn't commenting out the first select entirely that fixed the "Incorrect P1-P2 parameters" error, it was just changing the P1 parameter from 0x00 to 0x02 i.e. 

let cmd = NFCISO7816APDU(instructionClass: 0x00, instructionCode: 0xA4, p1Parameter: 0x02, p2Parameter: 0x0C, data: Data([0x3f, 0x00]), expectedResponseLength: -1).

i don't know for sure if it would work for your case since your error is different, but if you are trying things then I figured it would be worth a shot to make sure you are trying the fix correctly. like i said though, i only am sure that this fixes the "Incorrect P1-P2 parameters" specifically during the readCardAccess function.

longcon9999, 2024-05-21:
> let cmd = NFCISO7816APDU(instructionClass: 0x00, instructionCode: 0xA4, p1Parameter: 0x02, p2Parameter: 0x0C, data: Data([0x3f, 0x00]), expectedResponseLength: -1).
I got **Authentication method blocked** when send this cmd.
[image]
I tried send this cmd but it did not work, still got **Instruction code not supported or invalid** . Please let me know some solutions, @AndyQ thank you

yen-dang-ts, 2024-05-29:
I got same issue

lttrung2001, 2024-11-02:
i commented select master file apdu and just execute select card access file only with apdu: 00 A4 02 0C 02 01 1C then I received 9000 (success) response with my Vietnam ID card

longcon9999, 2024-11-04:
> i commented select master file apdu and just executed select card access file only with apdu: 00 A4 02 0C 02 01 1C then I received 9000 (success) response with my Vietnam ID card

[image]
Like this, right? I tried but it did not work

lttrung2001, 2024-11-06:
> ![image](https://private-user-images.githubusercontent.com/85325260/382734623-378214b5-cf47-40ec-9c50-d2bf79e63b73.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Try readCardAccess() with Data [0x01, 0x1C]

longcon9999, 2024-11-06:
> Try readCardAccess() with Data [0x01, 0x1C]

[image]
I tried like this, still sw1 = 109, sw2  = 0, data = 0 bytes, Any ideas, please help me, thank you

lttrung2001, 2024-11-06:
> ![image](https://private-user-images.githubusercontent.com/85325260/383454182-3ba92edf-1b95-4efe-a610-bef3212ff0b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I implemented select card access file apdu with Android base on this repo and it works. Have you try to change p1Parameter of readCardAccess() to 0x02? If not I'm afraid i can't help further.

luongpp, 2024-11-22:
> i commented select master file apdu and just executed select card access file only with apdu: 00 A4 02 0C 02 01 1C then I received 9000 (success) response with my Vietnam ID card

Hi @lttrung2001, When you receive 9000 what next you do for decode the response?

lttrung2001, 2024-11-27:
> > i commented select master file apdu and just executed select card access file only with apdu: 00 A4 02 0C 02 01 1C then I received 9000 (success) response with my Vietnam ID card
> 
> Hi @lttrung2001, When you receive 9000 what next you do for decode the response?

You should send read binary of card access file. But I don't know how to read binary sequently or decode response

HieeusNguyen, 2024-12-18:
Has anyone found a solution to that problem?

longcon9999, 2025-02-17:
> Has anyone found a solution to that problem? [@longcon9999](https://github.com/longcon9999)

I still don't have any solution

longcon9999, 2025-10-27:
> [@longcon9999](https://github.com/longcon9999) do you have any solution, i have this error too

Well, same answer, still don't have any solution, bro