AndyQ/NFCPassportReader #223: Error identifying Active Authentication Hash Algorithm - Chinese ePassport

rbrouwer, 2024-05-23:
The solution of #194 broke AA for recent Chinese passports.

Those will return 0 bytes when giving them an Expected Response Length of 65536. When undoing the changes to secure messaging and setting the Expected Response Length to 256. Those work just fine again.

I guess behaviour such as in JMRTD' AAAPDUSender is wanted; Guessing the correct length and even retrying with 65536 when 256 did not make sense.

rbrouwer, 2024-05-23:
Setting Expected Response Length to 231 on [this line](https://github.com/AndyQ/NFCPassportReader/blob/32cc702d9f66cd8370ba1f6d2e59ad7eae284f4a/Sources/NFCPassportReader/TagReader.swift#L55) works for those documents as well. Guessing the extended format is not being liked.

AndyQ, 2024-05-29:
Thats really annoying!  will have a re-think

rbrouwer, 2024-05-31:
I do not believe that would work. Some documents (like that Australian passport in #194) will actually return OK with a partial signature (cutting off whatever did not fit). The implementation in aa_test would return that response and not go for an extended read.

This of course would work fine for this one document I now mentioned in this issue, but that other one would again break.

AndyQ, 2024-05-31:
Hmmm thats a good point.  Wonder if it would work the other way round?  try first with extended read and if that failed then try standard?

AndyQ, 2024-05-31:
Ah of course - small change! - currently doesn't check error - any better?

I'm really interested if the passport will allow the normal read after the extended read fails!

rbrouwer, 2024-05-31:
Well... Apparently that will give you:
(Extended try)
sw - 0x67, sw2 - 0x00, and no data

and then 
(Normal try)
sw - 0x69, sw2 - 0x88, and no data

So apparently not.

furkanisik000, 2024-06-28:
?

baskurtbey, 2024-07-03:
@rbrouwer  Did you solve the problem

baskurtbey, 2024-07-03:
The `doInternalAuthentication` function is actually not working. Active Authentication (AA) is not being created.

AndyQ, 2024-07-03:
> Well... Apparently that will give you: (Extended try) sw - 0x67, sw2 - 0x00, and no data
> 
> and then (Normal try) sw - 0x69, sw2 - 0x88, and no data
> 
> So apparently not.

Thats a shame!  Not quite sure what to do here really.  I could add a flag to enable extended mode BUT you'd have to know that the passport supports/doesn't support that.  I'll did a little more into JMRTD code and see if I can see what they did to solve this.  I had a quick look thorugh and they had some functions in bouncy castle that could get certain info about the keys that I haven't yet figured out how to do (I'm sure its possible though).

AndyQ, 2024-07-03:
> The `doInternalAuthentication` function is actually not working. Active Authentication (AA) is not being created.

Could you try with version 2.1.1 and see if that works for you (that doesn't enable extended read) and has been working for most passports (except newer Australian and probably other newer ones)?

baskurtbey, 2024-07-03:
Hello, I tried but unfortunately there was an issue. It doesn't decode, most likely. The ID is a Turkish ID and it should normally be supported.

baskurtbey, 2024-07-03:
It works seamlessly on Android using JMRTD.

AndyQ, 2024-07-03:
So what I've done for the moment is instead of enabling extended mode for AA by default, we now have an additional flag set on PassportReader init - useExtendedMode.  Not ideal but will think more.

Also, as per #224 - We're now using OpenSSL 1.1.2300 which is a signed release and includes a privacy policy.

Currently main branch only,

baskurtbey, 2024-07-03:
The latest update solved my problem, thank you. You saved me from a big issue; I had been researching it for a week.  🙏❤️

danydev, 2024-07-09:
@AndyQ looks like it's working for @baskurtbey maybe it's time for a release? ;)

AndyQ, 2024-07-09:
It's definitely getting closer. I'm still not 100% happy as I do t yet know what passport support extended reads and which don't!  It looks like some Chinese ones don't but I'm sure there are others too. So I can't yet say when you should use extended read or not. JMRTD does some best guesses based on key lengths BITBi haven't yet figured out how to replicate that in OpenSSL (JMRTD uses bouncycastle which has different APIs).

tomgi, 2024-07-10:
@andyq for what it's worth, ICAO specifies in https://www.icao.int/publications/documents/9303_p11_cons_en.pdf that
> Note.— It should be noted that when using key lengths exceeding 1 848 bits (if Secure Messaging with 3DES is used) / 1 792 bits (if Secure Messaging with AES is used) in Active Authentication with Secure Messaging, Extended Length APDUs MUST be supported by the eMRTD chip and the Inspection System.

For Australian R-series EPassports, DG14 includes `id-PACE-ECDH-GM-AES-CBC-CMAC-256` ([0.4.0.127.0.7.2.2.4.2.4](https://oid-rep.orange-labs.fr/get/0.4.0.127.0.7.2.2.4.2.4)) as the only supported PACE algorithm. 
This is a 256-bit key, so based on that we know the chip MUST support extended length APDUs.

Logs from [here](https://github.com/AndyQ/NFCPassportReader/blob/89813bafed0490d1dab17e1c2c243e41bebcd455/Sources/NFCPassportReader/PACEHandler.swift#L94) are:
```
doPace - inpit parameters
paceOID - 0.4.0.127.0.7.2.2.4.2.4
parameterSpec - 933
mappingType - Generic Mapping
agreementAlg - ECDH
cipherAlg - AES
digestAlg - SHA-256
keyLength - 256
```

AndyQ, 2025-12-10:
The problem here is that you need to know ni advance whether extended mode is supported or not.  Its not something you can try and if it fails switch to non-extended mode.  The NFC session gets aborted and you have to restart with the discovery again (which is a pain). I did spend some time looking at this but haven't yet figured out a way round this.

closed 2026-11-09 by AndyQ