AndyQ/NFCPassportReader #224: ITMS-91065: Missing signature - OpenSSL.framework

ecubo, 2024-05-27:
Hello! Congratulations on the work. I'm running some tests and encounter this problem with Apple:

TMS-91065: Missing signature - Your app includes “Frameworks/OpenSSL.framework/OpenSSL”, which includes BoringSSL / openssl_grpc, an SDK that was identified in the documentation as a privacy-impacting third-party SDK. If a new app includes a privacy-impacting SDK, or an app update adds a new privacy-impacting SDK, the SDK must include a signature file. Please contact the provider of the SDK that includes this file to get an updated SDK version with a signature. For details about verifying the code signature for a third-party SDK, visit: https://developer.apple.com/documentation/xcode/verifying-the-origin-of-your-xcframeworks.

AVGThor, 2024-05-30:
you just need to update OpenSSL to new version (3.1.5001), rebuild your app and publish it again to test flight

dcrespim, 2024-05-30:
When update OpenSSL to version 3.1.xx, fail PACE authentication

AVGThor, 2024-05-30:
> authentication

Oh i use BAC so i don't know how to fix

AVGThor, 2024-05-30:
I failed on PACE can you show me how to fix it

dcrespim, 2024-05-30:
I'm working on it

hardeflo, 2024-06-18:
There is a new **signed** 1.1.X Release of OpenSSL. This should do for the moment, right? Upgrading to 3.X is probably tracked in https://github.com/AndyQ/NFCPassportReader/issues/217.

New Signed Version of OpenSSL: https://github.com/krzyzanowskim/OpenSSL/releases/tag/1.1.2300

AndyQ, 2024-07-03:
This has been tested and implemented in current main branch (no release quite yet) - please test and report any issues

jgtoriginal, 2025-09-10:
Hello @AndyQ 

What's the ETA for this to be released and available to download from Pod?