AndyQ/NFCPassportReader #232: Library not loaded: @rpath/OpenSSL.framework/OpenSSL

MMavrisPaleBlue, 2024-10-16:
Hi Andy, thank you for your great work.

When adding the NFCPassportReader as a dependency (SPM) of an SDK we are developing, we get this error/crash when we build and run for a device (it's ok for simulators):

```
dyld[15758]: Library not loaded: @rpath/OpenSSL.framework/OpenSSL
  Referenced from: <7B9E58BE-60E2-35D6-9D38-58D4A21BF161> /private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/Example.framework/Example
  Reason: tried: '/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/preboot/Cryptexes/OS/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/Example.framework/Frameworks/OpenSSL.framework/OpenSSL' (no such file), '/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/preboot/Cryptexes/OS/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/OpenSSL.framework/OpenSSL' (no such file), '/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/preboot/Cryptexes/OS/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/OpenSSL.framework/OpenSSL' (no such file), '/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/preboot/Cryptexes/OS/Users/michaelmavris/Library/Developer/Xcode/DerivedData/Example-Example-awzqsefhcffvvxcyvaaooyvprnrs/Build/Products/Debug-iphoneos/PackageFrameworks/OpenSSL.framework/OpenSSL' (no such file), '/private/var/containers/Bundle/Application/C7AAAE47-708B-4A47-9662-3B490879727A/Example-Example.app/Frameworks/OpenSSL.framework/OpenSSL' (no such file)
```

Any ideas what's happening?

closed 2024-10-18 by MMavrisPaleBlue

ersinkhr, 2024-12-20:
i am having same issue. did you find solution for this

MMavrisPaleBlue, 2024-12-20:
This was happening for the "example app" that was including our SDK. I added the OpenSSL as a dependency directly for the example app: https://github.com/krzyzanowskim/OpenSSL.git and got it working