AndyQ/NFCPassportReader #253: Add support for ISO/IEC 39794-5

jakub-vallo, 2025-07-29:
Hello, I would like to ask if support for ISO/IEC 39794-5 standard will be implemented in the near future, i.e. this year. ISO/IEC 39794-5 standard defines new format for storing biometric data in data groups DG2, DG3, DG4. I am mainly interested in DG2. Thanks.

TLOn1, 2025-09-02:
I have the same question. I'm also interested in DG2.

AndyQ, 2025-09-02:
I've currently no plans on supporting this yet (I currently don't have access to passports that support this), but when I do then its definately something that I'll be looking into to.  Of course, if anyone has access to DG2 info in this format I'd be happy to look into it.

TWilb, 2025-09-05:
Currently it is not allowed by ICAO to have this format in the documents, therefore you can't find public ones.
To prepare for the optional introduction in 2026 there is a sample dataset: https://github.com/ICAO-TRIP-ISO-WG3/39794-5-AP

SteveDove, 2025-12-18:
Hi there @AndyQ. Adding a new comment to this issue from a few months back, to register interest in the same (support for the  ISO/IEC 39794-5 standard, specifically DG2). I wondered if documents were now available to enable development / changes, or if the sample dataset from the previous comment provided any help. Many thanks in advance for any work completed in this area.